RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #78: Can this be firmware hacked?

NonTaxx, 2024-04-18:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

I-hate-2FA, 2024-04-18:
yes just the usual f130c8 but without gate driver

NonTaxx, 2024-04-19:
I don't know where to connect the STLink

I-hate-2FA, 2024-04-19:
either you are one of the bdc or you didnt read the wiki

I-hate-2FA, 2024-04-19:
gnd and 3v3 is correct

NonTaxx, 2024-04-20:
[image]
When I try to connect it beeps and I get an error message

NonTaxx, 2024-04-20:
Should I upgrade the firmware?
[image]

RoboDurden, 2024-04-20:
Sorry, no time (=happiness) to help you with this beginner steps.
Watch youtube and try autoflash.bat: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/BinariesReadyToFlash

I-hate-2FA, 2024-04-20:
it could be everything really if you ask me ill tell you to use serial flashing because itll 100% work

NonTaxx, 2024-04-20:
I succesfully connected it but I don't know which .bin should I flash in

NonTaxx, 2024-04-20:
I'm green on this stuff, I want it to be a remote controlled hoverboard so I could use it to control my lawn mower. I did my research and found on youtube that I need to flash custom firmware therefore I can connect it to my RC receiver.

I-hate-2FA, 2024-04-20:
autodetect

RoboDurden, 2024-04-20:
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/BinariesReadyToFlash

NonTaxx, 2024-04-20:
[image]

NonTaxx, 2024-04-20:
Which one should I flash in? There are a bunch of versions

RoboDurden, 2024-04-20:
please start using your brain !

NonTaxx, 2024-04-20:
Well you told it is 2.1.6 but there is no 2.1.6 .bin file in all of the repository

I-hate-2FA, 2024-04-20:
i told you to use autodetect,there is 4 option left,you know your mcu number so 2 left,i dont know which one to use either because the picture quality is shit no one can trace it

you are the type of person that dont deserve to use the firmware

RoboDurden, 2024-04-20:
Well everyone is free to use my firmware.
And no one at all deserves my help.
Because we all should be free :-)

closed 2024-05-31 by NonTaxx