RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #80: Questions about AutoDetect

hiliving, 2024-05-01:
My board is GD32F130C6T6, and the serial port pin is PA2 PA3. I flashed in the bin named hoverboard autodetect GD32F130 tx=PA2, rx=PA3. bin 
[image]

and open the serial terminal in arduino like this:
[image]
 
I used a self-locking button to bridge the power switch, followed by pressing the enter key, then pressing v, and then pressing s according to the prompts, but I found that my interface remained on this interface and did not display the list of automatically detected pins. Like in your video,
I have watched your  video many times, but I haven't found any errors. Could you please write down the specific steps so that I can check which step is problematic. thx very much.
[image]

RoboDurden, 2024-05-01:
Maybe you need to press enter in order to send the 'v' or 's' to the hoverboard.

Thank you for testing my autodetect.

hiliving, 2024-05-01:
Yes, I pressed v and also pressed s, but the process seems to be looping back to the beginning, trying to bridge the power switch again, or returning to the original interface

hiliving, 2024-05-01:
[image]
[image]
[image]
[image]

RoboDurden, 2024-05-01:
I do not see that you cycle through all analog pins  by hitting enternto find the correct one.
Your screensh

RoboDurden, 2024-05-01:
Is the motor spinning for 2 seconds every two seconds in this first stage ,?

hiliving, 2024-05-01:
yeah ,it spining like your video.

hiliving, 2024-05-01:
[image]
I am using a 24V power supply,When A4 appeared, I pressed V and then pressed S,When l is pressed, the motor restarts spinning

hiliving, 2024-05-01:
try A5
PA5: 39.47 V +- 0.00
PA5: 39.84 V +- 0.37

try A7
PA7: 48.70 V +- 0.00
PA7: 49.91 V +- 1.21
PA7: 51.31 V +- 1.40

try B1
PB1: 98.73 V +- 0.00

try A0
PA0: 98.86 V +- 0.00
PA0: 98.89 V +- 0.03

try A5
PA5: 39.83 V +- 0.00
PA5: 39.82 V +- 0.01

try A7
PA7: 47.19 V +- 0.00
PA7: 50.50 V +- 3.31

try B1
PB1: 98.72 V +- 0.00

try A0
PA0: 98.87 V +- 0.00
PA0: 98.88 V +- 0.00
PA0: 98.86 V +- 0.01

try A5
PA5: 39.81 V +- 0.00
PA5: 38.36 V +- 1.45

try A7
PA7: 47.68 V +- 0.00
PA7: 45.73 V +- 1.95
PA7: 48.75 V +- 3.02

try B1
PB1: 98.75 V +- 0.00

try A0
PA0: 98.86 V +- 0.00
PA0: 98.87 V +- 0.01
PA0: 98.87 V +- 0.00

try A5
PA5: 38.26 V +- 0.00

try A7
PA7: 48.58 V +- 0.00
PA7: 48.71 V +- 0.13

try B1
PB1: 0.34 V +- 0.00
PB1: 0.34 V +- 0.00
PB1: 98.72 V +- 98.38

try A0
PA0: 98.87 V +- 0.00

PA0: 98.86 V +- 0.01
PA5: 39.81 V +- 0.00
PA5: 39.83 V +- 0.02
PA5: 39.83 V +- 0.00

try A7
[... 1029 lines cut ...]
PA5: 38.21 V +- 0.75
PA5: 38.19 V +- 0.02
PA5: 39.66 V +- 1.47
PA5: 39.67 V +- 0.01
PA5: 39.66 V +- 0.01
PA5: 39.55 V +- 0.11
PA5: 38.30 V +- 1.25
PA5: 38.25 V +- 0.05
PA5: 39.65 V +- 1.41
PA5: 39.66 V +- 0.01
s

#define VBATT		PA0
now detecting SELF_HOLD pin,
'v'=battery voltage
'x'=delete pin	'l'=list,	'c'=reset and 's'=next stage
ENTER for next pin
now A0
press ENTER to begin or 's' for next stage.

:-)
:-)
:-)
:-)
:-)

hiliving, 2024-05-01:
I have replaced a serial port terminal that can display sending data. Perhaps you can see something, as if it is repeating the # define VBATT now detecting SELF_HOLD pin

hiliving, 2024-05-01:
You and I are both on the Discord channel. If it's convenient, you can find me there

RoboDurden, 2024-05-01:
Yes it seems that it is rebooting when stage AUTODETECT_Stage_Hold is run :-/
When this stage is initialized, all remaining pins are set as OUTPUT and HIGH.

```
	case AUTODETECT_Stage_Hold:
		for (i=0;i<COUNT_PinDigital; i++)	
		{
			if (!(aoPin[i].wState & STATE_HIDE))
			{
				//gpio_deinit(aoPin[i].i);
				pinMode(aoPin[i].i,GPIO_MODE_OUTPUT);
				digitalWrite(aoPin[iTest].i,1);	// every remaining io pin might be the SELF_HOLD pin
			}
		}
```

Perhaps your board has a pin that resets the board :-(

I have uploaded a new autodetect binary with different stage order so that you can skip this stage for now:

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesToTest/hoverboard%20autodetect%20B%20GD32F130%20tx%3DPA2%2C%20rx%3DPA3.bin

```
	#define AUTODETECT_Stage_Startup 	1
	#define AUTODETECT_Stage_VBatt 		2
	#define AUTODETECT_Stage_Led 				4
	#define AUTODETECT_Stage_Hall 			8
	#define AUTODETECT_Stage_HallOrder 	16
	#define AUTODETECT_Stage_Hold 			32
	#define AUTODETECT_Stage_Button 		64
	#define AUTODETECT_Stage_CurrentDC 128
	#define AUTODETECT_Stage_Results 	256
	#define AUTODETECT_Stage_Finished	512
```

hiliving, 2024-05-01:
Great, thx, I will try this version

RoboDurden, 2024-05-01:
Your board seems to be a variation (different pcb version) of https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/target_1%3DGD32F130/v6%3D2.5

There is a test firmware `hoverboard 2.1.5 master Dummy.bin` you could already try.
I do not remember if this board has been traced properly.
```
// ADC defines
#define VBATT	PA0			// Batman313v, might be CURRENT_DC !!!
#define CURRENT_DC	PA1		// Batman313v, might be VBATT !!!
```

But it would be nice if you continue testing my autodetect.bin. This will help to improve it.

RoboDurden, 2024-05-01:
i do not use these modern messangers..

I-hate-2FA, 2024-05-01:
thats the reason to not use pinmode output in autodetect
to just power off while detect self hold
and maybe the reason to have a main menu

RoboDurden, 2024-05-01:
Thanks AiLife for helping here.
I have changed the code to

```
		if ((buzzerTimer%3000) < 100)	// relase SELF_HOLD for a short time
		{
			pinModePull(aoPin[i].i,GPIO_MODE_INPUT,GPIO_PUPD_PULLDOWN);
		}
		else
		{
			pinModePull(aoPin[i].i,GPIO_MODE_INPUT,GPIO_PUPD_PULLUP);
		}
```

@hiliving , please give that `hoverboard autodetect C GD32F130 tx=PA2, rx=PA3.bin` a try.

hiliving, 2024-05-02:
i flashed the firmware you uploaded , it works， I  print the pin list like below：
```
//#define LED_RED		P??
//#define LED_ORANGE		P??
#define LED_GREEN		PB8
#define UPPER_LED		PB8
//#define LOWER_LED		P??
//#define ONBOARD_LED		P??
#define BUZZER		PB9

hall and phase currents: 'r'=restart	's'=save
try 1 = C14	HALL
try 2 = C15	
try 3 = F0	
try 4 = F1	
try 5 = A0	
try 6 = B11	HALL
try 8 = A5	
try 9 = A6	
try 10 = A7	
try 11 = B0	
try 12 = B1	
try 13 = B2	
try 14 = B10	
try 15 = A1	HALL
try 16 = B12	
try 17 = A11	
try 18 = F6	
try 19 = F7	
try 20 = A12	
try 21 = A15	
try 22 = B3	
try 23 = B4	
try 24 = B5	
try 25 = B6	
try 26 = B7	
try 28 = B9	
#define HALL_A		PC14
#define HALL_B		PB11
#define HALL_C		PA1

#define HALL_A		PC14
#define HALL_B		PA1
#define HALL_C		PB11
//#define PHASE_A		P??
//#define PHASE_B		P??
//#define PHASE_C		P??
now detecting SELF_HOLD pin,'v'=battery voltage
'x'=delete pin	'l'=list,	'c'=reset and 's'=next stage
ENTER for next pin
now A0
press ENTER to begin or 's' for next stage.

:-)
:-)
:-)
:-)
```

and I feel that some options can be optimized, such as “s” being the next stage in some places and saving in others, which is easy to confuse.  

int this section , I found that "l" didn`t work , Does it have any other function？
```
//#define VBATT		P??
'r'=red,	'o'=orange,	'g'=green,	'u'=up,	'd'=down,	'p'=pcb led,	'b'=buzzer
'x'=delete pin	'l'=list,	'c'=reset and 's'=next stage
ENTER for next pin
now C13
try C15

try F1

try B11

try A5

try A7

try B1

try B10

try B12

try F6

try A12

try B3
g
try B5

try B7

try B9
b
try C14

try F0

try A0

try A4

try A6

try B0

try B2

try A1

try A11

try F7

try A15

try B4
r
try B6

try B8

try C13

try C15

try F1

try B11

try A5

try A7

try B1

try B10

try B12

try F6

try A12

try B3

try B5
l
try B6
l
try B7
l
try B8
l
try B9
l
try C13
```

RoboDurden, 2024-05-02:
No, 'l' should print the definition list also in stage

hiliving, 2024-05-02:
right , this is B， now I will test C，sorry,  I wasn`t realized there was already C until you told me

RoboDurden, 2024-05-02:
According to your log ouput you have

```
#define LED_RED		PB4
//#define LED_ORANGE		P??
#define LED_GREEN		PB3
//#define UPPER_LED		P??
//#define LOWER_LED		P??
//#define ONBOARD_LED		P??
#define BUZZER		PB9

#define HALL_A		PC14
#define HALL_B		PA1
#define HALL_C		PB11
```

Make sure you have the 3 color led module attached.

autodetect C has the old stage order but AUTODETECT_Stage_Hold might succeed now.

RoboDurden, 2024-05-02:
I will setup a Gen2.1 test setup today or tomorrow so i can verify your findings for a better user interface :-)

RoboDurden, 2024-05-02:
@hiliving you have an impressive list of github forks ! Are your from China or Taiwan ? If you know someone who cares about peace and politics, please forward my world peace plan: www.RoboDurden.de , thanks !

RoboDurden, 2024-05-02:
I now see that the LED stage also uses OUTPUT:

hiliving, 2024-05-02:
> According to your log ouput you have
>
> ```
> #define LED_RED		PB4
> //#define LED_ORANGE		P??
> #define LED_GREEN		PB3
> //#define UPPER_LED		P??
> //#define LOWER_LED		P??
> //#define ONBOARD_LED		P??
> #define [...]

it doesnt matter for LED now ,it can be done in the end,  i want to control the motor through serial first .
by the way,supporting peace is a responsibility that everyone should bear, and I am glad to contribute.

hiliving, 2024-05-02:
This is the log for firmware C,
```
'v'=battery voltage
'x'=delete pin	'l'=list,	'c'=reset and 's'=next stage
ENTER for next pin
now A0
press ENTER to begin or 's' for next stage.
:-)
:-)
:-)
:-)
:-)
:-)
:-)
:-)
:-)
:-)
:-)
try A4
PA4: 24.05 V +- 0.00
PA4: 24.07 V +- 0.02
PA4: 24.05 V +- 0.01
PA4: 23.11 V +- 0.94
PA4: 22.85 V +- 0.26
PA4: 23.03 V +- 0.17
PA4: 22.59 V +- 0.43
PA4: 22.44 V +- 0.15
PA4: 24.05 V +- 1.60
PA4: 24.04 V +- 0.00
PA4: 24.06 V +- 0.02
PA4: 24.05 V +- 0.01
PA4: 24.06 V +- 0.01
PA4: 22.78 V +- 1.28
PA4: 22.55 V +- 0.24
v
try A6
PA6: 49.03 V +- 0.00
PA6: 49.29 V +- 0.25
PA6: 49.47 V +- 0.18
PA6: 47.69 V +- 1.78
PA6: 47.73 V +- 0.04
PA6: 47.74 V +- 0.01
PA6: 47.71 V +- 0.03
PA6: 47.74 V +- 0.04
PA6: 49.31 V +- 1.57
PA6: 48.84 V +- 0.47
PA6: 49.18 V +- 0.33
PA6: 49.35 V +- 0.18
s

#define VBATT		PA4
now detecting SELF_HOLD pin,
bridge the onOff button
press ENTER to begin or 's' for next stage.

992
0 now C14 : 991
0 now C15 : 991
0 now F0 : 991
0 now F1 : 991
0 now A0 : 991
0 now B11 : 991
0 now A5 : 989
0 now A6 : 991
0 now A7 : 991
0 now B0 : 991
0 now B1 : 991
0 now B2 : 991
0 now B10 : 991
0 now A1 : 991
0 now B12 : 991
0 now A11 : 991
0 now F6 : 991
0 now F7 : 991
0 now A12 : 991
0 now A15 : 991
0 now B3 : 991
0 now B4 : 991
0 now B5 : 991
0 now B6 : 991
0 now B7 : 991
0 now B8 : 991
0 now B9 : 991
release OnOff button
press ENTER to begin or 's' for next stage.

-991 = 1
1 now C14 : -991 = 0
1 now C15 : -991 = 0
1 now F0 : -991 = 0
1 now F1 : -991 = 0
1 now A0 : -991 = 0
1 now B11 : -991 = 0
1 now A5 : -990 = -1
1 now A6 : -991 = 0
1 now A7 : -991 = 0
1 now B0 : -991 = 0
1 now B1 : -991 = 0
1 now B2 : -991 = 0
1 now B10 : -991 = 0
1 now A1 : -991 = 0
1 now B12 : -991 = 0
1 now A11 : -991 = 0
1 now F6 : -991 = 0
1 now F7 : -991 = 0
1 now A12 : -991 = 0
1 now A15 : -991 = 0
1 now B3 : -991 = 0
1 now B4 : -991 = 0
1 now B5 : -991 = 0
1 now B6 : -991 = 0
1 now B7 : -991 = 0
1 now B8 : -991 = 0
1 now B9 : -989 = 2
SELF_HOLD PB9 //2
bridge OnOff button
press ENTER to begin or 's' for next stage.

991
0 now C14 : 991
0 now C15 : 991
0 now F0 : 991
0 now F1 : 991
0 now A0 : 991
0 now B11 : 991
0 now A5 : 991
0 now A6 : 991
0 now A7 : 991
0 now B0 : 991
0 now B1 : 991
0 now B2 : 991
0 now B10 : 991
0 now A1 : 991
0 now B12 : 990
0 now A11 : 991
0 now F6 : 991
0 now F7 : 991
0 now A12 : 991
0 now A15 : 991
0 now B3 : 991
0 now B4 : 991
0 now B5 : 991
0 now B6 : 991
0 now B7 : 991
0 now B8 : 991
0 now B9 : 991
release OnOff button
press ENTER to begin or 's' for next stage.

-992 = -1
1 now C14 : -991 = 0
1 now C15 : -991 = 0
1 now F0 : -991 = 0
1 now F1 : -991 = 0
1 now A0 : -991 = 0
1 now B11 : -990 = 1
1 now A5 : -991 = 0
1 now A6 : -991 = 0
1 now A7 : -991 = 0
1 now B0 : -991 = 0
1 now B1 : -989 = 2
1 now B2 : -991 = 0
1 now B10 : -991 = 0
1 now A1 : -991 = 0
1 now B12 : -991 = -1
1 now A11 : -991 = 0
1 now F6 : -991 = 0
1 now F7 : -991 = 0
1 now A12 : -991 = 0
1 now A15 : -991 = 0
1 now B3 : -989 = 2
1 now B4 : -991 = 0
1 now B5 : -991 = 0
1 now B6 : -991 = 0
1 now B7 : -991 = 0
1 now B8 : -991 = 0
1 now B9 : -991 = 0
SELF_HOLD PB1 //2
bridge OnOff button
press ENTER to begin or 's' for next stage.

991
0 now C14 : 991
0 now C15 : 991
0 now F0 : 991
0 now F1 : 991
0 now A0 : 991
0 now B11 : 989
0 now A5 : 991
0 now A6 : 991
0 now A7 : 991
0 now B0 : 991
0 now B1 : 991
0 now B2 : 991
0 now B10 : 991
0 now A1 : 989
0 now B12 : 991
0 now A11 : 991
0 now F6 : 989
0 now F7 : 991
0 now A12 : 991
0 now A15 : 991
0 now B3 : 991
0 now B4 : 991
0 now B5 : 991
0 now B6 : 991
0 now B7 : 991
0 now B8 : 990
0 now B9 : 991
release OnOff button
press ENTER to begin or 's' for next stage.
s

#define SELF_HOLD		PB1
now detecting BUTTON pin,
release OnOff button
press ENTER to begin or 's' for next stage.

//when i release the OnOff button , the power is off ,and when i re bright it , it came to beginning.

'v'=battery voltage
'x'=delete pin	'l'=list,	'c'=reset and 's'=next stage
ENTER for next pin
now A0
press ENTER to begin or 's' for next stage.

:-)
:-)
:-)
:-)
:-)
:-)
:-)
```

RoboDurden, 2024-05-02:
Okay so the change from OUTPUT to INPUT_PULLUP prevents the stage to crash.
However my too-clever trick to detect the hold pin does not seem to work.

For now i suggest that you keep the onOff button bridged and skip these stages with 's' and complete all the other steps.
It is good to have the three led as they can help to output debug info..
Be sure your motor is strongly mounted before you enter the current pin detection

A4 is you battery voltage ?

I already have uploaded test binaries for your 2.1.6.1 board. 
With no hold and button pin, these should already work for the slave board, that needs external power anyway.

RoboDurden, 2024-05-02:
You can also update your Gen2_GD32 fork and update defines_2-1-6.h yourself (if you have Keil installed)

RoboDurden, 2024-05-02:
@I-hate-2FA maybe you have some eeprom code for GD32 so i could implement your hold-pin detection..

RoboDurden, 2024-05-02:
@hiliving would be nice if you could upload some nice clean photos of frontside and backside of master and of slave.

I-hate-2FA, 2024-05-02:
using input pullup or output high wont effect what method of detecting, the only bad side effect it may have is that the signal is too weak it can no longer self hold(which does happen, on a mm32 board, but it is very rare), that way you need to trace manually no way around it

it is not needed to use eeprom to detect self hold, just print the pin you are going to release, and wait 200ms if the code is still running use backspace to delete that and print new pin untill the board is off, but this will have to be done last ofcourse

I-hate-2FA, 2024-05-02:
can i ask, what is the minimum necessary pin configuration in order to make the board boot normally in gd32
like if buzzer is not set board can still drive motor normally just no boot sound
in mm32 only 5 pin is verified on boot, if its not valid the firmware will not boot up (3 hall pin and 2 serial pin), all other is optional

I-hate-2FA, 2024-05-02:
btw your autodetect also will cause my 2.6.1 board to reset, with power button bridged, when detecting led, so the side effect of using pinmode output is already known back then, but you just removed the problematic pin from the list and call it a day

RoboDurden, 2024-05-02:
i have added #ifdef for more pins like ORANGE and HOLD to get it compiled.
I still have to insert #ifdef for red and green.

I do not remember to be so ignorant as to no woreship your contributions.

I have a 2.1.4 setup running now and it can also not detect the hold pin.
Yes, the led stage still uses OUTPUT instead of INPUT_PULLUP.

Today is the last sunny day here for at least a week.
I will continue tomorrow with this issue here.

hiliving, 2024-05-02:
> @hiliving would be nice if you could upload some nice clean photos of frontside and backside of master and of slave.

> @hiliving would be nice if you could upload some nice clean photos of frontside and backside of master and of slave.

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/80#issue-2273719019  done it,  and very thanks

I-hate-2FA, 2024-05-02:
> I do not remember to be so ignorant as to no woreship your contributions.

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/53#issuecomment-1900691584

hiliving, 2024-05-02:
> Okay so the change from OUTPUT to INPUT_PULLUP prevents the stage to crash. However my too-clever trick to detect the hold pin does not seem to work.
>
> For now i suggest that you keep the onOff [...]

yeah PA4 is my voltage for 24V。i am using the 24V LED power.    thanks for your compile, ill test it now

AILIFE4798, 2024-05-02:
The LEDs should not really effect the code, even if it's not set it'll just not be outputted, but for vbat and current DC if it's not set will the board still work normally, and the button too, the code may get stuck waiting for button to release

I-hate-2FA, 2024-05-02:
The LEDs should not really effect the code, even if it's not set it'll just not be outputted, but for vbat and current DC if it's not set will the board still work normally, and the button too, the code may get stuck waiting for button to release

RoboDurden, 2024-05-02:
code will work without vbat and currentDC:

bldc.c

```
float batteryVoltage = BAT_CELLS * 3.6;
float currentDC = 0.42;		// to see in serial log that pin is not defined

...

	#ifdef VBATT
		if (buzzerTimer % 100 == 0)
			batteryVoltage = batteryVoltage * 0.999 + ((float)adc_buffer.v_batt * ADC_BATTERY_VOLT) * 0.001;
	#endif

..

	#ifdef CURRENT_DC
		currentDC = ABS((adc_buffer.current_dc - offsetdc) * MOTOR_AMP_CONV_DC_AMP);
	#endif
```

button can be disabled in config.h

`	//#define CHECK_BUTTON		// disable = add '//' if you use a slave board as master
`

But yes, i shoud add

```
#ifndef BUTTON
  #undef CHECK_BUTTON 
#endif
```

Please @I-hate-2FA remeber that i did not write this firmware from scratch. And i have no happiness to fully recode it. So there will always be old code lingering around.

I-hate-2FA, 2024-05-02:
At least this is a good start to turn the unreadable gen1 mess into something workable given enough time, great work none the less, but it's why I decide to redo everything with mm and not merge into this

hiliving, 2024-05-02:
i test the 2.1.6.1  firmware with  dummy.bin  and single_uart.bin and single_uartbus_0.bin , unfortunately it can`t work.
well I found that the Hall pins in 2.1.7 are the same as those in 2.1.6.1, but in a different order。so I changed the Hall pin of 2.1.7 to be the same as 2.1.6.1.  then, it works normally and can be controlled through serial, but the button pins are incorrect and the power cannot be turned off. Only can be turned on.

RoboDurden, 2024-05-02:
Good to see that you can solve problems yourself :-)
please post your final pin definitions so i can update 2.1.6

RoboDurden, 2024-05-02:
[image]

I-hate-2FA, 2024-05-03:
Looks like a master to me with charger and a buck converter
Self hold circuit is not clear though

RoboDurden, 2024-05-03:
I posted my image of the 2.1.6(.0) slave because it shows the label v1.1.
The board of @hiliving might be the older v1.0

Here my master board, 2.1.6.0 master:
[image]

I-hate-2FA, 2024-05-03:
the one hiliving have is v1.3

RoboDurden, 2024-05-03:
I have added a menu to my autodetect:

```
autodetect menu:
0: do all
1: VBatt
2: Hold
3: Button
4: Led
5: Hall
6: HallOrder
7: CurrentDc
8: Results
```
6,7, and 8 only show up when the preceeding stage has been completed.

for Hold i added a manual mode:

```
now detecting SELF_HOLD pin,
bridge the onOff button
't' to toggle manual mode.
ENTER to begin, 's' for next stage, 'q' for menu
```

with `auto find: 0` the user has to click through every pin with ENTER. 
When the onOff button is released, the board will turn off and the user can see which pin it was.
Then after restart and entering manual mode again, the user has to cycle through the pins again with onOff bridged and send 'f' when the correct pin shows up again.
I would prefer to store the value in eeprom but i do not know how to do this.
And i am again very close to the 32 kB limit: Program Size: Code=30448 RO-data=1564 RW-data=564 ZI-data=2996  

I can not find a button (onOff) pin with my autodetect on Gen2.1.4:

```
now detecting BUTTON pin,
release OnOff button
ENTER to begin, 's' for next stage, 'q' for menu
 C14 C15 F0 F1 A0 B11 A4 A5 A6 A7 B0 B1 B2 B10 A1 B12 A11 F6 F7 A12 A15 B3 B4 B5 B8 B9 A2
bridge OnOff button
ENTER to begin, 's' for next stage, 'q' for menu
 C14 C15 F0 F1 A0 B11 A4 A5 A6 A7 B0 B1 B2 B10 A1 B12 A11 F6 F7 A12 A15 B3 B4 B5 B8 B9 A2
0 pins found. release OnOff button
ENTER to begin, 's' for next stage, 'q' for menu
```

Maybe this 2.1.4 does not have a onOff button detection ?
Board simply holds the power on once it has powered on..

currentDC is also not working. With this 2.1.4 Unable to detect backcurrents.
I am not sure about VBatt either :-/
All the adc does not seem to work.

@hiliving , i have uploaded a `hoverboard autodetect D GD32F130 tx=PA2, rx=PA3.bin` with this new autodetect. Would be nice if you give it a try.

I-hate-2FA, 2024-05-03:
2.1.4.1 have latch and on off button and i know which pin it was

to use eeprom,you need 2kb of free storage(last 1kb cannot be used or the chip may not be able to boot,and you need a full page allocated to be eeprom)

its nice seeing were improving together and leave the gen1 behind,today got f103 blinking,although it will be a long and hard way because the f103 is M3 core,mm32spin is M0 core, there is allot of compatibility issue, and ofcourse the clock on f103 is also problematic, just like the mm32spin

I-hate-2FA, 2024-05-03:
you are 99.4% full on storage idk why is that the case when there isnt allot of strings like in my firmware

I-hate-2FA, 2024-05-03:
so just use the terminal as eeprom,do you know you can read data back from terminal that you just printed?

RoboDurden, 2024-05-03:
With `#ifndef REMOTE_AUTODETECT` i really tried hard to remove all unnecessary code.
Maybe i will give the MicroLib a try again: Program Size: Code=26336 RO-data=1388 RW-data=564 ZI-data=1876  

> do you know you can read data back from terminal that you just printed?

I have never heard of that. But i could program my autodetect.ino to store and send back a byte with some special characters like i bridge your serial detection..
This however would then only work with my autodetect.ino

I-hate-2FA, 2024-05-03:
because you didnt start a new project for autodetect i cant really read it unfortunately
it may also help with reducing unnecessary code

I-hate-2FA, 2024-05-03:
i have reverse engineered the circuit button is active high so setting to pull up will not be able to detect anything

I-hate-2FA, 2024-05-03:
i think we have already looked at the button and latch circuit before in mm

RoboDurden, 2024-05-03:
As i alreay had a `iRepeat` counter it was easy to implement 4 cycles:

```
now detecting BUTTON pin,
release OnOff button
ENTER to begin, 's' for next stage, 'q' for menu
 C14 C15 F0 F1 A0 B11 A4 A5 A6 A7 B0 B1 B2 B10 A1 B12 A11 F6 F7 A12 A15 B3 B4 B5 B8 B9 A2
C13 C14 C15 F0 F1 A0 B11 A4 A5 A6 A7 B0 B1 B2 B10 A1 B12 A11 F6 F7 A12 A15 B3 B4 B5 B8 B9 A2
bridge OnOff button
ENTER to begin, 's' for next stage, 'q' for menu
 C14 C15 F0 F1 A0 B11 A4 A5 A6 A7 B0 B1 B2 B10 A1 B12 A11 F6 F7 A12 A15 B3 B4 B5 B8 B9 A2
C13 C14 C15 F0 F1 A0 B11 A4 A5 A6 A7 B0 B1 B2 B10 A1 B12 A11 F6 F7 A12 A15 B3 B4 B5 B8 B9 A2
BUTTON	A4
1 pins found. release OnOff button
```

```
void ScanInit(uint8_t iTestNew)
{
...
	pinModePull(aoPin[i].i,GPIO_MODE_INPUT,iRepeat%2 ? GPIO_PUPD_PULLUP : GPIO_PUPD_PULLDOWN);
```

@hiliving  I have updated all the autodetect binaries.

I-hate-2FA, 2024-05-03:
> Well for the 2.1.4 board, it has to be `GPIO_PUPD_PULLUP` Otherwise `uint8_t bOn = digitalRead(aoPin[iTest].i);` will not show a difference.

if this is true, then it dont make any sense to detect it anyways just treat it as without a button
because in main the logic is designed for normally low button, so it get stuck in the while imidiately

RoboDurden, 2024-05-03:
Good point, i already wondered why the button push did not turn off the 2.1.4 board.
It will need a #define BUTTON_PushedOnLow in those defines_2-x-y,h that need PULLUP instead of PULLDOWN.
Will test that tomorrow.

I-hate-2FA, 2024-05-03:
Arguably it's better to not set the self hold and button anyways
Because you can just connect a relay/opto coupler to the button pin and control the board with esp
That way you can ensure the boards power state is synced on the esp, and you can turn it on and off as you will

I-hate-2FA, 2024-05-03:
On a normal board with self hold circuit we discussed before, no additional hardware is needed even, you can just connect one of the pin of the switch that didn't go to vbat to one of the gpio on esp, when gpio have 3v3 board will turn on

hiliving, 2024-05-06:
Sorry for the late reply. I was outside a few days ago，its great  of you continuously improving autodetect. The UI has become more user-friendly,I tested the latest autodetect.bin . However, when testing the button pin, it still doesn't seem feasible on this board. according to the prompt, press release onoff, and it will shut down. After restarting again, it won't be able to continue the previous detection and return to the initial interface.
```
autodetect menu:
0: do all
1: VBatt
2: Hold
3: Button
4: Led
5: Hall
3
now detecting BUTTON pin,
release OnOff button
ENTER to begin, 's' for next stage, 'q' for menu
hit ENTER:
:-)

:-)
```

With the help of AiLiFE, I manually traced and confirmed that the Button pin is PA5, which is the same as 2.1.7. However, the actual compiled firmware cannot be shut down by clicking the button. I don't know why, but clicking the button in Autodetect can shut down normally

[27 more replies]