RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #81: Some boards need INPUT_PULLUP for onOff button to work !!!

RoboDurden, 2024-05-04:
Board Gen2.1.4 needs in defines_2-1-4.h

```
//#define BUTTON		P??
#define BUTTON_PU		PA4
```

(a boolean #define BUTTON_NEEDS_PULLUP would be more intuitive but harder to implement a different data type than PA4 to my autodetect.)

and in setup.c

```
			#ifdef BUTTON_PU
				pinModePull(BUTTON_PU,GPIO_MODE_INPUT,GPIO_PUPD_PULLUP);
			#else
				pinMode(BUTTON,	GPIO_MODE_INPUT);
			#endif
```

When i add an X like `#ifdef BUTTON_PUX` the button no longer works because `digitalRead(BUTTON)` always returns `0`
Strangely, with `GPIO_PUPD_PULLUP` setup `digitalRead(BUTTON)` returns 1 when pressed and 0 when released.

Of course i thought that with pullup, the buton should return 0 when pressed and 1 when not being pushed. But no, the digitalRead behavior does not change. So my defines.h reads like

```
#ifdef BUTTON
	#define BUTTON_PUSHED 1
#else
	#ifdef BUTTON_PU
		#define BUTTON_PUSHED 1		// very strangely, even so the button needs a pullup, digitalRead gives 1 when button pushed
		#define BUTTON BUTTON_PU
	#else
		#undef CHECK_BUTTON 
	#endif
#endif
```

`BUTTON_PUSHED` should be defined to be 1 for both cases. 

The onOff button of Gen2.1.4 must be inverted on the pcb to become a breaker that interrupts the connection to gnd. So when pushed, the internal pullup pulls the pin to high. Otherwise it is pulled to 0.
Normaly, the pin is pulled to gnd by an external resistor and becomes 1 when the button is pushed and VBatt via a resistor divider lifts the input voltage to somewhere near 3V.

@I-hate-2FA commented in https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/80#issuecomment-2093680987
> Arguably it's better to not set the self hold and button anyways Because you can just connect a relay/opto coupler to the button pin and control the board with esp That way you can ensure the boards power state is synced on the esp, and you can turn it on and off as you will

> On a normal board with self hold circuit we discussed before, no additional hardware is needed even, you can just connect one of the pin of the switch that didn't go to vbat to one of the gpio on esp, when gpio have 3v3 board will turn on

With my solar camper and my block band saw i already have two real life projects where the ESP32 does more than controlling one hoverboard and i can turn the hoverboard on and off with the hoverboard button.
And all the BDC with their SFP (stupid fun projects) will want the on/off button.

I-hate-2FA, 2024-05-04:
add `const uint8_t BUTTON_PUSHED =0;`
```
if(digitalRead(BUTTONPIN)==BUTTON_PUSHED){
//do sth
}
```
```
if(BUTTON_PUSHED == 1){
pinMode(BUTTONPIN,INPUT);
}else{
pinMode(BUTTONPIN,INPUT_PULLUP);
}
```

the use of internal self hold would only make sense in one circumstances,that is the esp is powered by the hoverboard 3.3v, and you only used 1 or 2 boards,but the problem is the hoverboard 3.3v is not powerful enough for a esp, you need to use a lm2596 connected after the PMOS to make the esp switched by the hoverboard while also providing enough power
which is not really the most convinient way

RoboDurden, 2024-05-04:
As I wrote, both cases have button pushed as 1 and button releases as 0!

We have already one Remote das does not need an esp32. I would like it if users would add a RemoteAdc.

I-hate-2FA, 2024-05-04:
when converting old appliances to smart ones,the toggle button is the most hated things ever,you cant just simply use a smart socket,insted you need to somehow use a relay to push the button, and have a gpio to read the power led on the appliances, so it can actually tell it is on or off and some slightly complicated tricks are also involved because the button sometimes is multiplexed

I-hate-2FA, 2024-05-04:
> As I wrote, both cases have button pushed as 1 and button releases as 0!

thats very weird,would be nice if you can provide clear picture of front and back with capacitor and plastic connector removed
> We already have one Remote that does not need an esp32. I would like it if users would add a RemoteAdc.

thats exactly what we dont want, or rather i exactly dont want

I-hate-2FA, 2024-05-04:
it would be super easy to add remote adc if there is 2kb of storage available, but we dont want that, we want standarized uartbus protocol on all boards

RoboDurden, 2024-05-04:
No, only Linux believers want only one standard and everyone to follow their leaders ;-)

With my Remote.h I emulated the object oriented style so I would be happy if people would adopt it.

I-hate-2FA, 2024-05-04:
i originally want to port uartbus to a gen1(with 2 slaveid on 1 board) but after working with them once on the gd32f103 board i gave up all hope on gen1 we should just move on

allot of people indeed requested adc without a esp, which would be doable, but why, there is only 32kb after all, the implementation will be the most primitive

RoboDurden, 2024-05-04:
Without autodetect and the MicroLib:

Program Size: Code=18542 RO-data=1158 RW-data=24 ZI-data=1528  

I would simply add the two additional adc to

```
typedef struct
{
  uint16_t v_batt;
	uint16_t current_dc;
} adc_buf_t;
```

The adc finished intterupt will probably still support the 16 kHz of the TIMER0
```
//----------------------------------------------------------------------------
// This function handles DMA_Channel0_IRQHandler interrupt
// Is called, when the ADC scan sequence is finished
// -> ADC is triggered from timer0-update-interrupt -> every 31,25us
//----------------------------------------------------------------------------
void DMA_Channel0_IRQHandler(void)
{
```

But i want some of the BDC do the job. connecting two potentiometers is very likely a SFP after all.

RoboDurden, 2024-05-04:
> thats very weird,would be nice if you can provide clear picture of front and back with capacitor and plastic connector removed

Sorry i will not touch a running board. So no removal of components.