RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #82: Arduino example 4Wheeler - misleading comment ( or my misunderstand ? )

pacraf, 2024-05-04:
I have a test setup consisting of 4 boards, each of them separate  ( not MASTER-SLAVE so lets call is 4 SINGLES ) connected on common UARTBUS

then to test it, you can use example 
Arduino Examples/Test_4Wheeler/Test_4Wheeler.ino

and there is lines:
#define REMOTE_UARTBUS 
//#define SEND2SINGLES  // uncomment this line if you want to send to two master that then will control their slaves

and I think that it is exactly opposite as this comment says , right? 
my setup works only when I uncomment this line...  
if you look into this .ino , it is obvious that drives 2,3 will never receive data.
disclaimer: I did so many stupid mistakes in the past, so honestly I am not sure if it is like I wrote above...

RoboDurden, 2024-05-04:
I will check later.
Honestly I do not really want to offer ready to use Arduino examples.

I only focus on test speed.ino which gets updated with uartBus changes..

These other .ino were only created once when a specific user needed it..

It always better to start with test speed and modify this to work for your project.

pacraf, 2024-05-05:
Dear   @RoboDurden, your firmware for 2 x 2.1.8 and 2 x 2.1.1 connected on common UARBUS works like a charm. 
examples for arduino are both very helpfull, so please keep them both...

KINDLY THANK YOU FOR THIS AWESOME WORK.

closed 2024-05-10 by pacraf