RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #83: Very very elementary guide for the newbie of imbedded programming

jeongsukchul, 2024-05-18:
Hi, I have to hack my hoverboard because of my projects in the mechanical class.
Thanks for your github, i think i can find a solution for the project. 
I'm using  GD32F130C8 and the board is same with the board in 
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/25
except for some small changes like ( the master and slave board are same.. or etc, the overall pinmap is same i think)

My goal is just 
1. sensing the tilt angle from imu 
2. move the single motor with normal motor control
3. move the single motor with FOC control and check difference between them.
4. implement normal Hoverboard code using FOC to control both master and slave.

Because I am just mechanical engineering student, and i didn't even study about embedded programming, only one that i can do is just use arduino..
So it was quiet difficult to understand what you are saying. So I want you to answer my some little stupid questions that can help me doing things in right direction.

I buy esp8266 and ST-LINK V2, i only finished just this,
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/How-to-Unlock-MCU-flash

Q1. Why I have to use UART communication, not just use SWD flashing and control the GD32F130C8?

Q2. What is the pin connection for esp8266 to use UART?

Q3. How to execute bin files?
You mension to test the board with binary files, but how? using st-link utility?
hoverboard 2.1.8 master Dummy.bin
hoverboard 2.1.8 master Uart.bin
hoverboard 2.1.8 single UartBus id0.bin
hoverboard 2.1.8 single UartBus id1.bin
hoverboard 2.1.8 single UartBus id2.bin 
hoverboard 2.1.8 single UartBus id3.bin
then, can you tell me what is the purpose of each 6 files?

Q4. To success my goals (1,2,3,4) what files that i have to execute?
I saw that you recommend to use keil.
So i open the keil file. And what is the next thing i have to do? I saw your videos in youtube, in there you modified the config.h, What is things that i have to modify to success the each goal 1,2,3,4?
[image]

It will be great if you explain the things!

I-hate-2FA, 2024-05-18:
A1:you can flash with swd, but to control the speed you need to use serial to communicate esp with motor driver
A2:for esp8266 only hardware uart is io3 and io1
A3:you can use stlinkutil to flash gd32f series, bt it is better t use pyocd, please read the wiki first!!!
A4:for all your goal, only 2 is acheiveable
     if you can settle for sine wave insted of FOC you can try my firmware insted!

RoboDurden, 2024-05-18:
I guess this is the IMU:
[image]

datasheet needed.

For esp8266 i would try SoftwareSerial first so that the one full hardware serial can be used for debug output/input

As you have Keil installed, please compile (F7) and flash (F8) yourself. I do not really know if all these dedicated binaries are still up to date. @I-hate-2FA , if you know how to add eeprom, it would be nice if you do a proof-of-concept eeprom pull request to the GD32 repo. I would like to get rid of all these dedicated binaries by storing the many defines in one main binary. With the option add boards with a second autodetect binary like you have done with MM32.
Your sim_eeprom.c is built on hal_flash.c and i do not see such a HAL in the GD32 firmware.

@jeongsukchul if you need to compare FOC to Block Commutation with a Gen2.1 (GD32F130) board you might want to have a look at my SimpleFOC firmware. 
[image]
The control via I2C did already work. But i more or less have abondend this project because the SimpleFOC library is too complex in my opinion.. And i have made it even more complex with my OOP obession :-(

I guess you better stay here with the simple Keil project.

RoboDurden, 2024-05-18:
If your board is not exactly the 2.1.8 layout, please post photos of front (and backside) here and do only test with a 2A CC power supply.

I-hate-2FA, 2024-05-18:
F103 series is too hard to support I've deemed it impossible without seperate project
Because f103 is core m3, it do not have pinmodeAF, have different gpio, it cannot support this structure
Gd32f130 on the other hand advertise as a core m3, but it is actually core m0, it is modified from stm32f031, you can look at the CPU structure, the gpio of course m0 is connected to AHB bus and m3 it is connected to apb2
I have a idea that can maybe make the firmware support gd32f130 insted, do you want to merge gd into mine insted, if not it's ok I'll work on lks and mm32f031 insted

RoboDurden, 2024-05-18:
@I-hate-2FA , i do not have the happiness anymore for big tasks like porting the firmware. Only little things like adding eeprom if someone gave me a code example :-/

And in your bldc.c is see 

```
#include "hal_tim.h"
#include "hal_conf.h"
```

So i fear that your motor control is not compatible with my gd32 bldc.c

I-hate-2FA, 2024-05-18:
i already have idea how can i support mcu that do not have hal library