RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #84: New board SMART-L-V2.1, two controller boards equipped with STM32F103C8T6)

zimmii91, 2024-05-20:
i have found 3 sets of these boards, but unfortunately no firmware as far as i understand it. found this issue a while ago, but still no sollution. https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/218

[image]

RoboDurden, 2024-05-20:
My firmwares can be compiled for the GD32F103 and i think that this mcu is compatible to the STM32F103. But noone ever tested my F103 binaries :-(
I have 2.2.1 boards here and a test setup ready for such a board to install. But no happiness to do it.
I will assign 2.2.3 to your layout. Please upload photos from backside and master / slave

zimmii91, 2024-05-20:
Sounds great! here are some pictures. i can gladly try, however i am a bit unsure of the ST-link connections. i have seen in the link i provided that the pins where soldered, and i have done the same, but i want to be sure before connecting. do you have any good suggestion to fint this out? 

[image]
[image]
[image]
[image]
[image]

RoboDurden, 2024-05-20:
check with a multimeter which of the 4 pins is GND and which is 3.3V. Then there are only two options for SWCLK and SWDIO

I-hate-2FA, 2024-05-20:
> My firmwares can be compiled for the GD32F103 and i think that this mcu is compatible to the STM32F103. But noone ever tested my F103 binaries :-( I have 2.2.1 boards here and a test setup ready for [...]

no your firmware does not work on f103 i have tested it, and it do not need a test to know

RoboDurden, 2024-05-20:
And you feel no need to explain :-(

I-hate-2FA, 2024-05-20:
[image]
target is f103, processor f130 is selected, i dont think any explaination is needed

I-hate-2FA, 2024-05-20:
back in #77 you already asked me to not use your firmware, theres no reason gen2 cannot work on gen1, additional motor will just be ignored, and itll work without issue if the correct set of hall sensor is detected

RoboDurden, 2024-05-20:
Ah thanks for the info. I may have corrected that:
[image]

Now i get 44 errors to be fixed :-)
Starting with 

```
compiling bldc.c...
Src/setup.c(37): error: call to undeclared function 'gpio_mode_set'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
        gpio_mode_set(pin&0xffffff00U, mode, GPIO_PUPD_NONE,BIT(pin&0xfU) );
        ^
Src/setup.c(37): note: did you mean 'gpio_bit_set'?
C:/Users/PAN CF-LX6/AppData/Local/Arm/Packs/GigaDevice/GD32F10x_DFP/2.0.3/Device/Firmware/Peripherals/inc\gd32f10x_gpio.h(460): note: 'gpio_bit_set' declared here
void gpio_bit_set(uint32_t gpio_periph, uint32_t pin);
     ^
Src/setup.c(37): error: use of undeclared identifier 'GPIO_PUPD_NONE'
        gpio_mode_set(pin&0xffffff00U, mode, GPIO_PUPD_NONE,BIT(pin&0xfU) );
                                             ^
Src/setup.c(38): error: call to undeclared function 'gpio_output_options_set'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
        gpio_output_options_set(pin&0xffffff00U, GPIO_OTYPE_PP, GPIO_OSPEED_10MHZ, BIT(pin&0xfU));
        ^                                           ^
```

Maybe i can fix it.

Yes, EferuFOC should work on these Gen2.2 boards. If the pin get traced manually.
I would prefer my autodetect :-)

I-hate-2FA, 2024-05-20:
> Now i get 44 errors to be fixed :-) Starting with

good luck, i have already gave up, supporting gd32f130 on my firmware is easier then supporting mm32f103 beause it is core m3
> Yes, EferuFOC should work on these Gen2.2 boards. If the pin get traced manually. I would prefer my autodetect :-)

#77 all pins is traced manually without error, but i still does not work, gen1 code is a nightmare to work with, anyone should not attempt to support none standard hardware, it is a complete waste of time and effort, and terrible customer support in the issue, working with you is much better

RoboDurden, 2024-05-20:
If it is a 64kB mcu, my SimpleFOC should run. I tested it with a Gen1 board.

I-hate-2FA, 2024-05-20:
it is 128 or 256k mcu
i am not familiar with the simplefoc version, but it is written by oop fan it should be understandable unlike the gen1

I-hate-2FA, 2024-05-20:
do not use CSDN website, it is complete trash, a android system developer copied code from there and caused a huge bug that involve infinite loop

core m3 do not have pinmodeAF, thats why i gave up

RoboDurden, 2024-05-20:
It might be better to start with https://github.com/Candas1/Split_Hoverboard_SimpleFOC
my fork adds aditional hoverboard classes which makes the OO structure more complex.. The Candas1 repo was the repo where we made the motor running..

I-hate-2FA, 2024-05-20:
> do not use CSDN website

https://youtu.be/kE9tUAPU-0c

RoboDurden, 2024-05-20:
searching for `GPIO_TIMER0_FULL_REMAP `  or `GPIO_TIMER0_PARTIAL_REMAP` only give me Chinese webseites :-/ GD is from China of course.
If i would find some other code, i might be able to copy and paste..

RoboDurden, 2024-05-20:
There is a gd32F103 Gen2 repo that seems to have done the port: https://github.com/JohMah/Hoverboard-JC2015.7.31.V2.0-Hack/blob/main/HoverBoardGD32F10x/Src/setup.c
But i can not yet get it compiled:
```

Build target 'Target 1'
Error instantiating RTE components
Error #544: Required Software Pack 'GigaDevice.GD32F1x0_DFP.3.1.0' is not installed
```

Good night from Germany.

RoboDurden, 2024-05-21:
Now this other repo comiles with

gd32f10x_misc.c : `NVIC->IP` -> `NVIC->IPR`
`    NVIC->IPR[nvic_irq] = (uint8_t)temp_priority;
`

main.c: `//#include "arm_math.h" ` - will this affect code speed ?

And a gd32f1x0_3.03 software pack had to be deselected in 
[image]

This Hoverboard-JC2015.7.31.V2.0 is Gen2.2.1 by the way, so i should be able to test it myself.
But there might be a difference between GD32F103 and STM32F103: 

> your bldc driver is different than gd32 board. its original stm32f103c8t6. the difference for gd32f103c have timer0=timer1 at stm32 etc. 
https://github.com/JohMah/Hoverboard-JC2015.7.31.V2.0-Hack/issues/3

I-hate-2FA, 2024-05-21:
> gpio_pin_remap_config timer0 leads me there but i can not speak Chinese..

i have read it anyways, but it do not explain complementary pwm, so it is not useful anyways
most people use stm32f103 library, because it is fully compatible with gd32f103(mm32f103 is not quite fully compatible)

> There is a gd32F103 Gen2 repo that seems to have done the port: https://github.com/JohMah/Hoverboard-JC2015.7.31.V2.0-Hack/blob/main/HoverBoardGD32F10x/Src/setup.c

the repo owner didnt invest time in the readme, i doubt it have nice code structure that allow you to change all the pins, so it would be just like gen1 very hard to modify, i would be happy if it wasnt the case but well im not holding my breath

I-hate-2FA, 2024-05-22:
i have checked, it is not as bad as i thought, the adc channel is editable in defines.h unlike gen1

RoboDurden, 2024-05-22:
According to the three issues, that firmware did actually work for some users. And take a look at the eagle/ folder!

My approach in Gen2.x is to provide wrapper defines/functions in target.h. This is sufficient to remove all errors from it.c

```
#elif defined GD32F103
	#include "gd32f10x.h"
...
	// it.c
	#define TIMER0_BRK_UP_TRG_COM_IRQHandler TIMER0_UP_IRQHandler
	#define adc_software_trigger_enable(a) adc_software_trigger_enable(ADC0,a)
	#define dma_interrupt_flag_get(a,b)	dma_interrupt_flag_get(DMA0,a,b)
	#define dma_interrupt_flag_clear(a,b)	dma_interrupt_flag_clear(DMA0,a,b)
```

But i guess it would be better to add a `setup_gd32f103.c` because of too many syntax changes. I stopped at 40% for ADC_init():

```
	#define RCU_ADC RCU_ADC0
	#define RCU_DMA RCU_DMA0
	#define RCU_ADCCK_APB2_DIV6 RCU_CKADC_CKAPB2_DIV6
	#define dma_deinit(a) dma_deinit(DMA0, a)
	#define ADC_RDATA ADC_RDATA(ADC0)
	#define dma_init(a,b) dma_init(DMA0,a,b)
	#define dma_circulation_enable(a) dma_circulation_enable(DMA0,a)
	#define dma_memory_to_memory_disable(a) dma_memory_to_memory_disable(DMA0,a)
	
	#define DMA_Channel0_IRQn DMA0_Channel0_IRQn
	#define TARGET_nvic_irq_enable(a, b, c){nvic_irq_enable(a, b, c);}
```

And 	`#define ADC_RDATA ADC_RDATA(ADC0)` is not working because you can not redefine a define..
So i need a new define `TARGET_ADC_RDATA` that either maps to `ADC_RDATA` or `ADC_RDATA(ADC0)`

But this guy has already made the full port, so i could test his code with my Gen2.2.1 and then integrate it in my Gen2.x

But i do not really see any future for my old block commutation motor control. 

It is a bit strange that you @I-hate-2FA Linux believer openly dislikes the EFeru firmware. You can find my rant here: https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/134 So i dream that you have begun to love the beauty of OOP.

I-hate-2FA, 2024-05-23:
> According to the three issues, that firmware did actually work for some users.

i never doubted if itll work on the repo owners board, but i doubted it is modifiable for other layout(i have checked, it is actually better then i thought, not as good as mine but it is acceptable)

> But i guess it would be better to add a `setup_gd32f103.c` because of too many syntax changes.

thats exactly what ill do , seperate initialize.c and hardware.c

> But i do not really see any future for my old block commutation motor control.

indeed, sine wave is really good, but square wave is better then nothing

> It is a bit strange that you @I-hate-2FA Linux believer openly dislikes the EFeru firmware.

i used to like gen1, but after actually working with it once i now understand it is terrible, and thats not the worst part, the maintainer is not helpful, it make me not want to use it

> So i dream that you have begun to love the beauty of OOP.

maybe i will, i have already learned allot from this project, before this i actually write the pin number in the code without using define

zimmii91, 2024-06-11:
Hi guys! can i do anything to help here? i was trying to connect the board with my ST-link as mentioned earlier, i found the 3.3v and GND using the multimeter, but the other 2 pins does not seem to work as expected. i do not get the program to recognize the chip. do you have any tip on how i may solve this?

would is help to run the autodetect firmware once i get connected properly?  @RoboDurden

RoboDurden, 2024-06-11:
Sorry i have no happiness to help people who go silent for three weeks :-(

As you can read here, there is another github repo that succeeded with stm32f103 Gen2 board.
When you have succeeded with that firmware with the pins of your board traced manually,
you can come back here to help.

I-hate-2FA, 2024-06-12:
for other 2 pin you can just try both combination
if it still doesnt work, it is not the swd pin, maybe rf remorte or serial
swd pin is reletively easy to trace manually

zimmii91, 2024-06-12:
Thank you for the suggestion, i do not have enough knowledge to follow everything you mentioned earlier, i only used the gen1 board which hade all this information available from different sources. 

when i look at the datasheet for this chip i find that PA13 is for SWDIO and PA14 for SWCLK, but i get confused of the different pinvariations mentioned that i assume is for variant models of the chip. however it seems that they should be placed in the top right corner, one on top right and one on 3rd pin down from right side. not sure what i should expect when i meassure the pins to be honest, was expecting to find a connection similar to how i was able to define GND and 3,3V, but unfortunately i am not able to. 

reading the datasheet i found this information, maybe this is helpful? 
"Serial wire JTAG debug port (SWJ-DP)
The ARM SWJ-DP Interface is embedded. and is a combined JTAG and serial wire debug
port that enables either a serial wire debug or a JTAG probe to be connected to the target.
The JTAG TMS and TCK pins are shared with SWDIO and SWCLK, respectively, and a
specific sequence on the TMS pin is used to switch between JTAG-DP and SW-DP. "

I-hate-2FA, 2024-06-12:
[image]
is it really that hard, just look at the wiki and the datasheet of the chip
remember, if it is gen1, theres no such customer support like im doing now!

zimmii91, 2024-06-14:
Hi again. For me, this was really a challenge.  i have no experience in compiling firmware or understanding the code, but i have given it some time now. however i have to admit defeat as i do not get anywhere. i really wanted to be able to contribute at this point in time. As i know it is not on you to teach these basics i can only say, sorry to have waisted your time guys.

I-hate-2FA, 2024-06-14:
sorry, working on the f103 is just a waste of time, because there is no easily adaptable firmware for this mcu, this repo is for gd32f130, i also have a repo for all mm32 series, which a brand new pair of board only cost $2, i dont understand why you would use other ones

zimmii91, 2024-06-16:
Oh that sounds great! i bought a couple of Hoverboards and have about 4 pairs of the same f103, it seems that is the only thing selling in Sweden. Tell me, where can i find the MM32 bords for $2? that would make it so much easier.

I-hate-2FA, 2024-06-17:
www,aliexpress,com/item/1005005959910034.html
you can only get it for $2 in china unfortunately, but i think $3.5 is still good

zimmii91, 2024-06-17:
Wow Thank you! This is really helpful! i was hoping to find something like this. Appreciate the help!

L1mp4, 2024-06-23:
Nice. Feels like a loss tho, does it not? Not being able to use perfectly good electronics lying on the table. Wish i was capable of solving this. I am not.

I-hate-2FA, 2024-06-24:
f103 is core m3 insted of m0(gd32f130 is also m3, but the structure is copied from stm32f030, which is m0), so it is impossible to adapt firmware for it, only way is to rewrite everything

I-hate-2FA, 2024-06-24:
you can always replace the mcu, but i think it is just not worth the effort, unless you need a board right now

L1mp4, 2024-06-24:
Nah. I ordered something else to get the motors working. Maybe someone with the skills will get hold of a lot of these boards in the future and make it happend. :-)

hiliving, 2025-05-08:
> My firmwares can be compiled for the GD32F103 and i think that this mcu is compatible to the STM32F103. But noone ever tested my F103 binaries :-( I have 2.2.1 boards here and a test setup ready for [...]

I have received a 2.2.1 board now, but I am unable to compile the firmware as there are too many errors. How can I resolve this issue, or do you have any pre compiled firmware for me to test？

[image]

RoboDurden, 2025-05-08:
Yes, i can confirm 12 errors :-( 12x
```
Src/../Inc/../Inc/target.h(50): error: 'stm32f10x.h' file not found
        #include "stm32f10x.h"
```

Sorry, i have no happiness to really get a 103 mcu working.
There is a EFeru port (103) for a gen2 board , and some other 103 gen2 fork...

I-hate-2FA, 2025-05-08:
f103 is m3 it is very difficult, althopugh f130 is also m3 it is more like m0 structure in terms of peripherals

RoboDurden, 2025-05-08:
This 

Src/../Inc/../Inc/target.h(50): error: 'stm32f10x.h' file not found
        #include "stm32f10x.h"

Error indicates that already the target is not properly setup for the Keil Ide.
I have no happiness for that.

I-hate-2FA, 2025-05-08:
When you solve that there's way more to come
It's just the file name incorrect you need to find the correct one

I-hate-2FA, 2025-05-08:
I think right click there's a option to include automatically but I haven't work on this for years I forgot

hiliving, 2025-05-08:
ok，I'm going to dismantle it，thx

WestlingPi, 2025-05-17:
I have come across a hoverboard ("andersson 3.1" with nice 10 inch airfilled wheels) that have splitboards that looks very similar to the photos in the original post of this issue. The board is labeled "SMART-L-V2.0" (i.e not V2.1) but the layout seems to be very similar (I havent compared them in detail yet) and it seem to have same STM32F103 mcu.

Anyone been able to get a firmware for similar board to work lately?

Reading this thread, the best option seems to be the https://github.com/JohMah/Hoverboard-JC2015.7.31.V2.0-Hack firmware. Anyone got it to work? I got it to compile with this https://github.com/blazepeak/Hoverboard_Utility_Gateway_System/blob/master/GigaDevice.GD32F1x0_DFP.3.1.0.pack (seems to be gone frome GigaDevice's official repo) and @RoboDurden 's patch with IP -> IPR above.

WestlingPi, 2025-05-17:
When trying to connect ST-Link V2 I get this from STM32 ST-LINK Utility:
Can not connect to target!
Please select "Connect Under Reset" mode from Target->Settings menu and try again.
If you're trying to connect to a low frequency application , please select a lower SWD Frequency mode from Target->Settings menu.

(and changing to "Connect Under Reset" doesnt help)

I think I have got the pins connected right from measuring with multimeter and comparing against the PDF in JohMah repo... Anyone seen this before?

RoboDurden, 2025-05-17:
Stlink utility usually doesn't work with gen2 boards.
Read the wiki here.
Install platformio or Keil.

WestlingPi, 2025-05-17:
Can I unlock with platformio or Keil? My hoverboard still have the original factory firmware. Or is unlocking not applicable to gen2 splitboards with STM32F103/GD32F103? Do I need J-Link or CMSIS‐DAP to unlock/download firmware?

No luck with download firmware with Keil uVision either (with ST-Link v2) :-(

Should I still shortcut powerbutton and power from 36V battery (like gen 1 instructions)?

RoboDurden, 2025-05-17:
Shortcut power button yes. But with new board for sure only connect 24-36V 2A constant current power supply.
Or use the 1.5A charger..

We here have had no success with stm32f103 boards.
Ask the one with the gen2 fork.
There is also a guy who ported the EFeru to single Bldc like gen2 for that Chinese scooter..

But yes, I guess you should unlock with st link utility like gen1 board.
Maybe you need the NRST pin... There are tutorials on this elsewhere...would be nice if you have success and I can finally add stm32f103 support :-)

I-hate-2FA, 2025-05-17:
no, gen1 instruction is totally wrong, you should not power from 36v, you should power mcu using the 3v3
what if during flashing the unstable state the 2 mosfet turn on at same time? its just total bs
you CAN use stlink utilities if you is f103, you just need the correct version as stated in wiki

WestlingPi, 2025-05-17:
No luck with unlocking with charger connected (and battery disconnected). The hoverboard just keeps beeping (3 long and 1 short beep) but not able to connect with ST-Link utilities. I start to wonder if there is something wrong with my soldering of the SWD pin header. Is it sensitive? My soldering skills are not that good...

@I-hate-2FA Are you sure about powering the board from ST-Link via the SWD 3v3 pin? I have not found any wiki that promotes that (and many claim you can brick the board by powering from ST-Link)...

WestlingPi, 2025-05-17:
Do you mean this gen2 wiki https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki or is there another one? I have read it several times but I guess I need to study it some more.
I will double check my soldering. Then I will try with stlink3.3v if still no luck with the gen 1 way.
I also have a cheap ST-Link USB dongle from aliexpress (it works with a bluepill, also from aliexpress).

I-hate-2FA, 2025-05-17:
yes, just use the stlink util version mentioned
you can open the stlink just by sliding the cover forward towards the usb, and check the chip, the first time i dont know and got one with fake(apm32), when i acidentaly destroyed it and the hoverboard i bought a real one with real stm32 directly from china

WestlingPi, 2025-05-18:
My stlink dongle have an unmarked chip. I guess it means it is a fake. Should I order a new from aliexpress and hope to get one with a real stm32?
The mcus on the hoverboard boards seem to be ST32F103 (they have ST logo, but I guess it could also be fake), i.e. not GD32F103. Can I still use GD packs in Keil uVision, or should I replace them with ST packs? Are ST and GD hex files compatible?

WestlingPi, 2025-05-18:
I did some testing with my bluepill (ST32F103). I compile the same project in Keil uVision with either STM32F1xx_DFP pack or GD32F10xDFP pack and both hex files works when downloaded to bluepill using ST-Link Utility. So it seems that the binaries are compatible (atleast with a simple blink firmware).
(However it doesnt work to download to bluepill from Keil uVision. When compiled with STM pack (and STM32 device) I get "Not a genuine ST Device! Abort connection". When compiled with GD pack (and DG device) I get some other error from Keil. However I can download to another Blackpill (STM32F401) from Keil (another blink project setup for STM32F401))

I-hate-2FA, 2025-05-18:
the stlink is supposed to have stm32f103c8t6, so you can just flash stlink firmware on the blue pill and let it be the stlink (ofcourse then you run into a problem you need another stlink to flash the stlink...but you can flash with serial i guess...)
the fake stlink can be flashed with daplink then itll work in keil and pyocd, but i dont know if it can unlock stm32, maybe try to unlock with pyocd, for mm32 i can unlock with keil by format the whole chip

WestlingPi, 2025-05-23:
I still have had no luck with connecting to stm32f103 on hoverboard with charger connected. I have tried numerous times with different stlink utility versions. I am about to give up and try connect 3v3 from stlink to 3v3 swd on hoverboard without charger/battery connected.
One thing I dont understand however is why fake or real stm32 mcu in stlink dongle would matter. As I understand, all colorful stlink dongles are fake. The original st-link/v2 from ST is white and the shape is different: https://www.st.com/en/development-tools/st-link-v2.html Would a fake (colorful) stlink dongle with a real stm32f103 mcu be better than a fake stlink with a fake mcu (apm32/unlabeled/whatever)? Or a bluepill with stlink firmware?

RoboDurden, 2025-05-23:
I never had problems with what you call fake dongles.
Read the wiki here !! There any ways to upload cose

RoboDurden, 2025-05-25:
This repo does not compile with stm32f103

Would be great if you get some other fork to compile with kein and test on the real hardware.
Then I could import that code to this gen2.x repo.

You could also start with platformio and some EFeru port. If you have the dual opamps on the board for phase currents

With platformio you could also test my old simpleFOC firmware.

But best would be that other Keil gen2 Firmware.

I-hate-2FA, 2025-05-25:
For swd I always just try both combination as it won't do damage and only need to try twice 
If you flash with 3v3 you do not need to connect the power button, in fact for gen 2 it's not recommend at all to short the power button because it will confuse the program for auto detect
And you can connect both the battery(obviously not the real battery but a constant current supply) and the stlini with 3v3 at the same time, they are all linear regulator, powering each other is ok it won't cause any problem, the power draw of both is insignificant 
Just everywhere you go do not use the eferu gen1 firmware, I just want to modify some Pinout for user on gen1 and it's already basically impossible, don't waste your time

RoboDurden, 2025-05-25:
You could also take a look at https://github.com/Kambodia91/Hoverboard-firmware-hack-FOC-MASTER-SLAVE
It is a platformio port of EFeru FOC firmware. So  @I-hate-2FA i guess new pin definitions can be done: https://github.com/Kambodia91/Hoverboard-firmware-hack-FOC-MASTER-SLAVE/blob/main/Inc/defines.h ?

But if you get JohMah's repo running, that is a Keil project and closest to this Gen2.x here :-)

I-hate-2FA, 2025-05-26:
I don't know, of he ported directly then there's still no define for adc and it'll not be suitable for other layout, if they did their job correctly with the adc channel configuration define then it's definitely possible I'll check when I have time

[147 more replies]