RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #88: Will I be able to use PinFinder firmware from here to flash my firmware into GD32F303?

LeonXBB, 2024-08-17:
Hi! Sorry if this is the wrong place to ask this question :)

I have my own custom firmware built on Emmanuel's original FOC version. It was originally intented to run on 103 chip just as it's supposed to be. However, during the tests one of the boards burned. Unfortunately, it seems like all hoverboards, eBikes, and electronics shops in my country have ran out of boards with 103 chips :( I could only find boards with 303 chips and some AT boards. This is an exhibition project, so I'm on a clock and don't have time to order from other countries. So I've ordered some boards with 303 chips. I know there already is firmware for AT boards, but it doesn't seem to include PWM which my project is build on.

After that I've started researching how I could port my firmware from F103 chip to F303. I was hoping it would be kind of easy, but after searching through wiki and issues of the original repository it seems to not be the case. I tried working with CubeMX, and, altough I understand the concepts, the practical realization seems to be rather above my level at the moment. So I've started googling in hopes of finding firmware for 303 chips. I didn't, but I found this project :)

My question is, if I connect my board with 303 chip when it arrives, and run the PinFinder procedure described in the wiki, would it work out of the box, or with some non-major adjustments?

Thank you in any case and sorry for the troubles!

I-hate-2FA, 2024-08-17:
if you cannot get board with f130 chips, you can consider the mm32, theres currently no support exist for f103 and f303

RoboDurden, 2024-08-17:
First time I now have heard of stm32f303 boards ?

My advice, search the second hand market for used hoverboards. The only way nowadays to get old stm32f103 or gd32F103 boards.

You might also want to watch my hoverboard repair videos on YouTube. If two mosfet burned and the other motor is still running after unsoldering them, there is a chance that two new mosfet will help.

Pinfinder firmware is only for MM32 boards, my autodetect firmware only for GD32.
And both only for single motor boards.

Adding pwm control to an existing firmware is not difficult .

LeonXBB, 2024-08-17:
[image]
[image]
[image]

This is what the seller sent me.

My repair would be slightly more complicated than replacing some transistors (there's a burnt hole on the board :) )
Alright, I'll look at the second hand market.

RoboDurden, 2024-08-17:
The older the used Hoverboard, the better

RoboDurden, 2024-08-17:
Looks like stm32F103 binaries run on gd32F303

https://www.eevblog.com/forum/microcontrollers/psa-gd32f303-is-not-a-drop-in-substitute-for-stm32f303/

LeonXBB, 2024-08-17:
@RoboDurden thank you so much for trying to help me :)
Yesterday I was looking through the original repository and found these two issues which might prove otherwise:

https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/191
https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/357

(That's how I found this version, by the way)

Anyway, I'm supposed to receive the boards no later than Monday so I will be able to test it.
I'll keep this issue open until then, if you don't mind.

RoboDurden, 2024-08-17:
No, the two links mostly only bother about different pin out. That has to be taken care of for sure.
I would start with the original Niklas Fauth (now Lucys Rausch) binary that has no foc and spins the motors forward and backwards without any user input

Make sure to only test binaries with a 2A constant current power supply.
I you have none then the 2A charger without a battery connected might work.

You can use my hoverboard online compiler to proceed from the test binary..

I-hate-2FA, 2024-08-17:
changing pinout on eferu is near impossible, unless you got too much time to waste just get a f130 or mm32

RoboDurden, 2024-08-17:
I think the lucys Rausch firmware is easier with the pin definitions.
And for the exhibition the foc is not absolutely necessary.

LeonXBB, 2024-08-17:
Thanks to all of you for advice :)
I'll get the boards on Monday and will let you know how it goes

I-hate-2FA, 2024-08-17:
> I think the lucys Rausch firmware is easier with the pin definitions. And for the exhibition the foc is not absolutely necessary.

not sure if adc is mandentory in that firmware
if not itll work
on my firmware it is not necessary

LeonXBB, 2024-08-19:
Got the boards. This is what they look like:

[image]

The chip is indeed GD32F303. 

So far I'm having troubles getting ST-Link Utility to detect it, even though my Google research shows that it's supported. Will keep this thread posted.

RoboDurden, 2024-08-19:
I guess that each pair of mofets is driven by one of these sot-8 half bridge drivers.
Find the datasheet, then you can easily check the 12 output pins of the MCU.

I guess the hardware of "Two 16-bit advanced-control timer (TIMER0 & TIMER7)" only has one option to drive two bldc motors.

This is the defines.h  you have to check for the simple and original Niklas Fauth firmware: https://github.com/lucysrausch/hoverboard-firmware-hack/blob/master/Inc/defines.h

There i see `#define LEFT_TIM TIM8` and `#define RIGHT_TIM TIM1`
But GD mostly starts counting from 0, so TIMER0 might be TIM1 and TIMER7 = TIM8

`PC6` also has `TIMER7_CH0` as alternate function.
So 

```
#define LEFT_TIM TIM8
#define LEFT_TIM_U CCR1
#define LEFT_TIM_UH_PIN GPIO_PIN_6
#define LEFT_TIM_UH_PORT GPIOC
```

PC6 , PC7 and PC8 for TIMER7_CHx
and PA7, PB0 and PB1 for TIMER7_CHx_ON are the only possible pins, so you do not really have to check the pins with a diode tester.

Simply flash this test binary and test with 2A CC power supply: 
https://github.com/lucysrausch/hoverboard-firmware-hack/blob/master/build/hover.hex

, so your board

I-hate-2FA, 2024-08-19:
try to use pyocd
but as i said its better to buy another f130 or mm32 supported board

RoboDurden, 2024-08-19:
It will help the hoverboard community more if he is able to get a Gen1 firmware running on his GD32F303 boards.

I-hate-2FA, 2024-08-19:
but the manufacture has already moved on to mm32 and at32, t iis waste of time to support gd32f303

RoboDurden, 2024-08-19:
Not if STM32F103 code runs without modifications on a GD32F303

LeonXBB, 2024-08-19:
I was able to connect the board to the ST-Link Utilty. I had to power it using said ST-Link and not via the battery as in original firmware. 
I'll get  back to work.

RoboDurden, 2024-08-19:
max 2A CC Power supply !!!

I-hate-2FA, 2024-08-19:
i really dont know why gen1 say to power the board from battery and not stlink, it doesnt make any sense, maybe its just eferu dont know what he is doing, if you are in gen2 community we will always tell you to upload with stlink 3.3v and test with current limited supply

LeonXBB, 2024-08-19:
Nothing so far, I'll let you know if anything happens

LeonXBB, 2024-08-19:
@RoboDurden What is the expected behaviour for when I flash the default hover firmware? It's currently connected only to power supply and nothing else.

Before flashing, it also used to beep when turning on and off. Not anymore. I understand, wrong pin for the buzzer in defines.h?

I-hate-2FA, 2024-08-19:
which firmware are you using

LeonXBB, 2024-08-19:
at the moment, I've flashed the hover.hex firmware provided by @RoboDurden above

I-hate-2FA, 2024-08-19:
ok, the f303 do not have any automatic pin definition detector like autodetect or pinfinder, so you need to trace all the pin manually first

RoboDurden, 2024-08-19:
As i already wrote, the 2x 6 mosfet output pins must be the same.
The 2x 3 hall pins could be different from

```
#define LEFT_HALL_U_PIN GPIO_PIN_5
#define LEFT_HALL_V_PIN GPIO_PIN_6
#define LEFT_HALL_W_PIN GPIO_PIN_7

#define LEFT_HALL_U_PORT GPIOB
#define LEFT_HALL_V_PORT GPIOB
#define LEFT_HALL_W_PORT GPIOB

#define RIGHT_HALL_U_PIN GPIO_PIN_10
#define RIGHT_HALL_V_PIN GPIO_PIN_11
#define RIGHT_HALL_W_PIN GPIO_PIN_12

#define RIGHT_HALL_U_PORT GPIOC
#define RIGHT_HALL_V_PORT GPIOC
#define RIGHT_HALL_W_PORT GPIOC
```

But i guess that the two motors connected should do something when the board is powered with the 2A CC power supply.

Did you bridge the on/off button after successfuly flashing the hex file and power off the board ?

I-hate-2FA, 2024-08-19:
im not sure if itll work, only the one written the firmware will know
if it check for on off button first or the adc then it wont work

LeonXBB, 2024-08-19:
Like I said before, the motors weren't connected, only the power :)
Also, I'm sorry, what do you mean by bridging the button?

RoboDurden, 2024-08-19:
Of course you have to connect both motors with their 3 phase cables and the hall connector.
These boards only have a push putton to turn the board on and off.
Normally, that button will trigger the hold-transistor to keep the board powered when the push button is released.
As we do not know the pin that controls that hold transistor, you need to permanently push the onoff button = bridge the 2 pin socket.

After flashing, you should disconnect the 3.3V from st-link dongle and connect the 2A CC power supply.

RoboDurden, 2024-08-19:
in main.c , the board will poweroff and get stuck in an wendless while(1){}; loop when battery voltage is too low:

```
   // ####### BEEP AND EMERGENCY POWEROFF #######
    if ((TEMP_POWEROFF_ENABLE && board_temp_deg_c >= TEMP_POWEROFF && abs(speed) < 20) || (batteryVoltage < ((float)BAT_LOW_DEAD * (float)BAT_NUMBER_OF_CELLS) && abs(speed) < 20)) {  // poweroff before mainboard burns OR low bat 3
      poweroff();
```

As the adc pin for battery voltage might be wrong, @LeonXBB  might need to remove this code lines ( add // in before). 
That requires recompilation, which can be done with my online compiler. But for some reason the project tree is no longer shown for LucysRaush firmware. 

Maybe i have time to investigate tomorrow.

LeonXBB, 2024-08-19:
Sorry, I didn't know I had to connect the motors. I was expecting error beep or something.
Neither about the hold transistor, I was pushing it on and off as normally.

Yes, I disconnect ST-Link after flashing :)

I'll rectify those mistakes tomorrow and will post the results.

RoboDurden, 2024-08-19:
This binary might be LucysRaush in test mode without poweroff: https://pionierland.de/hoverhack/download/lucysrausch_hoverboard-firmware-hack_hover-test_2024-08-19_21_24_48.bin

You can compile yourself here: https://pionierland.de/hoverhack/?p=robo

changes i made in main.c :

```

/*
    // ####### POWEROFF BY POWER-BUTTON #######
    if (HAL_GPIO_ReadPin(BUTTON_PORT, BUTTON_PIN) && weakr == 0 && weakl == 0) {
      enable = 0;
      while (HAL_GPIO_ReadPin(BUTTON_PORT, BUTTON_PIN)) {}
      poweroff();
    }

    // ####### BEEP AND EMERGENCY POWEROFF #######
    if ((TEMP_POWEROFF_ENABLE && board_temp_deg_c >= TEMP_POWEROFF && abs(speed) < 20) || (batteryVoltage < ((float)BAT_LOW_DEAD * (float)BAT_NUMBER_OF_CELLS) && abs(speed) < 20)) {  // poweroff before mainboard burns OR low bat 3
      poweroff();
    } else 
      */
      
```

I am not sure if there are older changes to the config.h i made when i used that robo/ firmware.

RoboDurden, 2024-08-19:
You get some warnings when compiling but that is okay. binaries still run on the old Gen1 boards.

LeonXBB, 2024-08-19:
Once again, @RoboDurden, _danke_schon_!
I'll try it tomorrow.
I will also check the new mapping from the creator of [https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/191](this) issue

RoboDurden, 2024-08-19:
Yes you have to modify defines.h to match

```
BUZZER = PC15
HALL LEFT U = PB8
HALL LEFT V = PB7
HALL LEFT W = PB6
HALL RIGHT U = PA8
HALL RIGHT V = PA12
HALL RIGHT W = PC10
```

from what @peerke444 had already found out.

I-hate-2FA, 2024-08-20:
have you used constant current supply

I-hate-2FA, 2024-08-20:
> I think the lucys Rausch firmware is easier with the pin definitions. And for the exhibition the foc is not absolutely necessary.

i will have to try this on the gd32f103 board, if the pinout is possible to edit

LeonXBB, 2024-08-20:
So I've changed pins to what @peerke444 has already found. I hasn't gotten to actually test it with motors yet, but I played around anyway. 

First, I tried uploading the original (non-FOC) firmware with pins changed and power-off disabled. It gave me five beeps. Then, out of curiosity, I've tried uploading my intented firmware with pins changed. It gives familiar power-on melody, then it immediately goes into power-off and stucks the last note.

I need to find the button pin. Also, I hope to be able to test it with motors later today.

RoboDurden, 2024-08-20:
Remove the code in main.c as I have suggested to see if the power off melody disappears.

Are you using my online compiler ?

LeonXBB, 2024-08-20:
For non-FOC version, yes. Github actions for FOC version (because I was testing it with my actual intented firmware).

Yes, I forgot to remove the code for the FOC version, I'll retry it now, my bad.

I-hate-2FA, 2024-08-20:
should use none foc because foc cannot change pinout

LeonXBB, 2024-08-20:
Sorry, I'm not really sure what you mean?
I've changed the buzzer pin in the foc version and it work just fine.
Also halls and phases but haven't checked them yet

I-hate-2FA, 2024-08-20:
in foc version you cannot change adc pin

LeonXBB, 2024-08-20:
@RoboDurden I've disabled power-offs and now the board doesn't do that. Just beeps 4 times, PWM time-out, expected behaviour.

I do have to find the pin for the button though. And try it with motors later today.

LeonXBB, 2024-08-20:
Sorry, I still don't get it. 
What is the function of adc pin? What pin it's actually on? And why can't it be changed?
Could you explain further, please?

LeonXBB, 2024-08-20:
FOC is not important per se, I'm more worried about the existing code base that is built on the FOC version that I hope to get running on this 303 chip.
In fact, we were experimenting with different input types and modes to see which fits us best, and I don't think our current firmware even has it enabled.
I absolutely don't mind focusing on the LucysRausch version for the sake of community and all the help you've given me :)

I-hate-2FA, 2024-08-20:
better not tell him what the project is for so the chat stays purely technical
foc is not really that important imo, sine wave is already very good
the eferu code base is just a mess,forget about it
so insted use the none foc version

RoboDurden, 2024-08-20:
EFeruFOC is best for Gen1 boards if you do not want to change the code.
So bringing EFeruFOC to GD32F303 is best for the community.
But to get the motors spinning it is better to work with the most simple firmware.
And LucysRausch is the very first firmware on which most other firmwares have been built on.

I-hate-2FA, 2024-08-20:
eferu is only good if you have baodrd with exactly same pinout
the gd32f103 boards does not work at all

LeonXBB, 2024-08-20:
Returned to it about an hour ago and still having troules finding the correct pin for the on-off button using multimeter. Any advice for finding pins? :(

LeonXBB, 2024-08-20:
Also I have noticed non-foc version doesn't have configurability for input pins. Does that mean they're not configurable?

I-hate-2FA, 2024-08-20:
if you really wanted to, there is a n transistor power the big p transistor, see where the n transistor gate is connected to

I-hate-2FA, 2024-08-20:
trace the latch/self hold/button is not for beginer without pinfinder/autodetect

RoboDurden, 2024-08-20:
Thats not really helpful.
This is the mcu on the board:
[image]

Minus
```

SWCLK = PA14
SWDIO = PA13
BUZZER = PC15
HALL LEFT U = PB8
HALL LEFT V = PB7
HALL LEFT W = PB6
HALL RIGHT U = PA8
HALL RIGHT V = PA12
HALL RIGHT W = PC10
LEFT UH = PC8
LEFT UL = PB1
LEFT VH = PC7
LEFT VL = PB0
LEFT WH = PC6
LEFT WL = PA7
RIGHT UH = PA10
RIGHT UL = PB15
RIGHT VH = PA9
RIGHT VL = PB14
RIGHT WH = PA8
RIGHT WL = PB13
```

I-hate-2FA, 2024-08-20:
on the f303RC theres quite allot more pins

LeonXBB, 2024-08-20:
I can confirm this board has 64 pins

I-hate-2FA, 2024-08-20:
i want to say it have more then typical 48pin from gen2 but anyways you can still eventually find the pins, but for hall sensor i sugest to trace them 
because hall sensor wiring is simple, with help of a flashlight on the back using visual tracing is possible in 10 minute
button and latch on the other hand is very hard for beginer to trace

LeonXBB, 2024-08-20:
So I have made the following table

[image]

I have noticed PA8 has two functions. Do you think it's possible or a mistake?

I-hate-2FA, 2024-08-20:
not possible, motor need its own 6 pins which is fixed

I-hate-2FA, 2024-08-20:
be careful how to read the mcu pin, this is the wrong orientation

> Thats not really helpful. This is the mcu on the board: ![GF32F303 64pins](https://private-user-images.githubusercontent.com/12238841/359539503-503da4ae-ca42-4537-b544-9d3e189dce08.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQxNzM2NDEsIm5iZiI6MTcyNDE3MzM0MSwicGF0aCI6Ii8xMjIzODg0MS8zNTk1Mzk1MDMtNTAzZGE0YWUtY2E0Mi00NTM3LWI1NDQtOWQzZTE4OWRjZTA4LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODIwVDE3MDIyMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJhYzg4ZTA4NmYxN2E0ZGYxNDllN2M5M2I1ZGM2YTcwOGYyOGVkN2QwM2RhMjAxZmNhZjNiZDlhNzdjNjcyZWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.H4YqO2MiIBj0Qm9l0rTfTUhN4pe9KUZBvI5kleVdL5E)
> 
> Minus
> 
> ```
> 
> SWCLK = PA14
> SWDIO = PA13
> BUZZER = PC15
> HALL LEFT U = PB8
> HALL LEFT V = PB7
> HALL LEFT W = PB6
> HALL RIGHT U = PA8
> HALL RIGHT V = PA12
> HALL RIGHT W = PC10
> LEFT UH = PC8
> LEFT UL = PB1
> LEFT VH = PC7
> LEFT VL = PB0
> LEFT WH = PC6
> LEFT WL = PA7
> RIGHT UH = PA10
> RIGHT UL = PB15
> RIGHT VH = PA9
> RIGHT VL = PB14
> RIGHT WH = PA8
> RIGHT WL = PB13
> ```

[9 more replies]