RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #91: Board request GD32F130 C8T6 3 gate driver

TheCure935, 2024-09-02:
[image]

I-hate-2FA, 2024-09-03:
what do you want

TheCure935, 2024-09-03:
> what do you want

The wiki said to put in a “board request” if I couldn’t find mine in the drop downs

I-hate-2FA, 2024-09-03:
it is not needed any more there is auto detect firmware

TheCure935, 2024-09-03:
> what do you want

End goal I want to be able to control it like a servo

TheCure935, 2024-09-03:
> it is not needed any more there is auto detect firmware

Oh ok, I’ll check that out then

I-hate-2FA, 2024-09-03:
> End goal I want to be able to control it like a servo

its not possible on gd32 you can forget about it
it do not even have sine wave mode

TheCure935, 2024-09-03:
I can’t read the position data through the board?

> > End goal I want to be able to control it like a servo
> 
> its not possible on gd32 you can forget about it it do not even have sine wave mode

TheCure935, 2024-09-03:
> > End goal I want to be able to control it like a servo
> 
> its not possible on gd32 you can forget about it it do not even have sine wave mode

I was trying to figure out how to just get the thing moving with an radio o and then from there using the encoder attached to it to control the postion(very inaccurately ofc)

I-hate-2FA, 2024-09-03:
> I can’t read the position data through the board?

you can, but commutation have no torque at low speed and cannot position hold

> I was trying to figure out how to just get the thing moving with an arduino and then from there using the encoder attached to it to control the postion(very inaccurately ofc)

i think serial may be too slow for it, its better implement the pid on the hoverboard main board

TheCure935, 2024-09-03:
> > I can’t read the position data through the board?
> 
> you can, but commutation have no torque at low speed and cannot position hold
> 
> > I was trying to figure out how to just get the thing moving with an arduino and then from there using the encoder attached to it to control the postion(very inaccurately ofc)
> 
> i think serial may be too slow for it, its better implement the pid on the hoverboard main board, if its too slow itlll oscilate allot

I see, welp I have an odrive control board and encoder ordered so I’ll probs just wait for that then

also I noticed the hoverboard I have has plastic wheels not aluminum, it seems pretty tough but probs not the same level, is there a guide that has the models with aluminum wheels?

I-hate-2FA, 2024-09-03:
most problematic is at high power the wheel will heat up and the plastic soften, the aluminum is rated 350w but you can push it to 500w no problem, the plastic is max250w, some of them even less, its better to buy a aluminum one

RoboDurden, 2024-09-03:
@I-hate-2FA you must have a bad day today ! Sorry for that.
But almost everything you said here is wrong.
You always have been negative towards the work of others, but today i think you simply have a hard time. So happy greetings to you :-)

In your world, every GD32 hoverboard would be forbidden and put to mass destruction ?

A 500W servo ?
A servo might have a robotic use case, then the less weight of a plastic motor might even been better than the old aluminium ones. I already said that a few threads before.

But yes @TheCure935 , watch my latest autodetect youtube video and post the full log here.

If your servo use case indeed needs position hold, you/we will need to add a control loop into the firmware.
Either as RemoteServo or as the new class-like structure PilotServo ..

I-hate-2FA, 2024-09-03:
what

peterpan-bit, 2025-12-05:
I tried again and still cannot get the button pins: it says 0 pins found.

//#define BUTTON P??
//#define   BUTTON_PU  P??

I also didn't get the HallOrder.
I flashed the firmware without the button pins and tested the firmware using the TestSpeed code. The the motor moves a little abruptly and stuck with large noise inside the motor and the board is buzzing. Is that because the pins? Please let me where is going wrong? Here is the list of pins I used:

#define HALL_A		PC14
#define HALL_B		PB11
#define HALL_C		PA1
//#define PHASE_A		P??
//#define PHASE_B		P??
//#define PHASE_C		P??
		
#define LED_RED		PB8
#define LED_ORANGE		PA3
#define LED_GREEN		PB9
#define UPPER_LED		PC13
#define LOWER_LED		PC15
#define ONBOARD_LED		PC14
#define BUZZER		PF0
		
#define VBATT		PA5
#define CURRENT_DC		PA6
#define SELF_HOLD		PB2
//#define BUTTON		P??
//#define BUTTON_PU		P??

peterpan-bit, 2025-12-07:
I am testing the speed using REMOTE_UARTBUS by connecting the slave board using  the original UART Bus cable.

The console only shows iSlave id 0, even after I try using si 1. Both motors are running the same. How do I assign the slave board as id 1, so I can control the two motors separately through the UARTBUS? I see the master firmware is assigned 0, but that part is greyed out in the slave firmware. Do I need to assign it in the config file of the slave firmware?

RoboDurden, 2025-12-07:
Slave might need the positive voltage on the masterslave cable to turn on.

Make sure the button check is disabled in config.h as the slave board has not on off button .