RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #92: Re-implment balance function with IMU

hoverboardhavoc, 2024-09-03:
It would be interesting to make a hoverboard balance again with this firmware. I think the chip marked with 16A 3C6A C3C might be an IMU. I've Googled and I can't find any information. How can I work out how to interface with the chip? I assume it must be a SPI or possibly I2C bus. I think that [extracting the original firmware](https://www.usenix.org/system/files/woot20-paper-obermaier.pdf) is tricky, at least for me. Maybe there is a way to run the original firmware and spy on the signals sent from the micro controller to the chip?

[image]

RoboDurden, 2024-09-03:
if you want this rather stupid self balancing toy, then buy a new one..
You could try to run the arduino  i2c scanner  code to get the address this little chips responds to.
That might give you a clue to what type of imu it is.
But IMU is not at all the scope of this firmware.

I-hate-2FA, 2024-09-04:
the imu is connected with i2c, but its better to use a new imu and connect it to your esp

hoverboardhavoc, 2024-09-04:
Thanks for the responses, I have a few of these Classy Walks to play with. 

I agree that just recreating the existing functionality is boring, but it might be fun to turn it into a two wheeled remote controlled vehicle.

I-hate-2FA, 2024-09-04:
just put a imu on esp and write balance code on it, 100ms latency should be ok to balance if the mechanical design is not terrible

I-hate-2FA, 2024-09-04:
but dont share your project with too much unnecessary detail on this repo, its for your own good

hoverboardhavoc, 2024-09-04:
ok thanks for the help. I'll close this one

closed 2024-09-04 by hoverboardhavoc

hoverboardhavoc, 2024-09-08:
I made a start on this https://github.com/hoverboardhavoc/HoverboardImu/

RoboDurden, 2024-09-08:
Yes nice 🙂 
Good start with a detailed read.me.

Maybe you want to look at all the different layouts here to find some more boards using this IMU.

RoboDurden, 2025-06-26:
Will your code also compile with the Keil IDE ?

RoboDurden, 2025-06-26:
no it is a normal ebike hub motor with 3phases and hall cables. Should work with a gen2.1 board.
I do not have the happiness to port the code myself into Gen2.x
And when i look at https://github.com/hoverboardhavoc/hoverboard-gen2.1-hack-GD-imu/tree/main/Src is see too many new files.
If you can merge all the i2c and imu code into one mpu6050.c and mpu6050.h file, that also contains the setup code with only two functions to be called from main() imuSetup() and imuUpdate(), then it would be easy for me to add to Gen2.x

hoverboardhavoc, 2025-06-28:
Another update: The code on the [branch](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-GD32/compare/main...hoverboardhavoc:Hoverboard-Firmware-Hack-Gen2.x-GD32:self_balance) works but is slow. I'm considering removing the mpu6050 library pasted from the https://github.com/EFeru/hoverboard-sideboard-hack-GD project and going for something much simpler that does not use interrupts since my IMU does not support DMP.

hoverboardhavoc, 2025-06-28:
https://github.com/user-attachments/assets/e69cd9da-d3ae-4ed9-b33d-d05f37e7b797

RoboDurden, 2025-06-28:
That video really looks great.
If you can isolate the imu code to just one imu.c / imu.h pair, i really would like to add the code gen2.x

@DDoru If you have such boards to spare, i would send your 5 Euro via paypal for shipping to Germany. Maybe also 10 Euro after i have searched my stock of hoverboards and found now gen2 with such an imu.

Maybe i could not only detect braking but also pedaling on an ebike as a periodic little rotation alongside the driving direction..

hoverboardhavoc, 2025-06-28:
I want 4 files 😱

generic i2c functions: i2c.h / c
mpu6050: mp6050.h / c
balance logic / pid controller: balance.h / c
Attitude And Heading Reference System (AHRS) algorithm (pasted code) MadgwickAHRS.c /h

You would only need the first two @RoboDurden if you want the raw data from the IMU. Obviously they will all be enabled/disabled with `#ifdef`s

hoverboardhavoc, 2025-06-28:
Join us on Platform IO @RoboDurden 🤗

RoboDurden, 2025-06-28:
I hope that i will get BLDC_SINE to work for 32f103 today.
So i would not mind f103 if your imu code runs on it.
here my email for shipping address and your paypal account: roland@robodurden.de
Where are you from ? Should be inside Europe for reasonable shipping cost and shipping time.

Have you tested the imu code both for f130 and f103 ?

hoverboardhavoc, 2025-06-28:
I've not made the F103 work yet at all. I sent you an email @RoboDurden

hoverboardhavoc, 2025-06-28:
well it works 🥳  Maybe some tuning required. I want to try on a bigger vehicle

https://github.com/user-attachments/assets/9a0cfdbc-5f63-47f3-b021-18f2e393f75c

RoboDurden, 2025-06-28:
i don't really know in how the imu is used for balancing. I think the balancing is mainly achieved via the pressure off the foot onto these opto triggered black pads.

hoverboardhavoc, 2025-06-28:
The two wheels fight with each other so I'm going to run the PID loop on the microcontroller, meaning our requirements are pretty similar @RoboDurden. I'll just send the raw values over the UART bus. I'm currently using `remoteUart`

	typedef struct{
		int16_t     x;
		int16_t     y;
		int16_t     z; 
	} Gyro;

	typedef struct{
		int16_t     x;
		int16_t     y;
		int16_t     z; 
	} Accel;

RoboDurden, 2025-06-28:
Please link to your github project when the code has been cleaned, so i can see how complex it is.

RoboDurden, 2025-06-29:
I may want to use one of my gen2.1.3 boards for the ebike.
This is the imu on that board.
Maybe you have some ideas:

[image]

hoverboardhavoc, 2025-06-29:
Maybe it's worth playing with https://github.com/hoverboardhavoc/HoverboardImu It has lots of different branches with different arduino IMU libraries. It will build from platform IO.

    ICM-20948
    ITG3200
    LSM6
    MPU6050
    MPU9250

My IMU gave an unusual WHO_AM_I response (`0x79`) but it still works like a MPU6050

RoboDurden, 2025-07-15:
> Not sure if it is useful, but I found out that the IMU used with the F130K6/8 is a Bosch BMI120. Not sure how they used it, as it is definitely wired in I2C mode, and those pins do not really support it..

@ThorsAvenger which layout do you mean ?
I already suspect the 2.2 to have a IMU connected via SPI, which the BOSCH BMI120 would support:

```c++
3.2.1 Primary Interface I2C/SPI Protocol Selection
The protocol is automatically selected based on the chip select CSB pin behavior after power-
up.
At reset / power-up, BMI120 is in I2C mode. If CSB is connected to VDDIO during power-up and
not changed the sensor interface works in I2C mode. For using I2C, it is recommended to hard-
wire the CSB line to VDDIO. Since power-on-reset is only executed when, both VDD and VDDIO are
established, there is no risk of incorrect protocol detection due to power-up sequence.
If CSB sees a rising edge after power-up, the BMI120 interface switches to SPI until a reset or
the next power-up occurs. Therefore, a CSB rising edge is needed before starting the SPI communication. Hence, it is recommended to perform a SPI single read access to the
ADDRESS 0x7F before the actual communication in order to use the SPI interface.
If toggling of the CSB bit is not possible without data communication, there is in addition the
spi_en bit in Register (0x70) NV_CONF, which can be used to permanently set the primary
interface to SPI without the need to toggle the CSB pin at every power-up or reset.
```
https://lcsc.com/datasheet/lcsc_datasheet_1912111437_Bosch-Sensortec-BMI120_C437657.pdf

ThorsAvenger, 2025-07-16:
According to your numbering it is v3=2.2. I got a response for it using the SoftI2C library, combined with DF_Robots BMI160 library, which I think makes use of bit banging. For SPI you need to be able to set the Chip Select pin, but on my board that pin on the imu is hardwired to 3V3. I've also tried to read the NV_CONF register, but is reads all zeros, meaning the spi_on bit has not been set.

RoboDurden, 2025-07-16:
I have two pairs of 2.1.3 and they nicely small. Would be great if you get the imu to work as i would like to use such a small gen2 board for my ebike project. So the SoftI2C library is not bound to the PB6/PB7 and PB8/PB9 pins ?
In the manual i read that the bosch imu can be configured to be in spi mode without the need put a rising flank on the spi/i2c pin..

Deepseek just helped me with the correct init code for GD32E230 (might be the same for GD32F130):

```c++
void i2c_config(void) 	// Deepseek
{
    rcu_periph_clock_enable(RCU_GPIOB);
    rcu_periph_clock_enable(RCU_I2C0);

	#ifdef PB6PB7	
		// Configure PB6 (SCL) and PB7 (SDA) as AF open-drain
		gpio_mode_set(GPIOB, GPIO_MODE_AF, GPIO_PUPD_PULLUP, GPIO_PIN_6 | GPIO_PIN_7);
		gpio_output_options_set(GPIOB, GPIO_OTYPE_OD, GPIO_OSPEED_50MHZ, GPIO_PIN_6 | GPIO_PIN_7);
		gpio_af_set(GPIOB, GPIO_AF_1, GPIO_PIN_6 | GPIO_PIN_7);  // AF1 for I2C0
	#else
		// Configure PB8 (SCL) and PB9 (SDA) as AF open-drain
		gpio_mode_set(GPIOB, GPIO_MODE_AF, GPIO_PUPD_PULLUP, GPIO_PIN_8 | GPIO_PIN_9);
		gpio_output_options_set(GPIOB, GPIO_OTYPE_OD, GPIO_OSPEED_50MHZ, GPIO_PIN_8 | GPIO_PIN_9);
		gpio_af_set(GPIOB, GPIO_AF_1, GPIO_PIN_8 | GPIO_PIN_9);  // AF1 for I2C0
	#endif	

    // I2C configuration (Standard mode: 100kHz)
    i2c_deinit(I2C0);
    i2c_clock_config(I2C0, 100000, I2C_DTCY_2);
    i2c_mode_addr_config(I2C0, I2C_I2CMODE_ENABLE, I2C_ADDFORMAT_7BITS, 0x00);
    i2c_enable(I2C0);
    i2c_ack_config(I2C0, I2C_ACK_ENABLE);
}	
```

Then on my 2.3.2 the ads1115 on PB6/PB7 was detected, and now there is an i2c device on PB8/PB9: 0x68
i hope, it is a MPU6050

RoboDurden, 2025-08-05:
[[image]](https://youtu.be/KDhp0SOtsDc)

Ah, now i undestand, accel is indeed the acceleration but it contains the 9.80665 m/s² Gravitation. That is why the xyz values change when i rotate the board.
So i should be able to detect braking. Like this small push in x direction:
[image]
Gravtition here in the y orientiation of the onboard mpu6500.

But the gyro data still seems like rotational speed to me and not rotation acceleration. These are the readings of a 90° left rotation around the x-axsis of the mpu6500:

[image]

If it would be acceleration, there should be a positive spike that initiates the rotation followed by a negative spike when the roation is finished.
But gyro.x stays negative for the whole rotation, being minimal in the middle - where the rotation speed is max. Just as i would expect it from rotational speed.
The first image of the translational acceleration instead does show the acceleration and braking with the push in x-direction.

Okay, with the kalman filter, the direction of the gravtition can assist the integration of the rotational acceleration to give a drift free rotation angle. Well only for the x and y rotation. for the z-rotation, the gravitaional vector will not change.  For hoverboard balancing however, you only need the x-rotation :-)
For a lawn mover however, the z-rotation is what we need to change direction :-(
So adding an compass module to the i2c bus might be a good idea.

As i do not need that for my ebike brake/pedaling detection, i am happy with the raw data.

if your  kalman filter indeed removes rotational drift for the x and y-axis, it should be run by the hoverboard. With optional i2c compass module ?
esp32/rpi might be busy with wifi and vision-ai.

Is there some optiocal mouse i2c module that could track the underground of a lawn mover to feed into a kalman filter together with the acceleration data ?

Deepseek, Is there a cheap i2c module that works like an optical mouse to track the underground of lawn mover giving speed and position of the driving robot ?

$9 PMW3901 Optical Flow Sensor Module : https://de.aliexpress.com/item/1005007333832846.html

>Yes, there are affordable I2C modules that function like optical mouse sensors, designed to track ground movement for robotics applications like lawn mowers. Here's a practical solution:
...
Position Drift: Optical flow sensors accumulate error over time.
Solution: Fuse with wheel encoders or IMU (accelerometer/gyro) using sensor fusion (e.g., Kalman filter).

Indeed, combining iOdom (wheel rotation) with the mpu6500 data in a kalman filter might remove the slip of the wheels on the gras :-)

RoboDurden, 2025-08-22:
You want to emulate the onboard i2c IMU with an EPS32 so that you can trick the boards to spin motors like you want them to do ?
Looks like a lot of work to me and every layout has different IMU.
I am not sure if my imu support for 2.1.20 will work for even a few other boards.

I think the original hoverboard logic also includes the two IR-PHOTO sensors on the backside.
But maybe these only check that the foot is firmly on the hoverboard placed.

But i was thinking if the old firmware could be read to a bin file.
And then simply let Gemini anaylse the assambler code to rework the i2c protocol for the onboard IMU.
Might not even need to recompile into c-code.
I2c communication is very simple and most imu have 0x68 as the i2c address.
So it could be quite simple for ai to find the read/write operations to 0x68..

Good use case for ai.

RoboDurden, 2025-08-22:
sniffing with such a cheap logic analyzer https://de.aliexpress.com/item/1005009676857208.html ?
Or with ESP32 as I2C slave and somehow outputing the sda stream ?

What layout are you working with ?
Why are you not happy with my firmware ?
Have you tried my new 	`#define BLDC_SINE_BOOSTER		// can boost speed by 15% starting from 87% throttle.` ?

But yes, these GD32E230 boards are so hard to flash, removing the imu and control the motors like these hovercarts, might indeed work.

[image]

WestlingPi, 2025-12-28:
Ok, sure. A define would be fine for me too. I can add it when I start on the RemoteROS2 changes.

Are you aware that there is a lowpass filter in the IMU hardware (configured by register 26 for MPU6050 and 26 & 29 for MPU6500)? Maybe you can achieve the same thing with a higher(?) value of register 26/29 and thus would not need an additional lowpass filter in the firmware software? Just a thought that might be worth exploring... Otherwise, I think the current configuration and a define for the software lowpass filter will work for me.

WestlingPi, 2026-01-11:
I have been working on sending IMU data from the RemoteROS2 (and I think it is ready now). However, when testing RemoteROS2 I ran in to some problems that I think is i2c related.

The symptom I get is that reading the sensor values works great for a while, then suddenly it stops working and never recovers. At first I thought it was just after a while (10-30s after power on) but later I discovered that it is probably related to when increasing the bldc speed from 0.
@RoboDurden I have seen some comments in the code that maybe you have experienced similar issues, eg:
`I2C_Init();	// try to re-init if i2c bus fails. Need when I2C_SPEED is 400000 instead of 200000`

So the last few days I have been debugging this issue. I added some debug counters to i2c_readBytes(), one for every error case, and it turns out that when it fails for me, it is always in this part of i2c_readBytes():

```
    i2c_master_addressing(i2c_periph, devAddr << 1, I2C_TRANSMITTER);
    tmo = 0;
    while (!i2c_flag_get(i2c_periph, I2C_FLAG_ADDSEND)) {
        if (++tmo > I2C_TIMEOUT) {
            i2c_stop_on_bus(i2c_periph);
            i2cE3++;
            i2c_hard_reset(i2c_periph);
            return I2C_ERR;
        }
    }
    i2c_flag_clear(i2c_periph, I2C_FLAG_ADDSEND);
```

At first I tried to toggle the devAddr between 0x68 and 0x69 on every failed attempt, but that didnt help. (There is an AD0 pin on mpu6050/6500 that selects i2c address 0x68 or 0x69 that I thought it could be something wrong with - but the pcb seems ok).

Then Gemini suggested to add a i2c_hard_reset():
```
void i2c_hard_reset(uint32_t i2c_periph) {
	// 1. Force the I2C peripheral into reset state
	i2c_software_reset_config(i2c_periph, I2C_SRESET_SET);
	for(volatile int d=0; d<1000; d++);
	i2c_software_reset_config(i2c_periph, I2C_SRESET_RESET);

	// 2. Re-initialize the I2C peripheral from scratch
	I2C_Init();
}
```

With this code, whenever i2c_readBytes() fail, it recovers and is able to read the next time. According to Gemini the i2c bus is sensitive to electromagnetic noise from the bldc driver and in particular the STM32F103 hardware implementation of i2c is more sensitive than the corresponding GD32 hardware.

@RoboDurden Do you think I should add the i2c_hard_reset() to i2c_readBytes(), or do you have any other idea? With this fix I am able to run the motor (accelerate, decelerate, continuous  speed etc). The i2c reads fails sometimes, but it always recovers. I dont experience any "hangings" anymore. I have not tried yet if it is enough with only I2C_Init() to recover. Also, do you prefer to have the recovery in i2c_readBytes() or in the users (in your case PilotHoverbike and my case RemoteROS2)?

RoboDurden, 2026-01-11:
Oh yes, great you solved this issue! I got quite frustrated when my hoverBike imu reading always failed.
So yes, your hard_reset should go into readBytes. 
But not always hardReset but only when readBytes fail, do a hardReset and repeat readBytes ? Maybe a short delay is then needed ?

WestlingPi, 2026-01-12:
Pull request submitted: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-GD32/pull/21