RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #93: PlatformIO support :-)

RoboDurden, 2024-09-04:
Thanks @hoverboardhavoc  for his pull request  https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-GD32/pull/3

Maybe adding stm32F103 support now has become closer.
adding this to platformio.ini might somehow do the trick:

```
[env:STM32F103RC]
platform = ststm32
//platform_packages = framework-arduinoststm32@https://github.com/Candas1/Arduino_Core_STM32.git
board = genericSTM32F103RC
debug_tool = stlink
upload_protocol = stlink
framework = arduino
build_flags = -g -ggdb
			  -D $PIOENV
			  -D GD32F103
			  -D TARGET=2
lib_deps = 
lib_archive = false
```

taken from https://github.com/RoboDurden/Split_Hoverboard_SimpleFOC/blob/main/platformio.ini

RoboDurden, 2024-09-06:
Concerning clock speed, this might help:

```
build_flags =
	-g -ggdb
    -D __PIO_DONT_SET_CLOCK_SOURCE__
	-D __SYSTEM_CLOCK_48M_PLL_IRC8M_DIV2=48000000
	-D $PIOENV

```

Electron-py, 2024-09-06:
Getting Error: This board doesn't support spl framework!, someone can help?

Electron-py, 2024-09-06:
My platformio.ini file: 
[platformio]
src_dir = Src
include_dir = Inc

[env]
platform = https://github.com/CommunityGD32Cores/platform-gd32.git
platform_packages = framework-spl-gd32@https://github.com/CommunityGD32Cores/gd32-pio-spl-package.git

[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = spl
build_flags = -D GD32F130
			  -D TARGET=1
			  -D USE_HSI_CLOCK
			  -DSTM32F100xB -DSTM32F10X_MD

Electron-py, 2024-09-06:
Found this: https://github.com/CommunityGD32Cores/gd32-pio-spl-package

RoboDurden, 2024-09-06:
Yes and you have it already selected in your [env] section.
But in the (overwriting ?) [env:genericSTM32F103C8] section you switched back from gd32 to stm32 :-/

Electron-py, 2024-09-06:
I'm trying to

Electron-py, 2024-09-06:
Please specify platform for 'genericGD32F130C8' environment

RoboDurden, 2024-09-06:
If you are running Windows, check your platformIO folders. Mine are found in 

C:\Users\PAN CF-LX6\.platformio\packages
C:\Users\PAN CF-LX6\.platformio\platforms

Electron-py, 2024-09-06:
Installed package but i still don't know how do i 'specify platform'?

RoboDurden, 2024-09-06:
[image]

Electron-py, 2024-09-06:
can you send me the gd32 file inside platform folder and the framework one?

Electron-py, 2024-09-06:
[image]

RoboDurden, 2024-09-06:
https://robosoft.de/forums/hover/

but these files should be auto downloaded when you compile the project.
Try to clear and clean the project..

I-hate-2FA, 2024-09-06:
will we get online editor for gen2 anytime soon

Electron-py, 2024-09-06:
Can't even clean because of this "Please specify platform". Sorry if I'm not understanding something

Electron-py, 2024-09-06:
I have to go. If you find a solution...

RoboDurden, 2024-09-06:
I am not at all experienced with PlatformIO. I also only used it for one project two years ago.
If you do no longer use PIO for this other project of yours, i would uninstall and reinstall PIO..

Electron-py, 2024-09-06:
ok

RoboDurden, 2024-09-06:
and goole your errors.
https://www.google.com/search?client=firefox-b-d&q=%22Please+specify+platform+for

Electron-py, 2024-09-06:
alredy tried that

RoboDurden, 2024-09-06:
@Electron-py actually i do not understand why you want to compile for a [env:genericSTM32F103C8] . You have a Gen2.1.1 which is GD32F130

When i add for the Gen2.2.t boards

```
[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = spl
build_flags = -D GD32F103
				-D TARGET=1
				-D USE_HSI_CLOCK
				-DSTM32F100xB -DSTM32F10X_MD
```

i get the `Error: This board doesn't support spl framework!`

that might help: https://github.com/platformio/platform-ststm32/issues/13#issuecomment-378051575

my `genericSTM32F103C8.json` is in `C:\Users\PAN CF-LX6\.platformio\platforms\ststm32\boards`

But i failed at step 2.
This is my stm32f1 folder:
[image]

the startup file needs to be changed in target.h line 48

```
	#include "startup_stm32f10x.c"
	//#include "gd32f10x.h"
```

But that is not working

```
Src\../Inc/../Inc/target.h:48:11: fatal error: startup_stm32f10x.c: No such file or directory
  #include "startup_stm32f10x.c"
           ^~~~~~~~~~~~~~~~~~~~~
```

And as a Keil/PIO beginner i do not really know how `#include "gd32f10x.h"` works successfully..

Electron-py, 2024-09-06:
I made it work. Now i have an error when compiling about RemoteCallback being defined more times.

RoboDurden, 2024-09-06:
> I made it work. 

What did you make work ?
And how did youmake it work ??

RemoteCallback is defined in every RemoteXY.h file. So i guess you did include multiple Remotes.

Electron-py, 2024-09-06:
I found a pakcage for gd32 and now i don't have enviroment errors anymore

RoboDurden, 2024-09-06:
this is not very instructive :-(
You found the package under your kitchen table ?
Still trying to compile stm32f103 ??
platformio.ini ???

hoverboardhavoc, 2024-09-06:
> build_flags =
> 	-g -ggdb
>     -D __PIO_DONT_SET_CLOCK_SOURCE__
> 	-D __SYSTEM_CLOCK_48M_PLL_IRC8M_DIV2=48000000
> 	-D $PIOENV

I tried this
```
[env:genericGD32F130C8]
board = genericGD32F130C8
framework = spl
build_flags = -D GD32F130
			  -D TARGET=1
			  -g -ggdb
			-D __PIO_DONT_SET_CLOCK_SOURCE__
			-D __SYSTEM_CLOCK_48M_PLL_IRC8M_DIV2=48000000
			-D $PIOENV
```
and this:

```
[env:genericGD32F130C8]
board = genericGD32F130C8
framework = spl
build_flags = -D GD32F130
			  -D TARGET=1
			-D USE_HSI_CLOCK
            -DSTM32F100xB
			-DSTM32F10X_MD
```

but the beeper tones are still lower than they should be

I-hate-2FA, 2024-09-06:
you need to use proper stm32f103 core if you want to set system clock

RoboDurden, 2024-09-06:
based on 

```
//----------------------------------------------------------------------------
// SysTick_Handler
//----------------------------------------------------------------------------
void SysTick_Handler(void)
{
  msTicks++;
}
```

in `it.c` you should be able to let the buzzer beep (or let red led blink) in 10 second intervals. Then measure the time it really takes and we might be able to see at how many Mhz the cpu is running.
This might give us a clue why the clock is wrong.

And if the motor is still running nicely, we might live with the wrong cpu clock and scale the beeper tone ..
But i guess that the uart baud rate would also have to bee scaled :-/

RoboDurden, 2024-09-06:
This platformio.ini did make the GD32F130 run with correct clock. Serial communcation was working.
https://github.com/RoboDurden/Split_Hoverboard_SimpleFOC/blob/main/platformio.ini

I-hate-2FA, 2024-09-06:
it is very clear, you are using gd32f103 core, its normal that clock cannot be configured, its same for mm32f103, the clock will be lefty as default

Electron-py, 2024-09-07:
I installed this: https://github.com/CommunityGD32Cores/platform-gd32 in PlatformIO and this is my platformio.ini file:
[platformio]
src_dir = Src
include_dir = Inc

[env]
platform = https://github.com/CommunityGD32Cores/platform-gd32.git
platform_packages = framework-spl-gd32@https://github.com/CommunityGD32Cores/gd32-pio-spl-package.git

[env:genericGD32F130C8]
board = genericGD32F130C8
platform = gd32
framework = spl
build_flags = -D GD32F130
			  -D TARGET=1

Now the enviroment works. The error that i have is that the RemoteUpdate function id defined multiple times: One in remoteUartBus.c and one in remoteUart.c . When Platform IO tries to link those f
[image]
[image]
iles gives errors because when it links them there are two RemoteUpdate functions.

Electron-py, 2024-09-07:
I made it compile! I simply renamed the conflicting functions in remoteUart adding a 2 and now it compiles.
[image]

Electron-py, 2024-09-07:
But the beeping is wrong. @RoboDurden  I'll try with the platform.ini from SimpleFOC

Electron-py, 2024-09-07:
https://github.com/user-attachments/assets/04be07be-f72d-448d-8dc0-405a6ee89684

hoverboardhavoc, 2024-09-07:
> based on
>
> ```
> //----------------------------------------------------------------------------
> // SysTick_Handler
> //----------------------------------------------------------------------------
> void [...]

I added some logic to turn on the led for 50ms every 5 seconds: https://github.com/hoverboardhavoc/Hoverboard-Firmware-Hack-Gen2.x-GD32/tree/hoverboardhavoc/debug_clock

In the videos below which show hoverboardhavoc/debug_clock running, you can see the LED flashing every five seconds but the behavior of the motor and buzzer is different. The tones are different pitch and the motor makes a kind of metallic sound on the Platform IO build

Keil:

https://github.com/user-attachments/assets/103863e0-74b4-4327-a064-ee0be5972f09

Platform IO:

https://github.com/user-attachments/assets/05b87db9-5f98-4c3c-9b65-fb16dd42c303

Electron-py, 2024-09-07:
@hoverboardhavoc i can confirm. Don't know why it makes that metallic sound.

Electron-py, 2024-09-07:
Maybe it's a different implmentation for PWM or idk

Electron-py, 2024-09-07:
How do I enable the Dummy on firmware?

RoboDurden, 2024-09-07:
> One in remoteUartBus.c and one in remoteUart.c . 

As I said, you must only uncomment ONE RemoteXy in config.h

Is uart working with 19200 baud ?

Bldc is running from a 16 kHz interrupt that is triggered when ADC dma has finished, if I remember correctly.

Add a counter and blink led every 160000  counts to that function in bldc.c to verify that it is indeed 16 kHz...

This firmware has the very simple block communication only.
If you set the speed to 1000 = 100% pwm duty cycle, then the 16 kHz is the only thing that should affect motor sound.

I think that the buzzer tone is also generated in that calculateBldc function !

Electron-py, 2024-09-07:
> > One in remoteUartBus.c and one in remoteUart.c .
> 
> As I said, you must only uncomment ONE RemoteXy in config.h
Right, sorry. Now it compiles with the original void names.

Electron-py, 2024-09-07:
What am I doing wrong? Using REMOTE_DUMMY

https://github.com/user-attachments/assets/b6eadf11-398a-4bf0-8ed2-0a405b0f0ff2

RoboDurden, 2024-09-07:
Looks like not all three hall sensors working . Are the three led blinking with always two overlapping ?

But it might be the wrong 16 kHz timing of calculateBldc that needs your push with a hand at low speed to make the motor start spinning

Electron-py, 2024-09-07:
But even when it starts spinning it doesn't want two. It seems like there are only two fases. And my hoverboard has only red and green light, not orange.

RoboDurden, 2024-09-07:
With a Gen2.1.1 you should have a two color led module and a three color led module.
And if not you could reasign the up-led or down-led to be the orange one in your defines file.
Was the motor spinning correctly with the precompiled dummy ? does it spin correctly with remoteUart and arduino/esp spinning forward and backwards ?

Electron-py, 2024-09-07:
I'll try with uart

Electron-py, 2024-09-07:
nope. still wrong

RoboDurden, 2024-09-07:
i can not really help you. With all your short one liners i never really know what you are doing :-(

Electron-py, 2024-09-07:
I tested with Uart and its the same of the video i sent.

RoboDurden, 2024-09-07:
By the way, there is a great visualization tool that also supports GD32F130: https://www.st.com/en/development-tools/stm-studio-stm32.html
If i remember correctly, you have to compile in debug mode, then choose the generated .elf file, and then you can log every global variable. No need to led an led blink..

I-hate-2FA, 2024-09-09:
1.use higher voltage
2.use field weakening

Electron-py, 2025-04-05:
is field weakening implemented right now?

RoboDurden, 2025-04-05:
No.
Doesn't field weakening needs FOC ?

My simpleFOC gen2 firmware might be able to do field weakening.

There is also the EFeru foc port for those gen2 boards with phase current opamps..

novval, 2025-04-05:
> There is also the EFeru foc port for those gen2 boards with phase current opamps..
Can I have a link, please?

I-hate-2FA, 2025-04-06:
field weakening can be done in sine wave no foc is needed, i can acheive in my firmware

I-hate-2FA, 2025-04-06:
there is no real support for it but i can change the strengh manually by editing a number in code, i tried it it significantly increase the speed with same voltage (at much higher current)

Electron-py, 2025-04-06:
I retried compiling with platform io all stock not changed a line from .in file but still same error. I'm not sure I have selected the right layout. How do you do that in the code? I don't remember.

RoboDurden, 2025-04-06:
I just checked and it comiles sucessfully.
But i get a ton of warnings like:

```
Src\../Inc/../Inc/target.h:188:78: warning: suggest parentheses around '+' in operand of '&' [-Wparentheses]
  188 |  #define PIN_TO_CHANNEL(pin) ((pin&0xffffff00U) ==  GPIOA ? (pin&0xLinking .pio\build\genericGD32F130C8\firmware.elf
fU) : (pin&0xfU+8) )
```

You should select the gd32f130 enviroment:

```
[env:genericGD32F130C8]
board = genericGD32F130C8
framework = spl
build_flags = -D GD32F130
			  -D TARGET=1

```
[image]

Then choose your layout in config.h

[image]

Yes, i only boosted the max speed from 10 kmh to 12 kmh with field weakaning for my solar car with EFeru firmware. Higher speeds were not really achieavable and power consumption raised (current) significantly.
With my ESP32 20A dcdc boster, i go to 22 kmh when it is flat by raising the battery voltage from 26V to 50V :-)

Electron-py, 2025-04-06:
So to select layout 2.1.1 I keep it like this, right?

Electron-py, 2025-04-06:
LAYOUT 1
LAYOUT_SUB 1

Electron-py, 2025-04-06:
I don't know what's the issue but it tries to spin but the motor acts weirdly.

[21 more replies]