RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #94: Gen2.1.1 Hoverboard Single Uart firmware turns on and off continuosly

Electron-py, 2024-09-06:
I have tried flashing single uart firmware and master uart firmware to one of the boards. But when i press the button it beeps with a red light and then it turns off. If i have the esp spamming uart signals it turns on, spins for 1 second, green light and turns off. I have the GD32F130C8 hoverboard Gen 2.1.1

I-hate-2FA, 2024-09-06:
did you remember to disable battery protection

Electron-py, 2024-09-06:
what battery protection?

Electron-py, 2024-09-06:
Never heard of it before. How do i disable that?

RoboDurden, 2024-09-06:
Yes, looks like you have set the wrong battery voltage in config.h

`	#define BAT_CELLS         	10        // battery number of cells. Normal Hoverboard battery: 10s
`

Electron-py, 2024-09-06:
I have a 25.2v battery, i set 6 cells.

Electron-py, 2024-09-06:
im using the precompiled binary. I have a 25.2v battery

RoboDurden, 2024-09-06:
Then you have to compile yourself with 7s or 6s.

Electron-py, 2024-09-06:
Is there a guide for Platform IO?

Electron-py, 2024-09-06:
If not what program should i use?

RoboDurden, 2024-09-06:
I have uploaded a uart master 7s for you: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/BinariesReadyToFlash

RoboDurden, 2024-09-06:
Yes, there is PlatformIO support since yesterday. Would be nice if you test that with your setup. If you have already platformIO installed.

Electron-py, 2024-09-06:
Yes, but I don't know how to properly compile the firmware.

Electron-py, 2024-09-06:
Then yes, I would be happy to test it.

Electron-py, 2024-09-06:
It works!

Electron-py, 2024-09-06:
https://github.com/user-attachments/assets/abc821a3-cb52-4646-8dc5-331865353167

Electron-py, 2024-09-06:
Trying to compile the firmware on PlatformIO, but I need to know how to 'select' the layout.

RoboDurden, 2024-09-06:
6s online
Now it would be nice if you would test the new platformIO support.

Simply open the folder of the project (that contains the platformio.ini file).
Then hit strg+b to compile
and if this succeeds hit strg+u to upload.

Electron-py, 2024-09-06:
Ok, how do I select Gen2.1.1?

RoboDurden, 2024-09-06:
There currently is only one target in the platformio.ini file:
```

[env:genericGD32F130C8]
board = genericGD32F130C8
framework = spl
build_flags = -D GD32F130
			  -D TARGET=1
```

And in config.h the LAYOUT 1 is already selected:

```
	#ifdef GD32F130		// TARGET = 1
		#define LAYOUT 1

```

RoboDurden, 2024-09-06:
for your setup you need

```
	#define BAT_CELLS         	6        // battery number of cells. Normal Hoverboard battery: 10s

	#if defined(MASTER) || defined(SINGLE)
		#define MASTER_OR_SINGLE
		
		//#define REMOTE_DUMMY
		#define REMOTE_UART
		//#define REMOTE_UARTBUS	// ESP32 as master and multiple boards as multiple slaves ESP.tx-Hovers.rx and ESP.rx-Hovers.tx
		//#define REMOTE_CRSF		// https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/26

```

RoboDurden, 2024-09-06:
And in bldc.c you need to remove the __INLINE directive in line 74:

//----------------------------------------------------------------------------
// Block PWM calculation based on position
//----------------------------------------------------------------------------
void blockPWM(int pwm, int pwmPos, int *y, int *b, int *g)		// robo 2024/09/04: remove __INLINE for PlatformIO
{

Electron-py, 2024-09-06:
ok, trying now to do all

Electron-py, 2024-09-06:
Server error: {"type":"AttributeError","args":["module 'platformio.platform.gd32' has no attribute 'Gd32Platform'"],"message":"module 'platformio.platform.gd32' has no attribute 'Gd32Platform'"}

RoboDurden, 2024-09-06:
hm :-/ it does compile on my computer. But i had some other GD32 project installed a year ago.
And i forgot how to install the gd32 platform.
I thought that with internet connection, it would be downloaded automatically..

Electron-py, 2024-09-06:
AttributeError: Traceback (most recent call last):
  File "C:\Users\djtho\.platformio\penv\Lib\site-packages\platformio\__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ^^^^^
  File "C:\Users\djtho\.platformio\penv\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\djtho\.platformio\penv\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\djtho\.platformio\penv\Lib\site-packages\platformio\cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\djtho\.platformio\penv\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\djtho\.platformio\penv\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\djtho\.platformio\penv\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\djtho\.platformio\penv\Lib\site-packages\click\decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\djtho\.platformio\penv\Lib\site-packages\platformio\run\cli.py", line 145, in cli
    process_env(
  File "C:\Users\djtho\.platformio\penv\Lib\site-packages\platformio\run\cli.py", line 208, in process_env
    ).process()
      ^^^^^^^^^
  File "C:\Users\djtho\.platformio\penv\Lib\site-packages\platformio\run\processor.py", line 81, in process
    install_project_env_dependencies(
  File "C:\Users\djtho\.platformio\penv\Lib\site-packages\platformio\package\commands\install.py", line 132, in install_project_env_dependencies
    _install_project_env_platform(project_env, options),
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\djtho\.platformio\penv\Lib\site-packages\platformio\package\commands\install.py", line 149, in _install_project_env_platform
    PlatformPackageManager().install(
  File "C:\Users\djtho\.platformio\penv\Lib\site-packages\platformio\package\manager\platform.py", line 52, in install    
    p = PlatformFactory.new(pkg)
        ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\djtho\.platformio\penv\Lib\site-packages\platformio\platform\factory.py", line 79, in new
    platform_cls = getattr(
                   ^^^^^^^^
AttributeError: module 'platformio.platform.gd32' has no attribute 'Gd32Platform'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

Now i'm trying to update PlatformIO because i didn't use it for some time.

RoboDurden, 2024-09-06:
always google error messages:

https://github.com/platformio/platformio-home/issues/42
https://github.com/platformio/platformio-home/issues/1859

Electron-py, 2024-09-06:
For the configuration file i found on the PlatformIO wiki to put this [env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8

Electron-py, 2024-09-06:
trying now

RoboDurden, 2024-09-06:
Yes, using the original stm32f103 core might be better anyway.
Still the GD32F103 works for me:

```
RAM:   [          ]   4.1% (used 332 bytes from 8192 bytes)
Flash: [==        ]  22.1% (used 14500 bytes from 65536 bytes)
Building .pio\build\genericGD32F130C8\firmware.bin
=========================================================================================================== [SUCCESS] Took 2.87 seconds ===========================================================================================================
```

But i have not yet the time to test it on some test hardware.
And the nice guy who added the platformio.ini file mentioned that the cpu clock might be wrongly initialized.

RoboDurden, 2024-09-06:
Might be better to use a 2A constant current power supply when testing new binaries !!!

Electron-py, 2024-09-06:
Don't have one

RoboDurden, 2024-09-06:
You can use the original charger for testing.

But as the code already works for you when compiled with the Keil IDE, it is not such a big risk to go on with battery.
But if the platformIO core does the mosfet activation wrong, you might ignite a short cut..

Electron-py, 2024-09-06:
Last thing, it gives me an error: board doesn't support spl framework! I think it is for wrong clock configuration. How do i set it to HSI?

Electron-py, 2024-09-06:
ok, setted the clock succesfully

Electron-py, 2024-09-06:
where is the config JSON file? I need to put "extra_flags": "-DSTM32F100xB -DSTM32F10X_MD", apparently

RoboDurden, 2024-09-06:
platformio.ini

Electron-py, 2024-09-06:
Still same error .Error: This board doesn't support spl framework!
0

Electron-py, 2024-09-06:
Tried to search online but it doesn't work

RoboDurden, 2024-09-06:
please continue in the PlatformIO issue: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/93
Maybe @hoverboardhavoc can help

closed 2025-05-05 by Electron-py