RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #95: MM32F031C6T6 and LKS32MC

MannyCuadros, 2024-09-16:
Hello, i hope this comment can be read soon.
I have adquired 2 hoverboard and i'm trying control it by my self. In tutorials I read that their motherboard firmware must be changed, in this case, both are gen 2 motherboard. So first, y tried to know wich one is the master side and wich is tha slave. Assuming that I'm using the correct side, I have connected it to an ST-LINK V2, and then this one plugged in to USB port. Then, using SMT32CUBEPromgrammer sofwtare in windows, I tried to connect the board, but It continually sends me an error (as if the board was not connected).
I don't know if I am using the correct programm (or if there is a way to do it in Ubuntu), maybe i have did something wrong from the beginning, if connection its ok, i really don't. I Noticed that the microcontroller of one hoverboard is and MM32F031C6T6 and the other one is and LKS32MC. This are the boards:

- MM32F031C6T6 side 1
[image]

- MM32F031C6T6 side 2
[image]

- MM32F031C6T6 microcontroller
[image]

- LKS32MC side 1
[image]

- LKS32MC side 2
[image]

- LKS32MC microcontroller
[image]

Thank you for your time first of all

RoboDurden, 2024-09-17:
Maybe @I-hate-2FA  can add MM32F031 support to his MM32SPIN0X/2X firmware https://github.com/AILIFE4798/Hoverboard-Firmware-Hack-Gen2.x-MM32
It is also a M0 core with 32 kB: 
https://crossic.com/wp-content/uploads/2021/12/DS_MM32F031xx_EN.pdf
https://wmsc.lcsc.com/wmsc/upload/file/pdf/v2/lcsc/2304140030_MindMotion-Microelectronics-MM32F031C6T6_C316709.pdf

I think there is no firmware at all yet for LKS32.

RoboDurden, 2024-09-17:
Please upload photos of the complete hoverboards as well as links where to buy them - so others can avoid buying these hoverboards :-/

RoboDurden, 2024-09-17:
MM32F031  only has one UART port, so easier to connect to.

MannyCuadros, 2024-09-17:
Thanks for replying.

Update.
I traced the pins on the MM32 microcontroller and realized that the configuration I had considered was wrong. This is the pin configuration I have now connected to the ST-LINK

[image]

I tried to use STM32CubeProgrammer connecting the ST-LINK to the USB port, which I do with the 3.3V cable (red VDD) disconnected due to a recommendation, and new questions arose: What would happen if I connect the 3.3V cable to the ST-LINK and the board while it is working? What if I test it with the cable connected but with the battery disconnected? When pressing "connect" in the program, is it always necessary to press the power button?
Anyway, when pressing "connect" it gave me some errors, which I don't know what they are due to, so I'm asking if there is something I can do to correct it.
[image]

> Maybe @I-hate-2FA can add MM32F031 support to his MM32SPIN0X/2X firmware https://github.com/AILIFE4798/Hoverboard-Firmware-Hack-Gen2.x-MM32 It is also a M0 core with 32 kB: [...]

So do I have to discard the hoverboard that uses the board with LKS32?

> MM32F031 only has one UART port, so easier to connect to.

How do I use UART port to control the motors by myself?, for example, using an Arduino or ESP32? 

> Please upload photos of the complete hoverboards as well as links where to buy them - so others can avoid buying these hoverboards :-/

The motherboards that contain those microcontroller are
- MM3232F031 -> DBK 4.3 control board
- LKS32 -> only found that says SLHT C270177

Which motherboard would you recommend so I can change the firmware and control it by myself? (Only the motherboard without any other components like wheels or LEDs)

I-hate-2FA, 2024-09-18:
> I tried to use STM32CubeProgrammer connecting the ST-LINK to the USB port

did you even read the wiki?

> So do I have to discard the hoverboard that uses the board with LKS32?

yes, when a mm32spin05 board is just $2 theres no reason to waste time on it, unless you are a enthuiast and want to play with new micro controllers, the lks32 compiler is availiable, so it would be possible to develop a firmware for it, its just not worth the effort, but contribution to the repo is welcome

> How do I use UART port to control the motors by myself?, for example, using an Arduino or ESP32?

there is currently no proper documentation of the remote UARTBUS protocol, but you can reverse engineer it from the example code

> The motherboards that contain those microcontroller are
>    MM3232F031 -> DBK 4.3 control board
>    LKS32 -> only found that says SLHT C270177

if you can, please join discord server and submit in the database

> Which motherboard would you recommend so I can change the firmware and control it by myself? (Only the motherboard without any other components like wheels or LEDs)

mm32spin05 board, because that board is tested, and very cheap, only $2

MarcoFranchi, 2024-09-28:
@I-hate-2FA where can I buy the mm32spin05 board for $2?

RoboDurden, 2024-09-28:
I think it is only $2 on taobao and it is difficult to buy there from outside China.
I added the aliexpess offer to the main page of this repo, but it does not offer shipping to Europe. And to the US it is rather $10 including shipping. And this MM32 board has no opamp for phase currents, so there will be no FOC. But the sinusodial firmware of @I-hate-2FA is already a big improvement over the simple block commutation of the GD32 firmware i offer here.

MannyCuadros, 2024-09-30:
Of all the gen 2 motherboards you have tested (STM32, GD32, MM32), which one do you recommend purchasing? And if you have the purchase links I would greatly appreciate it.

> I think it is only $2 on taobao and it is difficult to buy there from outside China. I added the aliexpess offer to the main page of this repo, but it does not offer shipping to Europe. And to the [...]

I am buying this one, but it will be shipped un 2 months.

I-hate-2FA, 2024-10-01:
I will suggest mm32, because I'm the mm32 dev

Stm32 is possible to be used now from what I heard, it's not tried yet, but you could be the one, and stm board always have the foc capability

MannyCuadros, 2024-12-06:
Finally, after a long wait, the mm32spin05 boards that you recommended me arrived

[image]
[image]

Now I will have to follow the steps indicated in the main source.
I would like to earnestly ask you, if you could please guide me, in addition to what appears in the Wiki, to do it correctly, and to avoid errors that could cause the board to fail. I would not like to wait so long again to acquire a new board.

Thank you very much in advance