RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #98: Open hardfware FOC driver

trondin, 2024-09-29:
My next idea is to design an open FOC motor driver.
(The target is to waste a time.) --- strikethrough.

The target is to design a very cheap driver that will have all the necessary features for BLDC FOC. The interfaces will be UART and CAN-bus, and the schematic will be mainly the same as the M365.

I plan to use EFery firmware as the first option, which I can implement easily.

What do you think? Can something be cheaper or better?

I am going to order parts now from AliExpress and a board a few days later. Anyway, the board will arrive in 2-3 weeks and the components in 6-8 weeks.

[image]
[image]
[image]

RoboDurden, 2024-09-29:
Many people still buy used hoverboard for 30€ including shipping and mostly get working mainboard and two motors. Sometimes the battery is still okay.
But with all the many Gen2 boards and the Atery Gen1 board, chances get lower and lower with new Gen2 boards popping up from China.

So i personally think you coud help the hoverboard community more if you make FOC firmwares also for GD32 and LKS32..

But if you can let your board be made at less the $10 inkl. shipping, i will order 5 pcs when you make a Kickstarter.

What mcu do you plan to use ? It should be a 64 kB version, not 32 kB. 
Candas and I made a SimpleFOC firmware for Gen2 boards and I2C control was working.. But with the SimpleFOC library it became nearly impossible to stay below 32 kB.

Brain degenerated consumers always ask for more speed. So an option for higher voltage then 12s (50V) would be nice. Maybe 72V.

And might add double TO220 holes to solder a second mosfet on top of the exiting six mosfets. maybe with an alauminium plate in between.

So that your board could power 2000 Watt (50+A) motors like these ones:

https://www.aliexpress.com/item/1005005919479947.html
https://www.aliexpress.com/item/1005005919479947.html
https://www.aliexpress.com/item/1005003258321477.html
https://www.aliexpress.com/item/1005006176433721.html

You should post your design to the simpleFOC community to get valuable feedback: https://community.simplefoc.com/

But i do not really like your new project. Always these linux nerds that waste all their time to do everything from scratch anew.

A platformIO EFeru FOC GD32 port would help more than yet another board that won't be below $10 anyway.

RoboDurden, 2024-09-29:
The many male headers to the right side could have 2.54 mm spacing so a protoyping board with female headers dowside could be plugged on top of your board.
For that, the onboard leds should be somerwhere on the side to still be able to shine..

trondin, 2024-09-29:
I have about 6 hoverboards, the latest was 15€ :) An it is fully working, battery is not terrible.
All my hoverboards have same old style single main board.
My new board can support about 400W motor, target was a low components price, to have fun for minimum money (my disability pension not so big).
Unfortunately I don't have any boards with GD32 and LKS32 :)
Probably LKS32 is interesting, if it has opamps. But I am not able to find RM with full description.
I don't like simpleFOC, it is too universal and take lot of resources.
I didn't manage to get M365 board working smoothly with hoverboard motor, looks like problem with PI control.
I am not going to make a Kickstarter, I am to old for toys lake that.

RoboDurden, 2024-09-29:
max power is only a question of R_dc_on of the mosfet ?
As i said, i was able to 1000w bldc on normal Gen1 board (with a cooler venting the aluminium back plate).
These hoverboard mosfet are quite cheap. So doubling them would allow for 2x amps = 2000 Watt.
And the additional  TO220 header pins will not screw your board.

I think you should try to offer something new.

Try to make the dcdc stepdown 80V proof to allow for 72V batteries..
Brain degenerated consumers will love that.

trondin, 2024-09-29:
**Try to make the dcdc stepdown 80V proof to allow for 72V batteries**
They are exist and chip cost is 0.2...0.6USD. Vin up to 800V :) Good example - PN9015.
They are very popular in crazy smart home.

trondin, 2024-09-29:
[image]

trondin, 2024-09-30:
2...4kW driver is doable, but TO220 is too small, should be used 100V 200A 2uOhm nMOSFETs (TO247 or D2PAK) and more powerful gate driver to charge big Qg fast. Price will be over 10€ :)

trondin, 2024-11-15:
Finally it works. Control - UART or CAN-bus.
[image]

RoboDurden, 2024-11-16:
I still don't know what to think of this.