RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #99: Board like 2.1.5 but MM32SPIN 05PF processor

cjoe-51, 2024-09-30:
Seems like totally same board as the 2.1.5 however the processor is differen t - its MM32SPIN 05PF
Here board A:
[image]

cjoe-51, 2024-09-30:
and here board B:
[image]

cjoe-51, 2024-09-30:
I have whole hoverboard so I can identify/confirm the connections/cables. On the other hand Im not in a hurry to delete the present firmware :)

I-hate-2FA, 2024-09-30:
the board is foc capable, you do not need to trace the board, the firmware do for you, if you need support please join discord group and submit in database

RoboDurden, 2024-09-30:
might be nice if you also post photos of the complete hoverboard and maybe a link where to buy.
So other know will be inside.

This looks like a dual OpAmp for phase currents B and C:
[image]

So this EFeru FOC port from @trondin might work on this board as well: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/96

But the pinFinder firmware from https://gitlab.com/ailife8881/Hoverboard-Firmware-Hack-Gen2.x-MM32 can confirm the correct pin definitions. 
@I-hate-2FA does your pinFinder detects phase current pins ?

I-hate-2FA, 2024-09-30:
no, it uses different algorithm then your

I-hate-2FA, 2024-09-30:
it is possible to add, but i do not have foc capable board

RoboDurden, 2024-09-30:
I did find the phase current pins with my RemoteAutodetect by pure chance when try to detect the hall sensors. The phase currents show the same frequency..

I-hate-2FA, 2024-09-30:
you can check my code, i dont read the hall like that, thats why my detection is so fast and work on all wheel and board, i try to move sensorless to all 6 position, if a pin do not match expection it is removed from the list, i do it with dynamic motor speed so all board can detect normally, it never had any issue, never false detect

I-hate-2FA, 2024-09-30:
[image]
this is only time tool report self check error, but in the end it is board hardware designed wrong, my tool is correct

I-hate-2FA, 2024-09-30:
there is actually quite some boards designed wrong, in that case you will ned to change the wriring, it is the downside to support sine wave, the hall pin need to be pinmodeAF not normal gpio

RoboDurden, 2024-09-30:
I do not understand your last two posts. 
What can be wrong with three inputs to read the hall sensors ?
If pinmodeAF, what alternate function is needed for the three hall sensors ??

I-hate-2FA, 2024-09-30:
in mm32, the hall sensor need to be handled by timer2 or timer3 input 1,2,3  there is internal debounce circuit and counter for the sine wave function

I-hate-2FA, 2024-09-30:
i do not know what will happen if you used a wrongly designed board, i dont think the code will work correctly

RoboDurden, 2024-09-30:
You mean that your sinusodial bldc code needs hall pins that have AF alternate function to trigger TIM2 or TIM3 so that the exact time since new hall position can be calculated somewhere like in hallhandle.c::91

`		u->HallTime[i] = halltimen ? TIM2->CCR1 : TIM3->CCR1;	 
`

So a "wrongly designed board" uses hall gpio which have no AF to trigger an onChange event for TIM2 or TIM3 ?

These are the hall defines of 2.1.5 which has the same PCB as this 2.4.5 : 

```
// Hall sensor defines
#define HALL_A PB0	// Batman313v
#define HALL_B PB5	// Batman313v
#define HALL_C PB4	// Batman313v
```

Is that okay with your firmware ?

I-hate-2FA, 2024-09-30:
[image]
[image]
[image]
these all work they are correctly designed board

I-hate-2FA, 2024-09-30:
> So a "wrongly designed board" uses hall gpio which have no AF to trigger an onChange event for TIM2 or TIM3 ?

yes

I-hate-2FA, 2024-09-30:
the pinfinder have a database of all combination, just run it, it will tell you

cjoe-51, 2024-10-03:
@RoboDurden - Im not yet prepared to kill the working Hoverboard, probably later. There is however a serial port by which the two boards communicate together. Did somebody analyzed this communication? is there a possibility to send commands and/or receive info to/from each board? Have you met such a project?
(I was not sure where to ask this, so please feel free to remove this question from here if its to much OT)

RoboDurden, 2024-10-03:
not that i have heard of. Yes, the slave board might be controlled if the protocol would be hacked. And it might contain the IMU data. I think @I-hate-2FA once had a shopping link for a very cheap GD32F130 slave board..

cjoe-51, 2024-10-03:
By hacking the communication I shall not destroy the device so I already have a board for potential experiments. 
So far I found only one guy who partially decoded the communication, however it seems he gave it up. 
Will search further.