RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #100: REMOTE_UARTBUS speed

hoverboardhavoc, 2024-11-03:
Hi, I had a look at [Esp32_PPM.ino](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-GD32/blob/main/Arduino%20Examples/Esp32_PPM/Esp32_PPM.ino) Does the UART bus only update every 3 seconds? If so, is that because it's multicast and there is no way to acknowledge the transmissions? I think 3 seconds might be a bit slow to change speed!

I-hate-2FA, 2024-11-03:
it is possible to update at least every 200ms acording to uartbus protocol

RoboDurden, 2024-11-03:
???

3000ms is only the cycling the hoverboard LEDs.

UartBus is sending at 100 Ms:

  ```
#ifdef REMOTE_UARTBUS
    if (iNow > iNext)
    {
      iNext = iNow + 100;
```

hoverboardhavoc, 2024-11-03:
Ahh, of course! Thanks for the help