RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #101: Gen2.1.1 - TestSpeed

YujiKF, 2024-11-12:
Hi! I'm using a Gen2.1.1 board and trying to use the [TestSpeed.ino](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-GD32/tree/main/Arduino%20Examples/TestSpeed) but I'm not sure if I'm doing this right.

I flahsed the pre compilated binarie [hoverboard 2.1.1 master Uart.bin](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesReadyToFlash/hoverboard%202.1.1%20master%20Uart.bin) to the master board and [hoverboard 2.1.1 slave.bin](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesReadyToFlash/hoverboard%202.1.1%20slave.bin) to the slave board. After that, I uploaded the TestSpeed.ino sucessfully to an ESP32.  (uncomment the line 24 was necessary as I'm using a ESP32)

But I only see this on Serial Monitor:
[image]

I saw a similar issue, where the problem was the define.h, since I'm using a pre compilated binarie to my board, I don't understand what I'm doing wrong. What should I do?

OBS: I tried to connect the slave and master board and connect the pin16 from ESP to PB7 and pin17 to PB6 (but I know this is wrong, it was just a test), and tried to connect the pin16 from ESP to PA3 and pin17 to PA2 but it didn't worked, and this way I can't connect master and slave. Switching pin17 and pin16 didn't worked as well.

RoboDurden, 2024-11-13:
First install https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesReadyToFlash/hoverboard%202.1.1%20master%20Dummy.bin on the master to verify that both hoverboard motors are spinning backward and forward. 

With RemoteUart the esp is waiting for incoming data from the master before it starts sending speed steer commands. 
So check the tx rx line first with that binary.

YujiKF, 2024-11-14:
Thanks for the answer! Just tested it and it worked well, both hoverboard motors are spinning forward and backward. However, the Serial Monitor doesn't change, even when the ESP is not connected to the hoverboard board the Serial Monitor remains the same.

RoboDurden, 2024-11-14:
Connect a led with resistor to the tx line of the hoverboard to verify that it is sending data.
If you don't have a dso yet, This is the best value for little money I think: https://a.aliexpress.com/_EJTmdFF

I could recompile the master binary..

RoboDurden, 2024-11-14:
RemoteDummy does not send serial data. You have to flash RemoteUart again.

YujiKF, 2024-11-14:
I have a oscilloscope! This is what i see in remote tx line of the hoverboard:
[image]

I'm strugling with the oscilloscope trigger but I think that there's no problem on tx line. Sucessfully flashed RemoteUart to the ESP and master uart to the master hoverboard and there's no difference, not sure if I'm doing something wrong...

RoboDurden, 2024-11-14:
This looks like valid serial data. So the problem is one the esp side.
Check with a USB uart dongle with 19200 baud.
There should be some incoming carnage. The hoverboard is sending a binary struct..

I-hate-2FA, 2024-11-20:
theres always a usb uart dongle included with every development board unless its some super cheap ones

RoboDurden, 2024-11-20:
my autodetect.ino acts as a usb uart dongle by passing incoming hoverboard data to the arduino serial monitor: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-GD32/blob/main/Arduino%20Examples/Autodetect/Autodetect.ino

YujiKF, 2024-11-21:
Okay, thank you! How do I use it? Just connect ESP32 pins in the hoverboard? Is it necessary to use a resistor or diode? I just killed an ESP32 trying to use the TestSpeed, not sure what happened...

RoboDurden, 2024-11-21:
First rule of project Mayhem: you do not ask questions.

I-hate-2FA, 2024-11-21:
esp32 is 5v tollerant but keep it away from 15v

CCALLEJAR, 2025-04-10:
YujiKF, did you solve the problem? I'm in the same situation as you, but I can't see anything on the serial number. It could be because I'm using an esp32 instead of a Wemos S2 Mini. Any help, please!

YujiKF, 2025-05-20:
> YujiKF, did you solve the problem? I'm in the same situation as you, but I can't see anything on the serial number. It could be because I'm using an esp32 instead of a Wemos S2 Mini. Any help, please!

Hi! Sorry for the long wait. Yes, I solved my problem, in my situation, it was just TX (ESP32) --- RX (Hoverboard) connection that had a problem, I changed the cable, and it worked fine. However, I made some code that may help you, it's [here](https://github.com/YujiKF/SerialHoverCR), it's totally based on the original TestSpeed. With this, you can read the hoverboard's serial data and see if everything is working correctly, let me know if you need more help!