RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #102: gd32F removing read protection

Nassimraz, 2024-12-18:
after removing read protection ( without holding the power button ) the board wont turn on again.
mcu  release the power latch and switch itself off.? help please

RoboDurden, 2024-12-18:
GD32f130 ?
Maybe you need to install a new binary because removing read protection did erase the original firmware ?
It needs a working firmware to keep the power on after a short press on the on button, which gives 3.3V to the mcu so the code can start.

Nassimraz, 2024-12-18:
i did complete chip erase and installed a new bin successfully(had to hold the power button ), but still wont turn on (vbat only shows 3.3v while holding power). i think the power latch circuit is somewhere on the board ?

Nassimraz, 2024-12-18:
this guy: 2.1.6 (todo)
[image]

RoboDurden, 2024-12-18:
so you did compile your 2.1.6 dummy firmware, yes ?
in config_2-1-6.h i read

`// todo: BUTTON not working but hiliving is sure to have traced the pin correctly with the help of ailife
`

So in the bginning you might need to bridge the on-button to keep the board powered.

If the motors are then spinning forward and backwards with the RemoteDummy, you can already be happy..

Good night from Germany.

RoboDurden, 2024-12-18:
Watch my autodetect youtube tutorial. You could retry the autodetect to get the board alife..