RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #103: Connecting board to ESP32(or similar)

lUAlSvail, 2024-12-19:
I was trying to follow the wiki but I could not find how to control the board after it was flashed. 
I flashed the board as per procedure in wiki:

- Used STM utility to flash the board. Master with 'hoverboard 2.1.1 master Uart.bin' and slave with 'hoverboard 2.1.1 slave.bin'. As I understand this is what I should use in combination with ESP32 or similar.(correct me here if I am wrong please)

But after this step how would I connect boards to my ESP32?
Do I need to connect both master and slave boards to ESP32 or just master will be enough(as master and slave connected)?
Should I use  REMOTE  G T R 5v  connection to do that?
Is there any sample code to use with ESP32 to test if everything connected properly and I go move from there?

Photos below is my board, and I assumed it is 2.1.1 version
[image]
[image]

RoboDurden, 2024-12-19:
Beginners should always begin with https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesReadyToFlash/hoverboard%202.1.1%20master%20Dummy.bin for the master board

Then flash https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesReadyToFlash/hoverboard%202.1.1%20slave.bin to the slave and connect the original master-slave uart cable.

Finally replace the master dummy with https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesReadyToFlash/hoverboard%202.1.1%20master%20Uart.bin

The other possibility is controlling both master and slave directly from esp32 with 

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki/Control-Methods#Remote-UART-Bus

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesReadyToFlash/hoverboard%202.1.1%20single%20UartBus%20id0.bin
and
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesReadyToFlash/hoverboard%202.1.1%20single%20UartBus%20id1.bin

lUAlSvail, 2024-12-20:
Thank you for replying!
I did follow process you described above, as well as tried to use example code for esp32 but I think something is wrong, after I connect ESP32 to the board(with already uploaded code TestSpeed only thing changed is pins as it is 16 and 17 for my esp) and click power button , I can see some weird symbols pop up in the terminal, so I assume I did something wrong.  Than I changed #define BAUDRATE 115200  from 19200 and that gibberish disappeared. But nothing else is appearing. I tried to modify code, to just go forward , but it did not change anything. Any advice on what I can start debug from?
[image]

RoboDurden, 2024-12-20:
first, tell me that the dummy fimrware does make at least the master board motor spin forward and backwards.

You imported the arduino code into platformIO. Should work, but i have not tested that yet.

hoverserial should speak with 19200 baud to the hoverboard, but with 115200 baud to the arduino console / putty / platformIO .

So i do not understand what you say about the console output.

lUAlSvail, 2024-12-20:
When I flashed master with dummy firmware and slave with 'hoverboard 2.1.1 slave.bin', after I pressed on button on the board nothing is happening. I can here sound of turning on, but I can not turn it off, only disconnecting the battery

lUAlSvail, 2024-12-20:
By nothing happening I mean wheels not spinning, but I can hear fans(so I assume board is powered)

RoboDurden, 2024-12-20:
You have to bridge the on-button to keep the board powered so motor can spin.
the startup melody is already a good sign.
Maybe the board is broken..
Flash the single-dummy firmware on both the master and the slave board. Connect them with the master-slave cable and bridge the on-button. Then the slave board should also startup and if the slave board motor is spinning but the master motor not, then there is something wrong with the master board.

You could also flash the autodetect firmware and see if this is capable of making the motor spin. Even without the serial communication, the motor should spin for half a second on startup.

https://www.youtube.com/watch?v=X7bpMX8tCE8

lUAlSvail, 2024-12-20:
After flashing with both boards with [hoverboard 2.1.1 single Dummy.bin](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesReadyToFlash/hoverboard%202.1.1%20single%20Dummy.bin). I can see board turn on, with different sound, and after clicking poweroff it powers off. But nothing else is happening. 

https://github.com/user-attachments/assets/dd961b64-f047-4b08-a0f1-c72e1e34115c

I will try autodetect firmware and update with the results.

lUAlSvail, 2024-12-20:
I also tried [hoverboard 2.1.1 master Uart 7s.bin](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesReadyToFlash/hoverboard%202.1.1%20master%20Uart%207s.bin) on master and [hoverboard 2.1.1 slave.bin](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesReadyToFlash/hoverboard%202.1.1%20slave.bin)  on slave and now when I click turn on it turns on, and when click turn off it turns off, but nothing else happening(wheels are not spinning).
Is my understanding correct that after flashing with single-dummy firmware on both the master and the slave board and not doing anything else, after clicking turn on button, wheels should have spin? 
Unfortunately I am unable to see what is the chipname as it looks like inks are erased so  I will be unable to flash with autodetect , as I do not know which one. I can try all of them , but should I flash both master and slave with the same autodetect?

lUAlSvail, 2024-12-20:
This is what I see now with serial monitor  [hoverboard 2.1.1 master Uart 7s.bin](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesReadyToFlash/hoverboard%202.1.1%20master%20Uart%207s.bin) on master and [hoverboard 2.1.1 slave.bin](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesReadyToFlash/hoverboard%202.1.1%20slave.bin) but still no wheels movement

[image]

lUAlSvail, 2024-12-20:
This is pretty much what I got with r [hoverboard 2.1.1 master Uart 7s.bin](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesReadyToFlash/hoverboard%202.1.1%20master%20Uart%207s.bin) on master and [hoverboard 2.1.1 slave.bin](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesReadyToFlash/hoverboard%202.1.1%20slave.bin). So based on the lights flashing, and receiving telemetry I can assume that I connected everything properly?
Also before flashing I made sure that hoverboard was working, so battery was good and wheel was turning when I was standing on a hoverboard.
So at this point I am confused what it can be
https://github.com/user-attachments/assets/b526aa25-90cc-4998-9d1a-bd4e2db307d0

lUAlSvail, 2024-12-21:
I was able to get number on the slave board and it looks like it indeed 2.1.1( not sure if it is the same number on the master as it looks erased.)
As per your advice I flash the autodetect firmware [hoverboard autodetect GD32F130 tx=PB6, rx=PB7.bin](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesToTest/hoverboard%20autodetect%20GD32F130%20tx%3DPB6%2C%20rx%3DPB7.bin) to both master and slave. After turning on  the board , master wheel turned(which I hope is a good sign), slave did not(once again both wheels was turning when I had board with original firmware). 

Sorry for 100 messages in this thread, but I was trying to get as much information as I can, to make some sense of it...

lUAlSvail, 2024-12-21:
[image]
GD32F130C8

RoboDurden, 2024-12-21:
> Is my understanding correct that after flashing with single-dummy firmware on both the master and the slave board and not doing anything else, after clicking turn on button, wheels should have spin?

yes.

> This is pretty much what I got with r [hoverboard 2.1.1 master Uart [...]

Good to see that you have the uart communication working.
Does the iOdomL and iOdomR change values when you rotate a motor by hand ?

The startup melody in your video does not really sound okay.
And the following beeping should only occur when the motor is spinning backwards.
But in the serial feedback log you can see the iSpeed starts from 0 going positiv, so the motors should spin forwards (which they don't) and no beeping sound shoud be heard.

As the hoverboard is communicating correctly with esp32, the firmware is running nicely.

As the motors do rotate with the autodetect firmware, i suggest you install autodetect.ino on the esp32 and follow the different stages of autodetect. Then we might see at which point the autodetect is failing.

Warning: autodetect should only be used with a 2A constant power supply.

Yes, you need the 7s = 24V binaries if you only connect a 24V battery. Otherwise the board will shut down because of low battery.

> As per your advice I flash the autodetect firmware [hoverboard autodetect GD32F130 tx=PB6, [...]

autodetect is only for one board at a time. But it should also work for the slave board. But the slave board might need the 5V of the master-slave serial connection to power up. With autodetect, first focus on the master board and see if you get the autodetect.ino user interface running..

You might want to install Keil V5 ide to compile your own firmwares.
But as you already have PlatformIO running, i think by now, this firmware here can also be compiled with PlatformIO.

lUAlSvail, 2024-12-21:
Does the iOdomL and iOdomR change values when you rotate a motor by hand ?
Yes IOdomL and iOdomR were changing when I was turning the wheels

Warning: autodetect should only be used with a 2A constant power supply.
I am not sure what do you mean by that, does fully charged hoverboard battery will do the job or I need some special tooling?

Also it is looks like even that I have similar board to 2.1.1 it is not exactly the same, I have some additional chip on the top(which 2.1.1 does not have) as well as white connectors also have more sockets (5 vs 3). Does this mean this is one of the reason why 2.1.1 is not working for me?

Could you please explain how should I use autodetect.ino. Do I need to first flash the master with [hoverboard autodetect GD32F130 tx=PB6, rx=PB7.bin](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesToTest/hoverboard%20autodetect%20GD32F130%20tx%3DPB6%2C%20rx%3DPB7.bin) and than connect esp32 with autodetect.ino uploaded to it? 

Sorry, I am pretty much noob with this. Thank you a lot for answering all the questions.

RoboDurden, 2024-12-21:
if iOdom works then the three hall sensor inputs are correct.

2A constant current power supply will not burn your mosfets if the autodetect firmware is accidentaly making a short cut. The battery would kill the mosfest in such a case.

As you have already fiddled around with the battery as the power source, it is unlikely that this will happen.

I only had a quick look at your photos and it look as a 2.1.1 board to me.
But yes, if the board is a new layout, then this could explain your bad luck. And the autodetect firmware might find the changes.

But button, hold, buzzer and hall inputs already seem to be correct, and the six mosfets outputs normally are so restricted by hardware that most of the GD32F130 have the same 6 pins.

But give my autodetect firmware a try: https://www.youtube.com/watch?v=X7bpMX8tCE8

If you have a little oscilloscope, it would help to check some of the six mosfet outputs to see if the pwm is looking good.
This one is nice: https://de.aliexpress.com/item/1005006890359302.html

RoboDurden, 2024-12-21:
No, your board is 2.1.1 for sure. You merely have a bluetooth module soledered onto the board and the led output pins have different white headers because your hoverboard has different led modules.

You said, that autodetect mades your master board motor spinning. So i suggest you continue with my youtube tutorial.

lUAlSvail, 2024-12-21:
[image]
When I send like 0 or 1 nothing happening, am I doing something wrong?
both boards flashed with [hoverboard autodetect GD32F130 tx=PB6, rx=PB7.bin](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesToTest/hoverboard%20autodetect%20GD32F130%20tx%3DPB6%2C%20rx%3DPB7.bin)

RoboDurden, 2024-12-21:
maybe your platformIO serial monitor is not good for my autodetect user interface.
Sending ENTER to the hoverboard should advance the code to the main menu.

try putty or the serial monitor from the Arduino IDE.

lUAlSvail, 2024-12-21:
Sorry, maybe I miss understood you, but the screenshot above is from Arduino IDE.

RoboDurden, 2024-12-21:
Indeed arduino ide. Then i would check your esp32-tx to hover-rx line.
But With RemoteUart you did see iSpeed increase so the hoverboard did receive the command from esp hoverserial.

try hitting enter several times.
try Putty.

any byte received should forward to the main menu:

```
	switch (wStage)
	{
	case AUTODETECT_Stage_Startup:
		if (!(buzzerTimer%16000) )	// 16 kHz
		{
			if (0 == iRepeat--)
			{
				iRepeat = 20;
				sprintf(sMessage,":-)\n");
			}
			else	sprintf(sMessage,"\r:-)");
		}
		if (cCmd)
			AutoDetectNextStage();
		return;
```

If you have a usb-uart dongle, try to send and receive data at 19200 baud directly from your computer to hoverboard.

lUAlSvail, 2024-12-21:
That is interesting that you said that, as even when I disconnect tx cable from esp32 I still can see iSpeed changing.
How is that possible?

https://github.com/user-attachments/assets/517e85a1-a883-4598-a4e9-fdf03f20e852

RoboDurden, 2024-12-21:
i have no explanaition.
With RemoteUart, iSpeed can only be set from incoming serial command:

remoteUart.c :
```
			if (pData->checksum == CalcCRC(aReceiveBuffer, sizeof(SerialServer2Hover) - 2))	//  first bytes except crc
			{
				iTimeLastRx = millis();
				//DEBUG_LedSet(SET,0) // 		(steerCounter%2) < 1
					DEBUG_LedSet(SET,0)
				speed = pData->iSpeed;
				steer = pData->iSteer;
				wState = pData->wStateMaster;
				wStateSlave = pData->wStateSlave;
				//if (speed > 300) speed = 300;	else if (speed < -300) speed = -300;		// for testing this function

				ResetTimeout();	// Reset the pwm timout to avoid stopping motors
			}
```

But now i see that in hoverserial.ino , the millis and iSpeed is not coming from hoverboard feedback struct but is the realtime data sent to the hoverboard:

```
      DEBUGT("millis",iNow-iLast);
      DEBUGT("iSpeed",iSpeed);
      //DEBUGT("iSteer",iSteer);
      HoverLog(oHoverFeedback);
```

So now i guess there is something wrong with your tx-rx communication.

Maybe because of the bluetooth module which might also make use of PB7 !!

Try to connect to 

```
	#define USART1_TX		PA2
	#define USART1_RX		PA3
```

which is the master-slave header.
Be careful with this header, there might be 15V on one pin !

try https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesToTest/hoverboard%20autodetect%20GD32F130%20tx%3DPA2%2C%20rx%3DPA3.bin

If now the tx rx communication works,
you will need RemoteUartBus for PA2/PA3 and control each motor directly from esp32. 
I can compile and upload such binaries for 7s and ids #1 and #2 - if you can not compile by yourself.

But first try the autodetect binary for PA2/PA3 on the master board

lUAlSvail, 2024-12-21:
Just to be clear those iSpeed without connecting tx I was getting with [hoverboard 2.1.1 master Uart 7s.bin](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesReadyToFlash/hoverboard%202.1.1%20master%20Uart%207s.bin) on master and [hoverboard 2.1.1 slave.bin](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesReadyToFlash/hoverboard%202.1.1%20slave.bin)

RoboDurden, 2024-12-21:
as i said, this iSpeed is not the confirmation of the hoverboard but only the speed command sent by esp32. Which i guess never reaches the hoverboard because of the bluetooth module which also sends data to the PB7 pin, i guess.

These bluetooth modules usualy do only receive data from the smartphone. So you could programm the esp32 bluetooth serial to send speed/steer to the hoverboard but not receive data from the hoverboard..

lUAlSvail, 2024-12-21:
used https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesToTest/hoverboard%20autodetect%20GD32F130%20tx%3DPA2%2C%20rx%3DPA3.bin on master.
And connected txrxgrnd where master slave connection was. Was able to send 0, 1,2 etc. As i was holding powerbutton, was unable to complete all of them.

hall and phase currents: 'r'=restart	's'=save
try 0 = C13	
try 1 = C14	HALL
try 2 = C15	
try 3 = F0	
try 4 = F1	HALL
try 5 = A0	
try 6 = B11	HALL
try 7 = A4	
try 8 = A5	
try 9 = A6	
try 10 = A7	
try 11 = B0	HALL
try 12 = B1	PHASE

#define HALL_A		PC14
#define HALL_B		PF1
#define HALL_C		PB11
#define PHASE_A		PB0
#define PHASE_B		PB1
//#define PHASE_C		P??

//#define LED_RED		P??
//#define LED_ORANGE		P??
//#define LED_GREEN		P??
//#define UPPER_LED		P??
//#define LOWER_LED		P??
//#define ONBOARD_LED		P??
//#define BUZZER		P??

So you was right about that bluetooth module?

lUAlSvail, 2024-12-22:
Could you please compile and upload such binaries for 7s and ids https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/1 and https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/2. And explain how I should connect everything to esp and what code should I upload to esp to control it.

RoboDurden, 2024-12-22:
You can test with a beeping diode tester (multimeter) that there is a connection from the PB7 rx pin to a pin of the bluetooth module.
Then you should also check if the PB6 tx pin does also connect to the bluetooth module.

Then you might be able to establish a fully bidirection bluetooth-serial connection from esp32 to hoverboard,
and could keep the pa2 pa3 serial header for the master slave connection.

does the motor spins smoothly after the the hall pin detection succeeded ?

I have uploaded `hoverboard 2.1.1 tx=PA2, rx=PA3 7s single UartBus id0.bin` and id1
First do only thest id0 on the master board without the slave.

your hall pin detection shows another ordering, but i have kept the original 2.1.1. order as your board is identical.

I have little time today.

lUAlSvail, 2024-12-22:
Unfortunately I do not have multimeter at the place where I am currently. But lets say it is connected, how would I achieve that?
Yes motor spins smoothly.
Thank you for the binaries!
How that my hall pins are different can impact the controlling the motors? Should I change that in the future.

Once again, thank you a lot for your help

RoboDurden, 2024-12-22:
https://randomnerdtutorials.com/esp32-bluetooth-classic-arduino-ide/

You would simply replace   `#define oSerialHover Serial1    // ESP32` with   `#define oSerialHover SerialBT` 
This might already do the job..

But normally these bluetooth modules do only receive data from smartphone and the PB6 tx pin will not be connected to the module.
But if your old smartphone app did show some data like battery capacity, then it might be possible that you have bidirectional bluetooth module.

If motor spins nicely, then hall sensors are fine.

lUAlSvail, 2024-12-22:
hoverboard 2.1.1 tx=PA2, rx=PA3 7s single UartBus id0.bin should go to master right? And I should use https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-GD32/blob/main/Arduino%20Examples/TestSpeed/TestSpeed.ino on esp32? Because that is what I did and I do not even receive the telemetry data now(

as per bluetooth, I bought it used for my project, and I did not even know there was a bluetooth

RoboDurden, 2024-12-22:
You have to uncomment this line in hoverserial to activate UartBus :

`//#define REMOTE_UARTBUS  // one serial bus to control them all :-)`

with uartbus there is no master and slave anymore. alle the boards are "single" now and the esp32 is the master.

lUAlSvail, 2024-12-22:
https://github.com/user-attachments/assets/8a04753d-6ffd-44f1-8047-fb5ed6683501

Looks like it is finally working! Tried with both master and slave and both spin! Thank you a lot for your help!

I have one more question(hopefully last one)
[image]
I saw this diagram on how to connect 2 of them simultaneously to esp32. Just to confirm I will need 2 diods to connect it and I should not connect without diods (if so which diods I should order)?
Also one more thing, how did you set power button to power on/off both motherboards at the same time, as they do not have master slave connection anymore?

RoboDurden, 2024-12-22:
You need to connect the 5V (or 15V) pin from the master-slave headers, so the slave will power up when the master is starting up.

any diode do. You can also use little 3mm LEDs.
With only two boards on the uartBus, it might also work without diodes.

Good night.
please do not ask questions, whose answers you can figure out yourself.

lUAlSvail, 2025-01-02:
Could you please share the source code for hoverboard 2.1.1 tx=PA2, rx=PA3 7s single UartBus id0.bin and id1?
And just to confirm that I understood it correctly, this firmware hack only helps to control the wheels and that is it. So for example on/off signal cannot be received by esp and passed to another board(following your advice on connecting boards with 5v I got both board powered up after clicking on button, but it only powers off one motherboard, looks like I will need to use some mosfet, was wondering if I have some control over that via esp32, or on the firmware hack level), as well as battery charging will not be handled anymore with power cable which was originally with hoverboard?

I-hate-2FA, 2025-01-02:
for controlling power programatically, i sugest to remove the definition of self hold and button pin, and then use another gpio on the esp connect it to the hoverboard button, theres 2 pin, one have full 42v on it, other have 0v, connect to the 0v one, if you send output high on that pin the board turn on, otherwise off, esp need common gnd 
it is so much nicer then send the off command cuz then you cant turn it on again whats the point its so stupid just turn it manually

battery charging is never present on these board, the charging connector is in parallel with the battery connector, with a diode for reverse protection, thats it

I-hate-2FA, 2025-01-02:
and i sugest to use all master board actually, because they have no bluetooth module and have the ability to read battery voltage, they are so cheap just get a bunch and only use the master
source code always is available

RoboDurden, 2025-01-02:
Yes, if the ESP32 is powered independently from the Hoverboard, and the trick with 3.3V digitalOutput to the input pin of the onOff button (and for sure NOT the 40V output pin) works, this might be the preferred way.

The slave board is powered by the 15V (or 5V) pin of the master slave cable.
So when the master turns off, the slave should also turn off.

My uart protocol has the option bit to turn off the master, take a look at the testspeed.ino code.

lUAlSvail, 2025-01-02:
The slave board is powered by the 15V (or 5V) pin of the master slave cable.
So when the master turns off, the slave should also turn off.

For some reason it does not happen... So I just connected on/off button to both  boards.  Maybe because both boards connected to the battery so it just get the trigger from 5v but than does not get any trigger when I click off button.

I-hate-2FA, 2025-01-02:
how is it possible to connect power button on slave

lUAlSvail, 2025-01-02:
[image]
[image]
They both have place to connect the pin

RoboDurden, 2025-01-02:
Yes, if you have Single UartBus firmwares on both the master and the slave, then the power hold function might also be active on the slave board.
But then the shutdown comand bit from my serial protocol should be able to turn off both "single" firmwares.

Sorry, i don't have the happiness to think deeper on this.

I-hate-2FA, 2025-01-02:
both board have the buck converter?most board the power is delivered on masterslave cable 15v which power the mosfet driver chip

RoboDurden, 2025-01-02:
yes, 2.1.1. seem to have dcdc converter and hold mosfets on both boards: https://raw.githubusercontent.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/main/Overview/Gen2.1.1.jpg

I-hate-2FA, 2025-01-02:
thats intresting, that mean if you run autodetect itll say its a master because effectively it just is, just treat it as 2 master

lUAlSvail, 2025-03-02:
@RoboDurden sorry to ping you again, but I have question regarding beeping in firmware hack. My board started double beeping constantly after I turn it on. And I am not sure what is wrong. I tried to google, but the best thing I got that it should be on the flat surface. Is there a way I can stop this beeping sound applying some code to ESP? Battery is fine and fully charged(when it is discharged it is constant single beeping so I know sound of that)

lUAlSvail, 2025-03-02:
Meanwhile everything else is working fine, wheels turning etc

RoboDurden, 2025-03-02:
Maybe low battery warning. 
I think in main.c you will find the different if clauses for beeping.