RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #104: Unable to control 2.1.1 board, dummy firmware works fine.

KiwiHuman, 2025-02-08:
I tried to get a single motor running with the 2.1.1 board (slave as I think I broke the master somehow as it wont connect to the ST link v2 tool) and whilst flashing the hoverboard 2.1.1 single Dummy.bin firmware causes the wheel to spin, I have been unable to control it with UART.
 I have tried 2.1.1 master Uart.bin, 2.1.1 single UartBus id0.bin - 2.1.1 single UartBus id4.bin, hoverboard 2.1.1 tx=PA2, rx=PA3 7s single UartBus id0.bin, hoverboard 2.1.1 tx=PA2, rx=PA3 7s single UartBus id1.bin conected to an arduino uno clone (the maker UNO Rev1.1) running  https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-GD32/blob/main/Arduino%20Examples/TestSpeed/TestSpeed.ino and Arduino Examples/Autodetect/Autodetect.ino

RoboDurden, 2025-02-08:
With UartBus, the hoverboard only answers after it received data with its id from arduino/esp.
So you better start with RemoteUart. This should always send the feedback struct to whoever is listening.
So you could connect to a usbUart dongle and see the garbage of the binary data.

Then succeed with receiving the data with testspeed.ino

ArduinoUno will need Softwareserial. Which should be able to handle the 19200 baud.

I have compiled and uploaded https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesReadyToFlash/hoverboard%202.1.1%20single%20Uart%20no%20button.bin
for you. This should run on the gen2.1.1 slave board with no onoff button. 

serial connection is on PB6/PB7 the second 4 pin header close to the st-link flash header..

For your bricked master board you might want to try the NRST pin .. 
It might need to modify your st-link dongle. There are some tutorials out there..

KiwiHuman, 2025-02-09:
I tried to connect to the second header you mentioned, however after soldering wires to the pins (labeled G T R 5V) (circled in the picture below)I am unable to upload firmware, the same thing seems to be what happened with the original master board. I'm not sure how just attaching some wires stopped it from working but I could flash firmware before adding the wires and I can't now. 

[image]

RoboDurden, 2025-02-09:
Interessting. Normally i would say that 5V can kill the 3.3V GD32F130
But seeing your photo i guess, it is 5VT 5v tolerant on the digital pins (no analog pin).

Check if you still have 3.3V on the flash header.

You should also try the different upload softwares.

like st-flash.exe with auto-flash.bat in https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/BinariesReadyToFlash

also read the wiki on flashing possibilities.