RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #105: Throttle trigger support using built-in ADC.

Eksol, 2025-03-10:
Is it possible to work without UART in this project? Using the gas and brake trigger as it is done in another project [hoverboard-firmware-hack-FOC](https://github.com/EFeru/hoverboard-firmware-hack-FOC)

RoboDurden, 2025-03-10:
Yes.
The master slave cable header usually is on the pa2 and pa3 pins which can be added to the analog input buffer.
If I remember correctly,.this buffer is defined in defines.h.

So you simply need to double this buffer and adjust the adc init code in setup.c

Then you can use these new ADC variables to calculate speed in main.c

You are welcome to add a RemoteAdc.c and make a pull request.

Eksol, 2025-03-10:
Maybe it will be better this way?  It can be confusing in the board versions. 

[image]

Eksol, 2025-03-10:
Why are two USARTs specified in the defines_2-1-18.c file? To connect two boards and control via one USART?

RoboDurden, 2025-03-10:
I think you should swap

```
#define USART1_MASTERSLAVE		// uncomment if this usart is used for master-slave communication
	//#define USART1_REMOTE		
```
So that PA2 and PA3 are correctly used for a RemoteXY.c

Then Pb6 pb7 in principle can be used for master slave. USART0_MASTERSLAVE

So you could attach a speed  and steer Potentiometer to the master and crossover connect the slave with the second serial Port, which is found close to the 4pin flash header.

Yes edit config.h to use board 2.1.18
Platform.ini was added by a nice user via pull request just recently.

RoboDurden, 2025-03-10:
Take a look at RemoteDummy.c 
That is where you should set speed and steer according to ADC buffer.
Not main.c, sorry.

Make your own RemoteAdc.c and .h based on RemoteDummy :-)

Eksol, 2025-03-10:
I don't see any braking function in the code. Is engine braking supported by the firmware?

RoboDurden, 2025-03-10:
No, not supported by now. And i do not really know how it should be done.
You can add a PID controller to control pwm via a real speed in rpm. Then define a breaking strength which decreases speed step by step.
But to break down to zero speed, i guess you would need a PID controller for torque.

But for a beginning, you could simply decrease speed (which is nothing but the pwm value) down to 0 within x seconds..

If you want braking, you need another (digital) input ?

I-hate-2FA, 2025-03-10:
braking just pwm the mosfet all to same level so all 3 short together and you release the energy stored in motor coil boosts to charge the battery via the mosfet protection diode
but because the uartbus protocol do not support it there is no point in adding because wouldnt be able to use it
i dont know why the attitude is changed to support other gen1 feature like adc again when before is said never will, i think none-programmatic control method is useless indeed

I-hate-2FA, 2025-03-10:
insted we should modify gen1 board to support uartbus, it is the protocol of the future (using 2 id per board for gen1 obviousely)

Eksol, 2025-03-11:
I found another interesting project. It already implements the use of the gas trigger. And it seems the brakes too. However, there is one problem. Difficulties with adding a new board configuration.

https://github.com/GreenBytes95/KickScooter

I-hate-2FA, 2025-03-11:
it is the same problem with gen1 too, the code is a mess and adc is hard coded, there is no chance to adapt for another pinout

RoboDurden, 2025-03-11:
Yes, the code is completely new to me.
Maybe @I-hate-2FA  wants to give a comment on it as he has done the mm32 firmware also with sine control.

https://github.com/GreenBytes95/KickScooter/blob/main/Keil/src/BLDC/bldc.c

```
// Title .........: BLDC - Управление двигателем.
// Arch ..........: GD32F130C8C6, GD32F130C6C6
// Author ........: GreenBytes ( https://vk.com/greenbytes )
// Version .......: 3.5.0.0
```

I wonder why hardware modifications are needed. Extremely annoying as 80% of all users have nearly zero experience with electronics.

I-hate-2FA, 2025-03-11:
i can check later when i have time, now i have to work with my android app projects first

Eksol, 2025-03-20:
Is it possible to set the RPM limit?

RoboDurden, 2025-03-20:
Reduce speed (pwm) when rpm is at limit..
Or implement a PID control loop to set rpm in remoteUart or alike
...

I-hate-2FA, 2025-03-21:
using my firmware you set the rpm directly, PID loop is built in

Electron-py, 2025-05-05:
Can someone add the remoteAsc.h or something similar? I do not have the skills to program it myself and copilot didn't help.

RoboDurden, 2025-05-05:
What does Asc stand for ??

Electron-py, 2025-05-05:
RemoteAdc, typing error

RoboDurden, 2025-05-05:
I don't really have the happiness to code the Remote ADC, because you probably also only do a stupid run project.

Deepseek or chatgpt can help you install the hoverserial.ino on a $2 esp32 S2 mini.
Then you can do ADC there.

I-hate-2FA, 2025-05-06:
> I wonder why hardware modifications are needed. Extremely annoying as 80% of all users have nearly zero experience with electronics.

i dont know, i dont know what is modified, it didnt show before and after not even the schematic, i can not tell
it is obvious one of the led gpio is modified to adc input other 2 i dont know probably disable over current protection or related to the 2 infrared sensor which doesnt even exist on the mm32 board

the guy developing the esp uartbus translator is gone too so now its how it is someone can fork it and add the adc

I-hate-2FA, 2025-05-29:
Did you remember to disable low battery protection

RoboDurden, 2025-05-29:
change code in main.c to see where the beeping comes from:

```
			// Show green battery symbol when battery level BAT_LOW_LVL1 is reached
			if (batteryVoltage > BAT_LOW_LVL1)
			{
				// Show green battery light
				ShowBatteryState(2);
				
				// Beeps backwards
				BeepsBackwards(beepsBackwards);
			}
			
			// Make silent sound and show orange battery symbol when battery level BAT_LOW_LVL2 is reached
			else if (batteryVoltage > BAT_LOW_LVL2 && batteryVoltage < BAT_LOW_LVL1)
			{
				// Show orange battery light
					ShowBatteryState(1);
				
				BuzzerSet(5,8)	// (iFrequency, iPattern)
			}
			// Make even more sound and show red battery symbol when battery level BAT_LOW_DEAD is reached
			else if  (batteryVoltage > BAT_LOW_DEAD && batteryVoltage < BAT_LOW_LVL2)
			{
				// Show red battery light
				ShowBatteryState(0);

				BuzzerSet(5,1)	// (iFrequency, iPattern)
			}
			// Shut device off, when battery is dead
			else if (batteryVoltage < BAT_LOW_DEAD)
			{
				ShutOff();
			}
			else
			{
				ShutOff();
			}

```

RoboDurden, 2025-05-29:
You have to add
```
// ADC buffer struct
typedef struct
{
  uint16_t v_batt;
  uint16_t current_dc;
  uint16_t speed_adc;
} adc_buf_t;
```
And than adapt the setup code

Eksol, 2025-05-30:
Wow. Have you started adding control using the gas triggers?

RoboDurden, 2025-05-30:
When @Electron-py  has succeeded, i will be happy to accept a `remotAdc.h`/`remotAdc.c` pull request or import his code if he publishes his fork.

Electron-py, 2025-05-30:
> When @Electron-py  has succeeded, i will be happy to accept a `remotAdc.h`/`remotAdc.c` pull request or import his code if he publishes his fork.

I can already share the changes but I need to fix this issue. Also I don't know how to add the remoteADC.h

I-hate-2FA, 2025-05-30:
yeah i dont know why robo changed the attitude to make uartbus the one and only the best control method i really like the idea but again the one responsible for the uartbus converter esp32 side also ran away and abandon the project so oh well

Eksol, 2025-05-30:
I have a question. How high is the PWM frequency possible to set the maximum? I'm comparing the maximum RPM from two controllers. And the controller based on projects from github gives less speed. Twice as much! I noticed that the controller that gives high speeds has a PWM frequency of 32 kHz.

Eksol, 2025-05-30:
Don't waste resources on weakening the field. This is a harmful feature.

Eksol, 2025-05-30:
[image]
[image] 

So what PWM frequency can be included in your code? If there is already support for gas triggers, then I can test the firmware!

I-hate-2FA, 2025-05-30:
the field weakening PR does not do SVPWM and should not be merged without more in depth look

I-hate-2FA, 2025-05-30:
> [image] [image]
> 
> So what PWM frequency can be included in your code? If there is already support for gas triggers, then I can test the firmware!

this waveform looks like sine wave to me tbh

RoboDurden, 2025-05-30:
what do you mean with good controller and bad controller ?? Different hardware = good control board vs bad control board ?
Or good motor vs bad motor ? The cheap motors only have 20mm magnets. The common ones 25mm magnets, and i found one hoverboard with 30mm magnest. All 6.5 inch: https://youtu.be/cYIpnW6RCSE
Or good firmware vs bad firmware ? And bad firmware = Gen2.x ? So what is the good firmware ?

All your oscilloscope photos look like sine modulation ! With Gen2.x and it's simple block commutation i would have expected straight horizontal blocks at speed = 1000 = 100% pwm ratio.

Eksol, 2025-05-30:
For some unknown reason. The scooter controller can accelerate the motor to 1000 rpm. While the homemade controller from the gyro scooter accelerates the motor only to 650 rpm. I am looking for reasons. I found the difference in the conversion frequency.

Eksol, 2025-05-30:
So far, I've been using the controller from the kugoo m4 scooter. This controller accelerates the motor wheel to 1000 revolutions. I haven't been able to use your firmware yet because I don't have any additional arduino boards for UART operation. That's why I'm waiting for a fork under the gas triggers. 
I used the kickscooter firmware as it supported operation from the gas triggers. But unfortunately, it has a very low PWM frequency of 8kHz, which makes the motor very noisy and inefficient. 
The motor and the supply voltage did not change. Only the controllers were changed.

Eksol, 2025-05-30:
i use this project https://github.com/GreenBytes95/KickScooter 

But I also used a project from Efera, and it also could not accelerate the engine beyond 650 rpm. I do not know exactly how the scooter controller manages to accelerate the motor without weakening the field.

Eksol, 2025-06-03:
Project files added by remote.adc? What is the maximum PWM frequency?

I-hate-2FA, 2025-06-03:
@RoboDurden is there no need to cancel init of the serial when adc is anabled because pinmodeAF is messed up

RoboDurden, 2025-06-03:
@Eksol , remoteAdc.h is included in remote.h

```c++
	#elif defined(REMOTE_ADC)
		#include "../Inc/remoteAdc.h"
	#endif
```

Thanks @I-hate-2FA for your help!
With my config/defines cleanup, the user should no longer be in need to edit a definesXY file. The usage of the two uart are now set in config.h

```c++
	#if defined(MASTER) || defined(SLAVE)
		#define MASTERSLAVE_USART		1 	// 	1 is usually PA2/PA3 and the original master-slave 4pin header
								//	0 is usually PB6/PB7 and the empty header close to the flash-header
	#endif
	
	#if defined(REMOTE_UART) || defined(REMOTE_UARTBUS) || defined(REMOTE_CRSF)
		#define REMOTE_USART				0 	// 	1 is usually PA2/PA3 and the original master-slave 4pin header
									//	0 is usually PB6/PB7 and the empty header close to the flash-header
	#endif
```

With REMOTE_ADC this gets overwritten in defines.h

```c++
#ifdef REMOTE_ADC
	#ifndef USART1_TX
		#error "REMOTE_ADC only works with USART1 PA2/PA3"
	#endif
	#undef MASTERSLAVE_USART
	#undef REMOTE_USART
	#if defined(USART0_TX) && defined(MASTER)	// only uart left
		#define MASTERSLAVE_USART 0
	#endif
#endif
```

There is a lot more ugly #ifdef stuff hidden in defines.h to finalize the two uart definitions for the different use cases :-/

But yes, with REMOTE_ADC defined, the setup code `USART1_Init(uint32_t iBaud)` gets deactivated, so no problem with `pinmodeAF`. 
But i had another look into setup.c and found that pinMode was missing in `GPIO_init(void)`:

```c++
		#ifdef REMOTE_ADC
			pinMode(PA2, GPIO_MODE_ANALOG);
			pinMode(PA3, GPIO_MODE_ANALOG);
		#endif
```

So now, both adc inputs adc_buffer.speed and adc_buffer.steer are functional.

I have for now removed the non-functional lowpass in bldc.c (gets overwritten by DMA anyway). my little video above demostrates, that the adc readings seem to be very stable/reliable.

Now i can add the steer parameter and test the masterslave from PB6/PB7 of the MASTER to the ordinary SLAVE with PA2/PA3

Eksol, 2025-06-03:
I'll have to test your firmware! I wonder what PWM frequency is supported?

RoboDurden, 2025-06-03:
config.h :  `#define PWM_FREQ         		16000     // PWM frequency in Hz
`

RoboDurden, 2025-06-03:
Okay, calibration mode is online :-)
Simply hold the on/off button until the startup melody restarts.
Then release the button and leave the joystick (the two potentiometers) in neutral position.
When the melody returns for 2 seconds, push speed to max
after another 5 seconds + 2 seconds meldoy: push speed to min
then steer to max
finally steer to min
Then the new oConfig (struct defined in defines.h) is stored to flash.

Tested for joystick with neutral position and joystick with neutral beeing pulled backwards. Then the original mid position is 50% speed and full push forward = 100%

If anyone wants to add parameters to oConfig, you have to increment `#define EEPROM_VERSION 1` so the old stored data gets reset.

defines.h
```c++
#pragma pack(push, 1)
typedef struct
{
  uint32_t iVersion;
  uint16_t wState;
  uint16_t iSpeedNeutral;		// = 2048 REMOTE_ADC
  uint16_t iSteerNeutral;		// = 2048 REMOTE_ADC
  uint16_t iSpeedMax;				// = 4096 REMOTE_ADC
  uint16_t iSpeedMin;				// = 0		REMOTE_ADC
  uint16_t iSteerMax;				// = 4096	REMOTE_ADC
  uint16_t iSteerMin;				// = 0		REMOTE_ADC
} ConfigData;
#pragma pack(pop)
```

setup.c
```
#define EEPROM_VERSION 1
//#define STATE_InverterOn  1
ConfigData oConfig;

void ConfigReset(void) 
{
	oConfig.iVersion = EEPROM_VERSION;
	oConfig.wState = 0;
	oConfig.iSpeedNeutral = 2048;
	oConfig.iSteerNeutral = 2048;
	oConfig.iSpeedMax = 4096;
	oConfig.iSpeedMin = 0;
	oConfig.iSteerMax = 4096;
	oConfig.iSteerMin = 0;
}
```

main.c
```c++
int main (void)
{
	ConfigRead();		// reads oConfig defined in defines.h from flash
```

remoteAdc.c
```c++
		case 5:	// read min adc_buffer.steer
			fSteer= (1-LOWPASS) * fSteer + LOWPASS * adc_buffer.steer;
			if (millis() > iTimeNextCase)
			{
				oConfig.iSteerMin = fSteer;
				int16_t iRangeMax = oConfig.iSpeedMax - oConfig.iSpeedNeutral;
				int16_t iRangeMin = oConfig.iSpeedNeutral - oConfig.iSpeedMin;
				if (iRangeMin != 0)
					if (iRangeMax/iRangeMin>10)	oConfig.iSpeedMin = oConfig.iSpeedNeutral;	// only forwards
				
				if (iRangeMax != 0)
					if (iRangeMin/iRangeMax>10)	oConfig.iSpeedMax = oConfig.iSpeedNeutral;	// only backwards
					
				ConfigWrite();
				iConfigMode = 0;
				BUZZER_MelodyDown();
			}
			break;
		}

```

RoboDurden, 2025-06-03:
I don't understand

RoboDurden, 2025-06-03:
Well, i would not need to store the static kown layouts into the flash, i could hard code them in the normal binary (only the autodetect binary gets close to 32kB).
The user then could choose the layout via uart and putty.