RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #107: Burned motors

hoverboardhavoc, 2025-04-12:
[image]

I finally got the chance to test my first hoverboard motor-powered project - the “el platform” (basically a pallet with four motors 🤷). It uses some old hardware I had lying around: a [Blue Pill](https://stm32-base.org/boards/STM32F103C8T6-Blue-Pill.html) connected to a FrSky SBUS radio receiver.

I was a bit skeptical about the bus protocol - partly because I thought the wires might be too long and because it seemed to be unidirectional. That is, it didn’t look like I could get any feedback from the hoverboard motors to the microcontroller (though maybe I’m wrong about that?).

Since the Blue Pill only has three UARTs, I implemented two additional software serial ports. This required some tweaking of the Arduino libraries, since only one SoftSerial is supported by default - even though the underlying hardware can handle more. Each motor is connected to its own UART.

Anyway it seemed to work well -  at least at first. Eventually two of the motors seemed to have burned out. Maybe the tank turns just drew too much current?
[image]

When I connect a known working motor to the board that previously had a burned motor, it’s hard to turn even with the power off. Does this mean some of the FETs have also failed?

Any tips for avoiding this in the future? Is there a way to limit current when the wheels aren’t spinning?

Thanks for the help

hoverboardhavoc, 2025-04-12:
BTW is there a discord to discuss this kind of thing? If not, maybe you want to make a discord @RoboDurden ?

RoboDurden, 2025-04-12:
Another stupid fun project :-/

36V battery ?
Normaly the motors can not get that hot at 36V ?
Gen2 boards ? Gen2.1 ? Which layout, Gen2.1.x ?
Why then not use my UartBus protocol ? You could control all four motors with one uart port and have serial feedback with speed and current consumtion..

If the boards have no power, the motors should spin freely.
Check the mosfets for a short cut with a multimeter (beeping diode tester)

RoboDurden, 2025-04-12:
Does your palet-car has a steering axis ?
If not and turning left and right is done by speed difference between left side and right side, this might cause high currents..

hoverboardhavoc, 2025-04-12:
> Another stupid fun project :-/
> 
> 36V battery ? Normaly the motors can not get that hot at 36V ? Gen2 boards ? Gen2.1 ? Which layout, Gen2.1.x ? Why then not use my UartBus protocol ? You could control all four motors with one uart port and have serial feedback with speed and current consumtion..

I'm using two lifepo4 batteries in series so about 26V. [This is the board](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-GD32/pull/2) It only one exposed UART unfortunately. 
> 
> If the boards have no power, the motors should spin freely. Check the mosfets for a short cut with a multimeter (beeping diode tester)

I'll give that a try

hoverboardhavoc, 2025-04-12:
> Does your palet-car has a steering axis ? If not and turning left and right is done by speed difference between left side and right side, this might cause high currents..

No only tank turns (speed difference) 😆 

[image]

RoboDurden, 2025-04-12:
26V can not really burn a motor..
But with tank steering only and a person sitting on it, i can imagine that the motors will overheat :-(
Former 2.20 now is 2.1.18:

[image]

hoverboardhavoc, 2025-04-12:
> 26V can not really burn a motor.. But with tank steering only and a person sitting on it, i can imagine that the motors will overheat :-( Former 2.20 now is 2.1.18:
> 
> [image]

Yes, they were crazy hot. I did not realize until it was too late. The image you posted is not my board. It looks like this:

[image]

BTW the motor that overheated (shown in the picture above) is hard to turn. I guess that means the coils are shorted?

RoboDurden, 2025-04-13:
Yes, shorted phases inside the motor is best explanation.
You need to add a steering mechanism to such a heavy transporter.

hoverboardhavoc, 2025-04-13:
I just noticed now, that the traces on the board are burned. It must have pulled a lot of current

[image]

RoboDurden, 2025-04-13:
This usually happens when a pair of mosfets burn and make a battery short cut.  See one of my first repair videos.
If you are lucky, the over current only came from the tank motors. 
If the board is still working without motor connected and the broken pair removed, you might try to solder in new mosfets. 
But maybe the gate drives have been damaged, then better  get new boards.