RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #108: 2.1.1 and motor without hall sensors

OM-3kChcBjgFR2COfP, 2025-04-15:
hi, can i control 3 phase motor without hall sensors by 2.1.1 variant plate?

RoboDurden, 2025-04-15:
No.

But i see at least two sensorless bldc firmwares on github:

https://github.com/sahil-kale/SensorlessESCSoftware
https://github.com/vladBaciu/Sensorless-BLDC-controller

And as we have the pin definitions of gen2.1.1, it might be possible to get such a sensorless firmware running on a gen2 board.

But they both utilize zero cross point  detection:
[image]

Gen2.1.1 on the other hand has phase currents. Mabe it is possible to use these to calculate the exact rotor position..

I-hate-2FA, 2025-04-15:
it is possible to drive sensorless with square wave without any sensing, i dont know how is it done but all modern ebike controller can do it, sine wave is possible to drive sensorless but the effiency is so poor its not worth it

RoboDurden, 2025-04-15:
Did someone trace these cheap bldc sensorless controller ?

[image]

Then we would see if hoverboard controllers could do the same..