RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #111: I believe this board hasn't been posted yet

Vicky-lp01, 2025-05-22:
[image]

[image]

The MCU is a GD32f130 C6t6, with 3 Gate Driver Chips. I'll be doing the reversing engineering and post here. It's similar to the 2.1.7 board, but with some big differences.

Vicky-lp01, 2025-05-22:
@RoboDurden I will !! But I don't have the soldering station today, any updates i'll post here.  

I'm just a little confused about whats the space with 4 above the buzzer. Comparing with other boars I think it could be an UART, but my other boards didn't have one, so i'm not sure.

Vicky-lp01, 2025-05-22:
@That's exactly it, thanks. But what's the purpose of the second UART ?

RoboDurden, 2025-05-22:
Esp32 Control program.

RoboDurden, 2025-05-28:
i have assigned the number Gen2.1.22 to this board and updated the wiki.
Photos of the two backsides would help manual tracing.
But running autodetect will be of more use: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/BinariesToTest
PA2/PA3 is the master-slave serial header,
PB6/PB7 that additional empty uart header

Vicky-lp01, 2025-05-28:
Ok, i'll try it now.
 I'm kindda new to this, should I run the bin file directly on Keil ? Or other software like the smt utility ?

After this I'll also upload a photo of the back and the slave board. This is what i've gathered this far, just with the hardware:

[image]

I-hate-2FA, 2025-05-28:
If you want programmatic control method is better to not use a button

RoboDurden, 2025-05-28:
Might be good if you quickly add a wiki page describing the the setup to do without the button pin and hold pin by simply setting the button input to high from the esp32.
And mention the danger to kill the esp32 by outputting to the wrong pin of the 2pin onOff button header.

I-hate-2FA, 2025-05-29:
Ok I will do later

I-hate-2FA, 2025-05-29:
I don't know how to disable the latch and button in your firmware

Vicky-lp01, 2025-05-29:
I'm sorry for the many questions i'm making, but just to be safe, I could also do this by simpling short circuiting the button connection on the board ? I belive i've seen some people saying something of the sorts

I-hate-2FA, 2025-05-29:
For auto detect you need to first short the button and follow the instructions board give you for main you can short the button to provide power but I don't know if the firmware will be able to work correctly for example my firmware will wait until the button it released before powering on so you need to disable it in code

RoboDurden, 2025-05-29:
config.h

`		#define CHECK_BUTTON		// disable = add '//' if you use a slave board as master
`

bldc.c
```
void CalculateBLDC(void)
{
	#ifdef CURRENT_DC
		// Calibrate ADC offsets for the first 1000 cycles
		if (offsetcount < 1000)
		{  
			offsetcount++;
			offsetdc = (adc_buffer.current_dc + offsetdc) / 2;
			return;
		}
	#endif
	
	// Calculate battery voltage every 100 cycles
	#ifdef VBATT
		if (buzzerTimer % 100 == 0)
			batteryVoltage = batteryVoltage * 0.999 + ((float)adc_buffer.v_batt * ADC_BATTERY_VOLT) * 0.001;
	#endif
	
  buzzerTimer++;	// also used to calculate battery voltage :-/
#ifdef BUZZER...
```

I-hate-2FA, 2025-05-30:
> Might be good if you quickly add a wiki page describing the the setup to do without the button pin and hold pin by simply setting the button input to high from the esp32. And mention the danger to kill the esp32 by outputting to the wrong pin of the 2pin onOff button header.

added: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki/Control-power-of-the-board-programatically

RoboDurden, 2025-05-30:
@I-hate-2FA  great that you still care for our wiki here :-)
I do not really understand at all what you mean with

> you can connect the low voltage side of multiple boards together, in series with a switch to one of the boards high voltage pin, that way multiple board can be controlled at the same time

what is a low voltage side ? GND ? and high voltag pin ? The 42V at the button header ?
Maybe a picture with multiple boards connected would help.

RoboDurden, 2025-05-30:
ah now i understand. This is not for activating multiple board from ESP/Arduino with its 3.3V output pin, but with an external power switch. As the title of the wiki page is 'Control power of the board programatically', this did not came to my mind.

I-hate-2FA, 2025-05-30:
when DISABLE_SELF_HOLD is defined, why do i need to care about the check_button, the program should automatically determane then since no self hold is present the button is useless and skip the check
```c++
if(LATCHPIN<PINCOUNT){    //have latch
		DELAY_Ms(100);    //some board the micro controller can reset in time and turn back on
		digitalWrite(LATCHPIN, 1);
		while(digitalRead(BUTTONPIN)&&BUTTONPIN<PINCOUNT){    //wait while release button
			if(BUZZERPIN<PINCOUNT){    //have buzzer
				digitalWrite(BUZZERPIN, 1);
				DELAY_Ms(2);
				digitalWrite(BUZZERPIN, 0);
				DELAY_Ms(2);
			}else{
				__NOP();
				__NOP();
			}
			IWDG_ReloadCounter();
		}	
	}

```
```c++
if(LATCHPIN<PINCOUNT&&BUTTONPIN<PINCOUNT){
			if(digitalRead(BUTTONPIN)){    //button press for shutdown
				TIM1->CCR1=0;    //shut down motor
				TIM1->CCR2=0;
				TIM1->CCR3=0;
				if(BUZZERPIN<PINCOUNT){
					for(int i=0;i<3;i++){    //power off melody
						digitalWrite(BUZZERPIN, 1);
						DELAY_Ms(150);
						digitalWrite(BUZZERPIN, 0);
						DELAY_Ms(150);
					}
				}
				while(digitalRead(BUTTONPIN)) {    //wait for release
					__NOP();
					IWDG_ReloadCounter();
				}
				digitalWrite(LATCHPIN, 0);    //last line to ever be executed
				while(1);//incase the hardware failed...
			}
		}

```

Vicky-lp01, 2025-05-30:
[image]
Ok, got the pinfinder working and this is what I got. 

The last thing is the Vbat, i'm using a only 15V power suply (is the only one I got). I think it may be the PA0, but should I try again with a higher voltage ?

Vicky-lp01, 2025-05-31:
@RoboDurden I actually do! I also had an MM32 board, so sometimes I get them mixed up.

I installed the autodetect feature and it works great. I'll try the firmware this weekend and make the pull request. Thank you so much for the support.

I do have one more question: I was hoping to use the FOC since I'll be working at low speeds. However, I read in another issue that it can't be done on the C6T6 due to memory limitations. Is there really no hope?

But even without FOC, I can still use sinusoidal commutation with this firmware, correct?

Vicky-lp01, 2025-05-31:
> pb12 is always the over current protect because of hardware limitation the photo is too blury to see but isnt pb1 still not used pf6 pf7 is never used because it doesnt exist on f103

Oh, that's good to know, thanks

Pb1 is not used indeed, but there are no traces connected to it. Pb2 does leed somewhere, but i'm not sure on what it does. I' also upload a better picture

I-hate-2FA, 2025-05-31:
the pb2 is untraceable using picture, it went to near the infrared at the side with large capacitor everything is blocked
this board have a single opamp so foc may be possible but not needed
mm32 hardware in incapable of foc but using sine wave at low speed it still works well

I-hate-2FA, 2025-05-31:
or maybe you think mm32 didnt perform well enough so you look here...

if  thats the case i can safely say mm32 perform way better then gd32

RoboDurden, 2025-05-31:
@I-hate-2FA  means that his mm32 firmware performs way better than the Gen2.x_GD firmware here, which is correct, as i only offer stupid simple block commutation.

If you are looking for a gd32f130 firmware with sine pwm, you could try https://github.com/GreenBytes95/KickScooter

but please still post your defines_2-1-22.h file.
The main purpose of this Gen2.x.y repo is to collect all the different layouts and at least offer a firmware to let the motors spin..
I have no happiness to add better bldc control..

Vicky-lp01, 2025-06-02:
> or maybe you think mm32 didnt perform well enough so you look here...
> 
> if thats the case i can safely say mm32 perform way better then gd32

No, it's just that I only have a GD32 board, and I was having some trouble figuring out where to start. 
I'm not really familiar with this kind of thing, and I got here through the YouTube videos. However, I will look into the MM32 firmware and see the differences. Also, thank you for answering my questions.

I-hate-2FA, 2025-06-02:
the gen1 performs better then mm32 i hate to say it but i have tested and its the case

the downside is that its hard to buy a real gen1 board and if you get one thats just slightly different the pinout of gen1 is not easily changeable like all gen2 is built with different pinout in mind

Vicky-lp01, 2025-06-04:
@RoboDurden  I'm so sorry to bother you again. I tested the remote_ADC code yesterday and tried connecting the slave board through the 15V and the cables from USART0, but it does not spin. 

I haven't turned on the slave board yet so I tried to use it as a single board and use the DISABLE_BUTTON option, also no luck. So I wanted to use the AUTODETEC code again, just to be sure.

To use the code on a slave board the only thing I have to do is to comment  #define CHECK_BUTTON on config.h ? Tried it but it's not being detected in the Arduino code.

 Currently the #define DISABLE_BUTTON is locked under the condicion that REMOTE_AUTODETEC is not defined.`  

    #ifdef REMOTE_AUTODETECT

	   #define REMOTE_USART 	1 	// 	1 is usually PA2/PA3 and the original master-slave 4pin header
								
                                    //	0 is usually PB6/PB7 and the empty header close to the flash-header

	   #define DISABLE_BUTTON`

Tried doing this manually but still no luck. I'm not sure on what to do.

RoboDurden, 2025-06-04:
Yes, the 2.1.22 slave board looks quite different, so a new autodetect might be a good idea. `DISABLE_BUTTON` is in defines.h only used for

```
#ifdef DISABLE_BUTTON
	#undef BUTTON
	#undef SELF_HOLD
#endif
```
and BUTTON and SELF_HOLD are pin definitions that are never used in autodetect code.
Simply strg+shif+f to search the entire project for SELF_HOLD when REMOTE_AUTODETECT is active and you will see that in only appears in greyed out code.
So you do not have to care about DISABLE_BUTTON with REMOTE_AUTODETECT

Normally, the slave turns on when it gets the +5/+15V from the masterslave cable.
And the PA2/PA3 autodetect binary should connect to the master-slave cable.
If this header on the slave is indeed pa2/pa3, which it always has been.

Check the 3.3V on the flash header, it should be 3.3V when the mcu gets power.

Vicky-lp01, 2025-06-04:
OHH, ok, that makes more sense now, thank you

> Normally, the slave turns on when it gets the +5/+15V from the masterslave cable. And the PA2/PA3 autodetect binary should connect to the master-slave cable. If this header on the slave is indeed [...]

Yeah, that's my problem right there. Just checked and even with the power on there is no power on the board, it only shows 0, both on the flash header and USART. Maybe the board is just broken, thanks for all the help anyways.

RoboDurden, 2025-06-04:
If you connect everything as it was inside the hoverboard originally, the slave should turn on, when you push the BUTTON of the master.
Normally there is a linear 5V regulator, followed by little ams1117 3.3V linear regulator on the board.
See where the 15V of the masterslaver header go... the 5V regulator could be this:

[image]

RoboDurden, 2025-08-09:
Check cables.

Use short cable from dongle to MCU

Is read out protection disabled ? St link utility and strg+B

On windows: autoflash.bat

Modify st link dongle to output rst and connect to nrst pin.

Serial flashing.

See the 2.3.3 issue for the latter two.

Vicky-lp01, 2025-08-27:
@RoboDurden 

Hey, this is probably not the right place, but I could't find any sort of e-mail to send a message. Just wanted to share that I just graduated my engineering degree and the I did my dissertation based on your work.

We have a project in my University to provide mobility aid for children with disabilities. This next semester we'll be receiving 3 more hovers through the gorvenament. This project is already showing some real change in the world. I wanted to take some time to thank you.  For all the work you do and also all the support you give everyone in this community. (All in portuguese, I'm sorry: https://www2.ufjf.br/noticias/2021/08/16/projeto-promove-mobilidade-para-criancas-com-deficiencia/)

I think I've properly given the credits to everyone, but I also personaly thanked you in the beginning of my text. I made some minor alterations on the code for ADC calibration and reduced speeds. And a simple aplication using ESP_NOW. I'll post the code on my Git probably next week. In the meantime, I'll look into the new updates you've posted and also try to my hands on the PID closed loop (this is my expertise, can't wait).

RoboDurden, 2025-08-27:
Nice to see that my RemoteAdc indeed is used fore something else than a stupid fun project.
No need to credit me.

Yes, would be nice if you would look at the PID code. I did not have the time to fine tune the parameters and i do not really want to understand the PID logic..

We plan to add a wiki page for github projects build on Gen2.x. If it is not there when your git is online, please create such a wiki page here yourself.

RoboDurden, 2025-08-31:
Okay but i must say that this red bobbycar again looks like a SFP stupid fun project to me :-/

And i am also disappointed that people wanting to help disabled people only come up with motorizing an ordinary wheelchair -
When the stock hoverboard firmware can already do balancing: 

[[image]](https://www.youtube.com/watch?v=o_8ng8ITOuM)

I think that life would be and feel much more equal for people confined to a wheelchair, if they would sit at eye level !!!
Living on wheels could even be faster and quicker than on two legs!

And already mounting a minmal cage for head and body on a stock hoverboard might work. With the cage being able to extend as the person grows.

But some saftey legs that unfold when the system fails might be nice, because these could also be used to climb stairs

So there are lots of opportuinites..

Vicky-lp01, 2025-09-05:
@RoboDurden First, sorry for the delay in response

The red bobbycar is kindda off a SFP, but it's what we have available for now, mostly through donations. We are using a the hover firmware as a new line of research, but there is a lot to be done still.

About the wheelchairs, I totally  agree !!  Phill malone acctualy used this kind of support, where you can stand on it, but it's so incredibaly expensive. Trying to look for a way that's afordable but also safe.

[image]

There's a lot to do, but I think we can get there