RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #115: how to add flash/eeprom ?

RoboDurden, 2025-05-31:
@I-hate-2FA  i found some nice flash read/write functions in that russian hoverboard firmware https://github.com/GreenBytes95/KickScooter

```c++
# define TRUE												0x01
# define FALSE												0x00

/* Clears a page of microprocessor memory */
void flashErase(uint32_t address) {
	fmc_unlock();
	fmc_flag_clear(FMC_FLAG_END | FMC_FLAG_WPERR);
	fmc_page_erase(address);
	fmc_lock();
}

/* Reads 4 bytes from microprocessor memory */
uint32_t flashRead(uint32_t address) {
	return *(uint32_t*)address;
}

/* Writes 4 bytes to microprocessor memory */
uint8_t flashWrite(uint32_t address, uint32_t data) {
	uint8_t fflash = FALSE;
	fmc_unlock();
	fmc_flag_clear(FMC_FLAG_END | FMC_FLAG_WPERR); 
	if (fmc_word_program(address, data) == FMC_READY) fflash = TRUE; 
	fmc_lock(); 
	return flash;
}

/* Write a memory buffer to the microprocessor memory */
void flashWriteBuffer(uint32_t address, uint32_t pbuffer, uint8_t len) { 
	int16_t icount = 0; 
	while (1) { 
		if (flashWrite(address+icount, *(uint32_t*)(pbuffer + icount)) == TRUE) { 
		if (icount > len) break; 
		icount = icount + 4; 
		}
	}
}

/* Read into the microprocessor memory buffer */
void flashReadBuffer(uint32_t address, uint32_t pbuffer, uint8_t len) {
	int16_t icount = 0;
	while (1) {
		*(uint32_t*)(pbuffer + icount) = *(uint32_t*)(address+icount);
		if (icount > len) break;
		icount = icount + 4;
	}
}
```

The fmc_ functions are defined in `gd32f1x0_fmc.c` but this file does not seem to be included in my keil project.
If i switch to the GDE230 target, the code compiles successfully.

This KickScooter project looks nice:

[image]

Whereas my project looks like a mess:

[image]
[image]

and this `gd32f1x0_fmc.c` file is missing :-(
Maybe you can easily tell me how to fix this project here.

I-hate-2FA, 2025-05-31:
when i have time i can look

RoboDurden, 2025-05-31:
i asked Gemini (2.5pro preview) :-)

[image]

[image]

[image]

I-hate-2FA, 2025-05-31:
did you implemented it sucessfully

RoboDurden, 2025-05-31:
At least it compiles successfully. but i have not uploaded it yet to this repo.
It will only be usefull for RemoteAdc, to store the potentiometer min and max values..
Then i could also use it for autodetect to store the self_hold pin and remove my akward tempory storage on the esp32.

RoboDurden, 2025-05-31:
This will really change how i code. I now have to decide what it is that i can do better and let ai do the rest.
My remoteXY oo-style for example is something that ai would not have come up with.