RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #117: Controller FOC support

I-hate-2FA, 2025-06-09:
Sorry it's unrelated, @RoboDurden can you identify if this board have FOC capabilities 

[image]

[image]
[image]

[image]

RoboDurden, 2025-06-09:
Is this a dual motor board or does it have two mosfet in parallel for each of the single six mosfet outputs to double the max current ?

> The 74HC595; 74HCT595 is an 8-bit serial-in/serial or parallel-out shift register with a storage register and 3-state outputs.

So driving two motors with only 6 gpio outputs of the MCU ?

> The CD405xB analog multiplexers and demultiplexers are digitally-controlled analog switches having low ON impedance and very low OFF leakage current.

So where is the single/tripple mosfet driver ?
May these will a have builtin amplifiers for phase currents ?
You could trase the gate of the low-side mosfets and see if they go to where the on-voltage (phase currents) get amplfied.

RoboDurden, 2025-06-09:
Why is FOC so important ? 
I am quite happy with my BLDC_SINE. Motor efficency does not increase with FOC ?
Only better PID for torque or speed ?

RoboDurden, 2025-06-09:
stupid fun project ?
What is the battery voltage ?
You can boost with two 30A step up modules..

With my little solar camper i can only boost speed by 20% with the field weakening of EFeru, otherwise, the power consumptaion gets to high / efficiency to bad.

I-hate-2FA, 2025-06-09:
48v,14s, i don't really want to use boost if not necessarily although it'll maintain a fast speed when battery is discharged