RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #122: Adding ROS2 control / protocol

RoboDurden, 2025-07-08:
@WestlingPi  wrote:

> I am going to start to implement an alternative protocol for remoteUart for integration with ROS2. I think it is mostly the message format and encoding/decoding of message that is going to be different. I am thinking of introducing a new ifdef for the alternative protocol (similar to the flash code have 2 formats, one for autodetect and another for remoteAdc). [@RoboDurden](https://github.com/RoboDurden) Would that be something that you would be interested in to upstream to your gen2.x repo? I think maybe more people would be interested in a protocol that integrates with a ROS2 driver (https://github.com/WestlingPi/hoverboard-driver)...

Yes, but please add a RemoteROS2.c and RemoteROS2.h !
Simply copy RemoteUart into these new files and change to your protocol

Maybe you can use this PID controller for ROS2: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/120#issuecomment-3005454189
I am no PID expert at all, but maybe it can also be used for position iOdom control ?

WestlingPi, 2025-07-10:
I dont think I will need a PID controller, atleast not to start with. The ROS2 driver I plan to use seems to have a PID controller on the ROS side (which is a a bit strange, I agree, but for not it suits my purposes).

WestlingPi, 2025-08-06:
I have done some more testing and thinking and I think I have a plan now. It seems like the "open_loop" parameter of ros2_control diff_drive_controller is only used for the odometry calculations, i.e. it does not effect the motor velocity commands. So I guess it means that ros2_control assumes that there is a PID loop on the hardware side (i.e. hoverboard in this case), although I havent found it explicitly stated. It also makes sense to me.
Also, I think I will keep the speed a steer parameters, it will just complicate the firmware code to have different input for different controllers. But I will move and continue the PID controller discussion in the PID issue https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/120

RoboDurden, 2025-08-06:
PID is such a common use case that Deepssek or Gemini should have no problem at all to give you working code.
Problem is only where to put the code.
If you keep the speed/steer framework than you might have to disable the low pass in main.c
speed and steer are only 16bit so a future pid on iOdom with 32 bit might not work.
And the pid loop might better be implemented in calculateBldc with its up to 16 kHz call rate.
That is why i already mentioned a possible third virtual method in remote.h 

```c++
	void RemoteCalculate(int16_t* pBldc_Pwm);			// must be implemented by all remotes
```

Which could be called at

```c++
	RemoteCalculate(&bldc_outputFilterPwm);
	bldc_get_pwm(bldc_outputFilterPwm, pos, &y, &b, &g);
```

allowing every RemoteXy to adjust or replace the speed+lowpass value..

WestlingPi, 2025-08-06:
I think a PID loop could be useful for most remotes, not just RemoteROS2 (atleast RemoteUart and RemoteUartBus also). So my plan is to put to it in a new "virtual" function so that it can be used in combination with whatever Remote you want to use (or keep using the current code). My plan for the PID version is to change the unit of speed to rad/s. Then based on iOdom calculate the actual speed in rad/s and then let the PID control do its magic to adjust pwmMaster/Slave so that the actual rad/s follows the desired rad/s.
I agree that PID have been implemented many times before and Deepseek or Gemini should be able to help to generate the code. I have no experience with PID loop so I am not sure if main loop frequent enough or if it is better to put it in calculateBldc. Maybe Deepseek/Gemini can help with that too...

WestlingPi, 2025-08-06:
I only plan to do pid based on speed, not torque or something else.
Currently I worry mostly about the resolution of hall sensor. 90 ticks per revolution?
If not iOdom, do you have any suggestions what I should base actual speed estimate on? iOdom is just hall ticks, right?
I have no experience with PID control from before so I am a bit naive. I think I will just start with something simple and see if works (learn something). Also, my usecase is a slow moving robot (max 2m/s, typically 0.5m/s?) so I think it will work (or maybe it is just me being naive).

RoboDurden, 2025-08-06:
Okay, `realSpeed` is calculated based on one electrical turn which is 6*4° = 24°
15 of the electrical turns will make the 360° mechanical full rotation.
So indeed, you could increase responsivness with counting buzzerTimer for every single hall step.
But this might get very noisy for high speeds.
Best approach would be to have a `speedCounterSlow` used for low speeds, and switch to `speedCounter`  when `speedCounterSlow` gets below say 50 for such one hall step.

This might work:

```c++
revs = (PWM_FREQ/90) / (float)speedCounterSlow;  // 90 = 360/4
..
revs = (PWM_FREQ/15) / (float)speedCounter;   // 15 = 360/(6*4)
```

Assume 1 rev/s and PWM_FREQ = 12000
Then on full rotation will cause 12000 ticks. So for one hall step speedCounterSlow = 12000/90

(12000/90)/ (float)(12000/90)  = 1.0 :-)

The PID should better run on uint16 or uint32, instead of float!
So you might want to go for a 

```c++
revs32 =  (SCALE32*PWM_FREQ/90) / speedCounterSlow;
..
revs32 =  (SCALE32*PWM_FREQ/15) / speedCounter;

```

SCALE32 should be based on a max speed which could be 14 rpm * 50V * 60s = 42000 which is the max rev/s, a bldc motor should be able to reach.
revs32 must not overflow, so not get greater than 4294967295.
speedCounter can go down to 1, so 4294967295  = SCALE32*PWM_FREQ/15
which leads to SCALE32 = 4294967295/(WM_FREQ/15)

RoboDurden, 2025-08-06:
@WestlingPi  i am currently working to implement such a speed pid in bldc.c, so stay tuned.

RoboDurden, 2025-08-06:
@WestlingPi  code is online.
Will make a youtube video now.

Only test `#define DRIVING_MODE 1	//  0=pwm, 1=speed in revs*1024, (not yet: 3=torque, 4=iOdometer)`
with RemoteDummy and constant-current powersupply set to max 2A 

The deepseek pid is only achiving 80% of the revs/s, but these indeed get controled to be constant.
There is an initializition bug that makes the motor shake when on first startup.

But now you can replace the pid code easily with whatever you have in driver.c

WestlingPi, 2025-08-06:
Great, I will check the code...

RoboDurden, 2025-08-06:
The shaky start is because RemoteDummy does its sine speed based on msTicks:

`		speed = ((iMax*30)/1000) * (ABS((	((int32_t)msTicks/90+100) % 400) - 200) - 100);		// for longer max speed
`

Therefore, after the board has start up, the first speed command jumps directly to 750 = 0.75 revs:

[image]

I do not fix that yet because that is a good test scenario to tune the PID parameters.
The green graph shows the output of the pid filter (which equals the input of the rank 13 low pass) `bldc_inputFilterPwm` :-(
The orange curve shows that input after being filtered to `bldc_outputFilterPwm` .
After roughly 1 second, the pid loops begins to stabelize.

RoboDurden, 2025-08-08:
I do NOT like your suggestions :-(
I spend a day to setup a framework where you can implement your own

`			return PID_Update2(&pid, input, revs32);
`

If you don't like it, than do whatever you like. And if your PID implementation works better i will see how i can import your code into my framework.

So stop dreaming and focus on your pid code.

WestlingPi, 2025-08-08:
No worries, I think I can work with the current stucture if you prefer. But I believe most of the work is done. The PID code is generated by some AI and you have done the integration into the code base. I think it is mainly tuning of PID parameter, testing and maybe some minor tweaks that remains? I can help with that (but first in need to reassemble my test setup that I dismantled after my vacation). Thanks for your work.

RoboDurden, 2025-08-09:
Okay, PID from ChatGPT5 seems to work and is online.
But it was a very frustrating chat with many many prompts and luckily, right with the last answer before free quota was used up, the motor started spinning.
I still have to scale the target by a factor 14/8 to somehow match target with real speed, but pid controller now works for ultra low speeds like 0.2 revs/s: 

[image]

RoboDurden, 2025-08-09:
But speed drops a bit under load and gets back to "normal", when i remove my hand from the motor:

[image]

RoboDurden, 2025-08-09:
Tomorrow i will continue the unwanted dispute with ChatGpt5 so that the PID really hits the target speed 100% without silly factor 12/8 (that only works for a specific target speed anyway).

RoboDurden, 2025-08-10:
I will think about masterslave steer when the pid for one motor is working nicely.
This already is to slow for my currently tested pid:

```c++
	if (buzzerTimer%20==0)	
		bldc_inputFilterPwm = Driver(iDrivingModeOverride,iBldcInput);		// interpret the input as pwm/speed/torque/position.
```

And as long as the pid is done in calculateBldc, i do not really care if it is called every cycle or only every 10th cycle.
Only if the pid would run in the main loop, this would help usart as this has a lower interrupt priority as calculatebldc.
But my current pid code is now tuned for 12 kHz.

RoboDurden, 2025-08-10:
I have given up on ChatGpt :-(
What a waste of time, in the time wasted on stupid ai, i could have coded my own speed control with my good old physics approach of a damped harmonic oscilator ten times :-(

WestlingPi, 2025-08-10:
Like I said, I dont think the problem is the implementation of PID_Update() but the tuning of the P, I, D parameters... Sure there are other ways of implementing closed loop control, but PID seems to be the commonly accepted solution to the problem. I hope you learned something, otherwise I woud agree it is a waste of time :-)

RoboDurden, 2025-08-10:
No, i learned nothing. waste of time :-( Shitty chat bots.
If i have read the wikipeadia article on PID, i would have learned more. But i do not want to learn about PID !

Have uploaded the latest chatGpt5 pid, which is a bit better then yesterdays.
I will stop now working on constant speed. I don't need it anyway.

Please tell me when you have a pid that indeed reaches target speed from 0.3*1024 to 3.0*1024 = 3.0 revs/s and doesnt stutter.
Otherwise, leave me alone :-)

WestlingPi, 2025-08-10:
Understood. :-)
I can take over working on PID (but at a slower pace than this weekend, otherwise my family wont be happy).

RoboDurden, 2025-08-10:
YESSS, i could not resist to give Gemini a chance, and with this prompt, Gemini 2.5pro did give a working PID in the very first response:

> Please give me the c code for a PID controller that outputs an int16_t pwm value ranging from -1250 to +1250 for a bldc motor based on a int32_t revs32 speed value that will be revs/s*1024. 
The PID controller and the revs32 will be updated at a rate of PWM_FREQ = 12000 Hz. 
The code will run on a GD32F130, so no floating point engine.
The PID controller code should only work with  integers but no 128bit integers and no 64bit division.
An Init function should accept the usual float values like void PID_Init(PIDController *pid, float kp, float ki, float kd, int16_t min_pwm, int16_t max_pwm, float max_i) 
This shall be the update function being called at 12 kHz: int16_t PID_Update(PIDController *pid, int32_t setpoint, int32_t actual) 

Code is online.
Works at 25V from 0.15 revs/s up to 5 revs/s.
There is some oscillations when the target speed jumps, but i am confident that this could be solved with tuning the pid init values and reducing the final lowpass filter rank from 13 to 10 or even less. ChatGpt_2 could do without that bldc.c low pass filter entirely. 
But i want to keep that filter as it is needed for emergency-shutdown.
So it might be better to slow down the Gemini pid so that that low pass filter does not confuse the fast pid.

@WestlingPi , please test it. If it also works for your setup, then we can discuss the masterslave-steer issue :-)

RoboDurden, 2025-08-10:
Okay, i also added constant torque in NewtonMeter:

```c++
//#define SLOW_SINE

void RemoteUpdate(void)
{
	int32_t iMax = 250;	// pwm 300 as default
	switch(iDrivingMode)	//  0=pwm, 1=speed in revs*1024, (not yet: 3=torque, 4=iOdometer)
	{
		case 1: iMax = 0.3 *1024; break;	// 1.5*1024 = max speed 1.5 revs/s
		case 2: iMax = 5.0 *1024; break;	// 1.5*1024 = max speed 1.5 Nm (Newton meter)
	}
	
	#ifdef MASTER_OR_SINGLE
		steer = 0;
		#ifdef SLOW_SINE
			speed = ((iMax*30)/1000) * (ABS((	((int32_t)msTicks/90+100) % 400) - 200) - 100);		// for longer max speed
		#else
			speed = ((iMax*16)/1000) * (ABS((	((int32_t)msTicks/30+100) % 400) - 200) - 100);
		#endif
	
		speed = CLAMP(speed , -iMax, iMax);
```

Did not need to change any of the Gemini PID constants :-)
Problem only was to calculate the torque based on  `// iTorque = 0.8 * batteryVoltage * currentDC * (90/2*PI) *speedCounterSlow / (PWM_FREQ  *(iOdom-iOdomLast))`

Seems to work:

[image]

Last thing would be using the pid for position control iOdom..

RoboDurden, 2025-08-10:
Okay, with position control, DRIVING_MODE is complete now :-)

[image]

iOdom = +90 is equivalent to a 360° rotation. So now 4° position control for robots :-)

With such a strong motor i already dream for some years to 3d print a 1:32 differential planetary gearbox.
That would allow for 1/8 ° degree resolution.
With a 1 meter long robot arm, that would be 2mm positioning. And +-4mm error for a 2 axis robot arm with a reaching diameter of 4 meters.

For iOdom position control, i did need to tune the Gemini PID parameters:

```c++
	case 2:	// input will be taken as NewtonMeter * 1024 
		#ifdef PID_Gemini
			PID_Init(&pid, 0.2f, 1.0f, 0.0005f, -1250, 1250, 1250.0f);		
		#endif
	case 3:	// input will be iOdom position in hall steps = 4°
		#ifdef PID_Gemini
			PID_Init(&pid, 50.0f, 1.0f, 0.005f, -1250, 1250, 1250.0f);		
		#endif
```

I will remove the Deepseek and ChatGpt5 pid codes in the near future.

WestlingPi, 2025-08-10:
Great! I will rebase my minor updates and see if they are still valid with the latest code. But it is getting too late now so I save it for another day.

RoboDurden, 2025-08-11:
[[image]](https://youtu.be/VcAJosAGlok)

RoboDurden, 2025-08-17:
I am currently fully busy implementing a 15% speed boster..

RoboDurden, 2025-08-17:
uncomment `//#define DEBUG_RX    // additional hoverboard-rx debug output` to see if any bytes are incoming.

WestlingPi, 2025-08-20:
I didn't see the new issue yesterday, I will reply there instead about the pilot (I agree null pilot is probably better for ROS usecase)...

WestlingPi, 2025-08-21:
I am still working on a setup where I can troubleshoot why it is not working on my 2.2.3 (10" wheel) and tune PID parameters.

RoboDurden, 2025-08-21:
StmStudio or McuViewer !
10" wheel should not make a difference.

RoboDurden, 2025-08-27:
Okay i have connected my 2.2.7 test setup which is a stm32f103 Gen1 board with right motor chosen.
And i can confirm that the buzzer frequency is far to low and the constant speed with bldc_bc is not able to achieve revs/s 1.0

I don't want to spend much time on this, as i need to move forward with my new liion battery for my old little solar bike camper.
But i i will be happy to test for you in the next few days.

RoboDurden, 2025-08-27:
Well, after a little push, constant speed PID is working for 1.0 revs. But MCU frequency is definitely to low:

https://github.com/user-attachments/assets/a4f54fe7-a55a-429c-bf97-540544b9a5a8

RoboDurden, 2025-08-27:
Oh, the Keil binary gets the buzzer sound right, but the PID still fails:

https://github.com/user-attachments/assets/e65be9bf-9dd9-40c2-baa2-1d661be88998

So I think I should do a bit of debugging.

Still, getting the MCU clock right for platformio binary should be your first step.

RoboDurden, 2025-08-27:
> Did @hoverboardhavoc already have the clock configuration code for STM32F103? Why not include it in the gen2.x code?

Well his changes have been added, maybe they don't work anymore:

```
platform_packages = framework-spl-gd32@https://github.com/hoverboardhavoc/gd32-pio-spl-package.git#hoverboardhavoc/add__PIO_DONT_SET_CLOCK_SOURCE__
...
    -D __SYSTEM_CLOCK=72000000UL  # Add explicit system clock definition
    -D __SYSTEM_CLOCK_72M_PLL_IRC8M_DIV2=72000000UL
    -D __PIO_DONT_SET_CLOCK_SOURCE__
```

WestlingPi, 2025-08-27:
My Keil binary is running calculateBldc()/buzzerTimer at 4 kHz on my 2.2.3 with STM32F103.
Deepseek also claims that there is difference TIMER_COUNTER_CENTER_BOTH between STM32F103 and GD32F103 (interupt triggered twice as often on GD32F103) 
Now I have more than enough things to debug when get home from work... :-)

WestlingPi, 2025-08-27:
I have tested with a modified TestSpeed.ino:

```
millis: 100	iSpeed: 0	msTicks: 3797	buzzerTimer: 14185	msTimer: 2532
millis: 100	iSpeed: 0	msTicks: 3897	buzzerTimer: 14585	msTimer: 2598
millis: 100	iSpeed: 0	msTicks: 3997	buzzerTimer: 14985	msTimer: 2666
millis: 100	iSpeed: 0	msTicks: 4097	buzzerTimer: 15385	msTimer: 2732
millis: 100	iSpeed: 0	msTicks: 4197	buzzerTimer: 15785	msTimer: 2798
millis: 100	iSpeed: 0	msTicks: 4297	buzzerTimer: 16185	msTimer: 2866
millis: 99 	iSpeed: 0	msTicks: 4397	buzzerTimer: 16586	msTimer: 2932
millis: 100	iSpeed: 0	msTicks: 4497	buzzerTimer: 16986	msTimer: 2998
```
`millis` is from ESP32 and shows time difference on ESP32 in ms, i.e. 100ms between each sample.
`msTicks` is from STM32F103 and shows as expected 100 difference between each sample.
`buzzerTimer` shows only 400 difference between each sample. I am expecting 1200 since I use BLDC_SINE.
I added `msTimer` which is stepped in `TIMEOUT_IrqHandler()`. It only shows about 66 difference between the samples. I am expecting 100.

For buzzerTimer i get 1/3 of expected and for msTimer i get 2/3 of expected. Maybe if I change timerBldc_paramter_struct.alignedmode = TIMER_COUNTER_CENTER_BOTH to TIMER_COUNTER_CENTER_DOWN, then buzzerTimer will also be 2/3 of expected (according Deepseek)? To be continued...

RoboDurden, 2025-08-27:
Why don't you use StmStudio or McuViewer?

RoboDurden, 2025-08-27:
Don't know if this helps,but a factor two can be achieved with putting calculateBLDC back the interrupt_flag_get clause:

```c++
	CalculateBLDC(); //moved behind flag_clear by Deepseek, Safe: NVIC blocks re-entrancy

	if (TARGET_dma_interrupt_flag_get(DMA_CH0, DMA_INT_FLAG_FTF))
	{
			TARGET_dma_interrupt_flag_clear(DMA_CH0, DMA_INT_FLAG_FTF);
	}	
```

RoboDurden, 2025-08-27:
> Because I wanted to compare against another clock (ESP32) and not just relative within the STM32F103.

But with realtime monitoring you simply take a stop watch to verify that msTicks is counting correctly..
Than you can compare buzzertimer to msTicks.

WestlingPi, 2025-08-27:
I will try `CalculateBLDC();` inside `if (TARGET_dma_interrupt_flag_get(DMA_CH0, DMA_INT_FLAG_FTF)` tomorrow...

On my 2.2.3 I get the following values:
```
iTimeSecond=3000
iPwmTicks=125
iAdcTicks=250
```

By why is iPwmTicks != iAdcTicks? I dont understand...
@RoboDurden I see you got different values of iPwmTicks vs iAdcTicks on your 2.2.1 setup but same on 2.2.7.
I would expect them to be same...

RoboDurden, 2025-08-28:
Good morning.
Ps. There is a factor 4 i had to skip in setting the target speed that I never understood why:

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-GD32/blob/c953cbc8c3d13c27fc0f2cdca15bbc7a46b1b2d7/HoverBoardGigaDevice/Src/bldc.c#L120

Speed should be *1024 which is a <<10 left shift. But revs32 is calcult rank 12 which would be >>12

Maybe the code indeed runs 4 times to slow..

Different counts for the interrupt handlers are due to interrupt priorities. ADC has lower priority then timer0, I think.
Or calculateBldc has not finished when next ADC dma triggers interrupt handler..

That is why I limit pwm_rate to 12 kHz for bldcSine. Whose extra external interrupts on the three hall sensors have higher priorities then ??
Search vor nvic in Setup.c and bldcSine.c

Now for breakfast.

RoboDurden, 2025-08-28:
Indeed, my 2.1.1 g32f103 seems to run 4x to fast:

[image]

The drop from 64k to 48k for adc probably is because calculateBldc takes to long and one dma_finished interrupt gets skipped.

bldc.c
```
	iBug = PWM_FREQ;
	if (msTicks > iBuzzTime)
	{
		iBuzzTime = msTicks + 1000;
		iBuzzRate = iBuzzCounter;
		iBuzzCounter = 0;
	}
	else iBuzzCounter++;
```

Same code for TARGET_TIMER0_BRK_UP_TRG_COM_IRQHandler and TARGET_DMA_Channel0_IRQHandler.
I have uploaded that code to github.

Results for 2.2.7 = stm32f103

[image]

Here my 2.1.4 = gd32f130

[image]

msTicks correctly counts 1000 increments per second.

2.1.4 with bldcSine:

[image]

calculateBldc moved inside the interrupt_flag clause:

[image]

I do not understand why timer0 currectly fires 4x as often.

Here platformIo binaries:

2.1.4 bldcSine
[image]

2.1.4 bldcBC
[image]

2.2.1 bldcBC
[image]

Now the rate is 1/4 !!! No wonder that PID fails.

Before PID, wrong rate only affected buzzer tone, which i did not care for. But PID is sensitive to how often 
`	bldc_inputFilterPwm = Driver(iDrivingModeOverride,iBldcInput);		// interpret the input as pwm/speed/torque/position.
`gets called per second.

RoboDurden, 2025-08-28:
With the above changes, i can also use the correct speed mapping in bldc.c :-)

```c++
void SetBldcInput(int32_t input)
{
	switch (iDrivingMode)
	{
	case 0:	// pwm
		iBldcInput =  BLDC_TIMER_MID_VALUE*(input/1000.0);	// thanks to WizzardDr, bldc.c: pwm_res = 72000000 / 2 / PWM_FREQ; == 2250 and not 2000
		iBldcInput =  CLAMP(iBldcInput,-BLDC_TIMER_MID_VALUE, BLDC_TIMER_MID_VALUE); 	
		return;
	case 1:	// speed in revs*1024 = revs<<10
		iBldcInput = input << (REVS32_SHIFT-10);
		return;

```

RoboDurden, 2025-08-28:
My Gemini free quota exceeded with

<blockquote>
I only need TIMER0_BRK_UP_TRG_COM_IRQHandler to start adc conversion which triggers DMA_Channel0_IRQHandler when finished.
Can i do without TIMER0_BRK_UP_TRG_COM_IRQHandler and trigger adc conversion with an different TIMER0 setup ?
</blockquote>

<blockquote>
Yes, you absolutely can, and it's the recommended and more efficient way to do it.

You can set up a hardware trigger so that TIMER0 directly starts the ADC conversion without needing the CPU to intervene through an interrupt handler. This method gives you perfect timing, has zero CPU overhead, and simplifies your code.
...
</blockquote>

But then it gave me code not matching out old 2.0.1 library.
My prompt for tomorrow:

```
I am compiling with library version 2.0.1 and get undeclared errors for ´timer_master_output_trigger_source_config´ and 'TIMER_TRIG_OUT_SRC_O3CPRE'.
I do not know how to correct this code line.
And another undeclared error for  ´ADC_EXTTRIG_REGULAR_T0_TRGO´.
These are the possible defines from my gd32f1x0_adc.h :
/* external trigger select for regular  channel */
#define CTL1_ETSRC(regval)              (BITS(17,19) & ((uint32_t)(regval) << 17))
#define ADC_EXTTRIG_REGULAR_T0_CH0      CTL1_ETSRC(0)                       /*!< TIMER0 CH0 event select */
#define ADC_EXTTRIG_REGULAR_T0_CH1      CTL1_ETSRC(1)                       /*!< TIMER0 CH1 event select */
#define ADC_EXTTRIG_REGULAR_T0_CH2      CTL1_ETSRC(2)                       /*!< TIMER0 CH2 event select */
#define ADC_EXTTRIG_REGULAR_T1_CH1      CTL1_ETSRC(3)                       /*!< TIMER1 CH1 event select */
#define ADC_EXTTRIG_REGULAR_T2_TRGO     CTL1_ETSRC(4)                       /*!< TIMER2 TRGO event select */
#define ADC_EXTTRIG_REGULAR_T14_CH0     CTL1_ETSRC(5)                       /*!< TIMER14 CH0 event select */
#define ADC_EXTTRIG_REGULAR_EXTI_11     CTL1_ETSRC(6)                       /*!< external interrupt line 11 */
#define ADC_EXTTRIG_REGULAR_NONE        CTL1_ETSRC(7)                       /*!< software trigger */
/* external trigger select for inserted channel */
#define CTL1_ETSIC(regval)              (BITS(12,14) & ((uint32_t)(regval) << 12))
#define ADC_EXTTRIG_INSERTED_T0_TRGO    CTL1_ETSIC(0)                       /*!< TIMER0 TRGO event select */
#define ADC_EXTTRIG_INSERTED_T0_CH3     CTL1_ETSIC(1)                       /*!< TIMER0 CH3 event select */
#define ADC_EXTTRIG_INSERTED_T1_TRGO    CTL1_ETSIC(2)                       /*!< TIMER1 TRGO event select */
#define ADC_EXTTRIG_INSERTED_T1_CH0     CTL1_ETSIC(3)                       /*!< TIMER1 CH0 event select */
#define ADC_EXTTRIG_INSERTED_T2_CH3     CTL1_ETSIC(4)                       /*!< TIMER2 CH3 event select */
#define ADC_EXTTRIG_INSERTED_T14_TRGO   CTL1_ETSIC(5)                       /*!< TIMER14 TRGO event select */
#define ADC_EXTTRIG_INSERTED_EXTI_15    CTL1_ETSIC(6)                       /*!< external interrupt line 15 */
#define ADC_EXTTRIG_INSERTED_NONE       CTL1_ETSIC(7)                       /*!< software trigger */
```

Directly triggering adc conversion without the pwm handler should(might) fix the 32f103 issue with only 1/4 calls..

RoboDurden, 2025-08-28:
Okay, i have continued with ChatGpt and after removing all the dma stuff and converting adc to "inserted adc", i now have 16 kHz for both gd32f130 and gd32f103.

Here my 2.2.1 gd32f103

[image]

iPwmRate is no longer needed as adc is directly triggered by the BLDC TIMER0
And adc value are read in the adc event handler

```c++
extern adc_buf_t adc_buffer;
#ifndef TARGET_ADC_CMP_IRQHandler
	#error "TARGET_ADC_CMP_IRQHandler not defined for active target in target.h"
#endif
void TARGET_ADC_CMP_IRQHandler(void)
{
	if (TARGET_adc_interrupt_flag_get(ADC_INT_FLAG_EOIC)) 
	{
		adc_buffer.v_batt = TARGET_ADC_IDATA0;	// first inserted channel result
		adc_buffer.current_dc = TARGET_ADC_IDATA1;	// second inserted channel result

		if (msTicks > iAdcTime)
		{
			iAdcTime = msTicks + 1000;
			iAdcRate = iAdcCounter;
			iAdcCounter = 0;
		}
		else iAdcCounter++;
		
		CalculateBLDC(); //moved behind flag_clear by Deepseek, Safe: NVIC blocks re-entrancy
		
		TARGET_adc_interrupt_flag_clear(ADC_INT_FLAG_EOIC);
	}
}
```

I can not push the new code that easily because of the TARGET_ macros which i also have to write for GD32E230.
Inserted adc only allows for max 4 channels on gd32f130 according to ChatGPT. 
So RemoteADC with its two additional joystick inputs might still work.
But ChatGpt suggested to go for a hybrid approach to do additional adc channels with the good old DMA. 
And RemoteADC does not need 16 kHz sampling anyway.
But now, chatGpt quota is also empty.

RoboDurden, 2025-08-28:
Well, i broke the Keil binary for 2.2.1 and forgot what i changed.
2.2.1 Platformio wroks but again with way to low adc/buzzer rate:

[image]

2.1.4 works both for Keil and PlatformIO.

I am tired now.
How can i turn my local code into a github branch ? Because the current adc_insert code can not replace the root repo.

RoboDurden, 2025-08-28:
But we can also stay with the main branch:

2.1.4 both Keil and PlatformIO binaries work 16 kHz
2.2.1 only Keil works with 16 kHz and PlatformIO again bad:

[image]

Maybe platformio library has a bug for adc that takes so long that only less then 4 kHz are possible..

WestlingPi, 2025-08-28:
I am still trying to figure out how the clock tree is configured in Keil uVision (and PlatformIO). After some Gemini chating it seems to be the `SystemInit()` defined in RTE/Device/GD32F103C8/system_gd32f10x.c (or the likes for other MCUs) that does the clock configuration. In from `SystemInit()` a `system_clock_config()` that use the #define that is "removed" from @hoverboardhavoc fork.

@hoverboardhavoc Do you know how it is supposed to work? Should you define one of the defines to select one of the clock configurations? Could you undefine `__SYSTEM_CLOCK_108M_PLL_HXTAL` from compile command instead of your patch? Gemini suggested to edit the system_gd32f10x.c but that seems a bit strange way of working to me... Is there some documentation on how to configure clock for GD32 for a Keil (or PlatformIO) project? It seems like something all projects would want to do...

RoboDurden, 2025-08-28:
?? with the current main branch, msTicks runs correctly for my 2.2.1 gd32f103 and platformio binary. Is is the adc conversion that brings the calculateBLDC rate down.

WestlingPi, 2025-08-28:
@RoboDurden What is your SystemCoreClock on 2.2.7 = stm32f103 (Gen1 board)?

My stm32f103:

[image]

RoboDurden, 2025-08-29:
My Gen1 setup:

[image]

And yes, the halfing of my iAdcRate is not because adc takes to long but CalculateBLDC.
And it is this simple line:

`		currentDC = ABS((adc_buffer.current_dc - offsetdc) * MOTOR_AMP_CONV_DC_AMP);
`

That is old floating point code !! So the software float implementation of gd32f103 might be way worse than the gd32f130 implementation.
So good that i always insited on integer-only code for my PID and BLDC_SINE.
Now i dream that i only need to convert the old currentDC to currentDC32.

@WestlingPi , i see in your StmStudio screenshot, that iPwmRate is already half of PWM_FREQ.
Try disabling 
´		if (interrupt_toggle)		// Only execute every second call as libray/hardware will trigger on up AND down, ignoring timerBldc_paramter_struct.alignedmode = TIMER_COUNTER_CENTER_DOWN
´
in it.c
But in setup.c you should have ´alignedmode = TIMER_COUNTER_CENTER_BOTH´ and we only want one call per period. So i do not really understand why you end up with only half the PWM_FREQ.

RoboDurden, 2025-08-29:
With neither CURRENT_DC and VBATT being defined and removing the last line that does float computation

`		//testing without any flots  realSpeed	= (realSpeed32_reg >> RANK_realSpeed32) / 1024.0;
`
My 2.2.7 rate can go up to 12 kHz, but only with no 25V. With 25V and the motor spinning, the rate halfs to 6 kHz as one adc-finished-handler gets skipped.
This was for `DRIVING_MODE 0` and `BLDC_BC`.
With BLDC_SINE, it starts of with 6 kHz but when my linear interpolation kicks in (i think), the rate drops to 1/3 = 4 kHz. So two handler calls get skipped.

For gd32f130, the platformIO binaries work just as fine as the Keil binaries. And SystemCoreClock is 72 MHz for both platformio and keil binaries.
For gd32f103, only the Keil binaries work nicely, and platformio binaries work poorly with SystemCoreClock not matching the 72MHz of the Keil binaries but show 80 Mhz for 2.2.7 (stm32f103) and 2.2.1 (gd32f103)

As the Platformio binaries use the same

```c++
platform = https://github.com/CommunityGD32Cores/platform-gd32.git
platform_packages = framework-spl-gd32@https://github.com/hoverboardhavoc/gd32-pio-spl-package.git#hoverboardhavoc/add__PIO_DONT_SET_CLOCK_SOURCE__

```
for f130 and f103, indeed the platformio clock settings for 103 might cause the poor performance (for some reason).

f103 has additional line `-D __SYSTEM_CLOCK=72000000UL  # Add explicit system clock definition` in its `[env:genericGD32F103C8]` section, but removing it does not have an effect, i still get 80 MHz.
@hoverboardhavoc  might immediately know, why 

```c++
    -D __SYSTEM_CLOCK_72M_PLL_IRC8M_DIV2=72000000UL
    -D __PIO_DONT_SET_CLOCK_SOURCE__
```

have no effect for f103 target in his 

```c++
platform = https://github.com/CommunityGD32Cores/platform-gd32.git
platform_packages = framework-spl-gd32@https://github.com/hoverboardhavoc/gd32-pio-spl-package.git#hoverboardhavoc/add__PIO_DONT_SET_CLOCK_SOURCE__
```

@WestlingPi maybe you can hard code the 72Mhz somewhere in the system files as your chatbot proposed to test if this indeed resolves all the skiped handler calls.

WestlingPi, 2025-08-29:
Great! I will try it this evening. Is the fix for PlatformIO only? Does it work for Keil aswell?

WestlingPi, 2025-08-29:
I have done some testing on my 2.2.3 (stm32f103) with latest in gen2.x repo from today. I am getting better rates than before, but not all the way yet:

[image]

Now I get iPwmRate 16000, but I care mostly about iBuzzRate since buzzerTimer is used to calculate revs32 for speed PID (which is what I was trying to do when I fell into this clock configuration hole). If iBuzzRate is not close to PWM_FREQ, revs32 calculation will be wrong and then wrong input and kp, ki, kd parameters tuned for other revs scaling will make the PID not work....

@RoboDurden Are you sure you got 72 MHz SystemCoreClock with 2.2.7 = stm32f103 (Gen1 board)? According to Gemini Pro, highst possible system clock speed for a stm32f103 with internal oscilator (HSI) as clock source is 64 MHz (with gd32f103 it is possible with 72MHz due to wider range of PLL multiplier (max 108 MHz for gd32f103)). The reason I get 12 MHz on stm32f103, I think, is because the multiplier for 72MHz (18 * 4) is out of range for stm32f103 and instead interpeted as 3 and 3 * 4=12MHz. But now I am out of Gemini Pro quota, and Deepseek have only given me bullshit all evening. I will try again to increase clock speed to 64 MHz tomorrow...

Another alterative would be to use iBuzzRate in the revs32 calculations to compensate for the lower bldc loop rate. But isnt 8kHz a little low for the bldc loop? But I am not there yet, I have not given up on changing clock speed to 64 MHz for stm32f103 yet (but I think Gemini Pro have proven me wrong on trying to acheive 72Mhz on stm32f103)...

RoboDurden, 2025-08-30:
Disable calculateBldc in event handler to see if your then get iAdcRate 16 kHz. Then the too long execution time of calculateBldc is the reason.

As you already get 16 kHz for iPwmRate, interrupt_toggle is doing it's job correctly.

Yes, we should keep iBuzzRate to make revs32 truly independent of this mess. 
But halfing of calculateBldc should really be happening.

RoboDurden, 2025-08-30:
Ps.  Maybe buzzerCounter is not the fastest counter anyway. The internal Timer0 register that is counting up and down might give a far higher time resolution for revs32 and bldcSine linear interpolation.
The TimeoutTimer is counting only upwards...
Or simply add another Timer.

RoboDurden, 2025-08-30:
My Gen2.2.7 = stm32f103RC (RC = 256 Kbytes of Flash memory, 72 MHz CPU)  with platformIO binary:

[image]

Everything fine.

RoboDurden, 2025-08-30:
I just recently added a [env:genericGD32F130C8_macbookARM] to the platformio.ini, so i we can add a [env:genericSTM32F103C8] as well. And as the gd32f103 env already contains clock defines, we should stick to that place, i think.

But your 48 MHz do not really explain why 72Mhz is fine on my Gen1 2.2.7 setup, both for Keil and PIO binaries ?
https://www.st.com/resource/en/datasheet/stm32f103c8.pdf
> The STM32F103xx medium-density performance line family incorporates the
high-performance Arm® Cortex®-M3 32-bit RISC core operating at a 72 MHz frequency,
high-speed embedded memories (Flash memory up to 128 Kbytes and SRAM up to
20 Kbytes), ...

The random numbers in your screenshot show, that the code did not startup. As you do F5 do get the debug binaries, you should see that debugging does not stop at main(), but either before or never. Then hit the break button and you probably will see that the code execution hangs in a final while(1); loop because cpu clock could not be initialized correctely. Happens for my gd32e230 hardware sometimes..

WestlingPi, 2025-08-30:
Max 72MHz is for external oscilator. When internal oscilator (HSI) is used there is first a /2 to prescaler before the PLL multiplier. Max for PLL multipler for stm32f103 is 16. So (8MHz/2)*16=64MHz.

If your Gen1 2.2.7 can do 72MHz, then I would GUESS that it is a fake. I have some chinese "bluepills" with stm32f103. But you can not trust the STM32 label on them. Maybe the same is the case for your 2.2.7?

Also I think there is an "Option 2b": Add also a `#include "../../Inc/defines.h"` to system_gd32f10x.c. Then 48MHz or 72MHz can be defined in defines_2-2-x.h (and also a default to 72MHz if undefined in defines_2-2-x.h). Then there is no need for a new Keil target. I prefer this option. I want something that works with Keil.

RoboDurden, 2025-08-30:
Indeed:
>When the HSI (=high-speed internal clock signal) is used as a PLL clock input, the maximum system clock frequency that can be achieved is 64 MHz.

Gen1 does not have an external oscillatior and i have checked the msTicks with a stop watch.  So SystemCoreClock which is 72M is ignored by the system ?

Please no modifications to system files like system_gd32f10x.c. I think these get easily replace while making changes to the RTE runtime enviroment ?

Better add a new Keil target and add the define there:

[image]

```c++
#ifdef __SYSTEM_CLOCK_48M_PLL_IRC8M
uint32_t SystemCoreClock = __SYSTEM_CLOCK_48M_PLL_IRC8M;
```
comes first, so the default 72M are ignored.

With this i get 
[image]
on my Gen1 board and msTicks again verified with stop watch.
But 48MHz is not optimal anyway ? How would you achieve 64MHz ?

[3 more replies]