RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #123: LKS32

unixdt, 2025-07-10:
I have a board exactly like the LKS32 V2 in this repo but I don't see it in the Keil files. Am I missing something? Is this board supported?

RoboDurden, 2025-07-10:
We don't have lks support 
But I still collect photos of every gen2 board for the database. If lks boards become more often, I might ask deepseek or Gemini to add code..

NaniGabber, 2025-09-08:
have also, lks32mc081c8t8 
Hoverboard name:Smart Balance 10.5 Elite Lux
Board: dualboard, slht 228499-B
drivers: u3115s
gyroscope: MPU-6887

NaniGabber, 2025-09-08:
[image]
This one. Keil package is avialable on this mcu. There are even keil project examples.

RoboDurden, 2025-09-08:
If you get a Timer0 example to compile I am willing to try adding the code to gen2.x

PR77, 2026-10-03:
Any updates on officially support for the LKS chip? I have sourced a board with the LKS32MC088 controller. PCB layout looks more-or-less identical to LKS V2, apart from the micro markings.

Are there any pre-compiled binaries available?

I-hate-2FA, 2026-11-03:
i can (try to) take oficial support for lks32, if anyone have one can send me the sample board set, will port the one from mm32

I-hate-2FA, 2026-11-03:
the only difference really is integrated opamp(mm32spin05 also have on spin05pfop), and integrated PGA on allot of the modelsfor the dc bus current sense

RoboDurden, 2026-11-03:
@I-hate-2FA  that would be great. I would add USD 10 for you if you can order an LKS board/kit.
Maybe you can ask here if they have a LKS32 board: https://www.alibaba.com/product-detail/Balance-Scooter-Control-Board-PCBA-Solution_1601826318496.html

I-hate-2FA, 2026-11-03:
marking is too unclear to tell, i need garunteed one, you have worked with me before you know everything is complex

I-hate-2FA, 2026-11-03:
i need the most common one seen so its actually useful i do have allot of token but it should be used wisely on useful project people actually want

I-hate-2FA, 2026-11-03:
[image]

[image]
this is more promissing where LKS is clearly visible but not what mcu

I-hate-2FA, 2026-11-03:
this is the cheapest brand new replacement board now so should be common enough

I-hate-2FA, 2026-11-03:
[image]

I-hate-2FA, 2026-11-03:
intrestingly, the board is 3.3v not 5v, perhaps the imu is too t roublesome, as all the LKS can handle up to 5.5v