RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #129: framework complete ?? Remote_W -> Pilot_X-> Driver_Y -> BLDC_Z

RoboDurden, 2025-08-19:
With my Hoverbike project, i have introduced the concept of a pilot, so user code running on the hoverboard firmware can be seperated from the main.c code, with the PilotDefault code not (yet?) moved to a pilotDefault.c file:

```c++
			#ifdef PILOT_XY
				Pilot(&pwmMaster,&pwmSlave);
			#else
				if (iDrivingMode == 0)
				{
					// Calculate expo rate for less steering with higher speeds
					expo = MAP((float)ABS(speed), 0, 1000, 1, 0.5);
					
					// Each speedvalue or steervalue between 50 and -50 (STAND_STILL_THRESHOLD) means absolutely no pwm
					// -> to get the device calm 'around zero speed'
					scaledSpeed = speed < STAND_STILL_THRESHOLD && speed > -STAND_STILL_THRESHOLD ? 0 : CLAMP(speed, -speedLimit, speedLimit) * SPEED_COEFFICIENT;
					scaledSteer = steer < STAND_STILL_THRESHOLD && steer > -STAND_STILL_THRESHOLD ? 0 : CLAMP(steer, -speedLimit, speedLimit) * STEER_COEFFICIENT * expo;
					
					// Map to an angle of 180 degress to 0 degrees for array access (means angle -90 to 90 degrees)
					steerAngle = MAP((float)scaledSteer, -1000, 1000, 180, 0);
					xScale = lookUpTableAngle[(uint16_t)steerAngle];

					// Mix steering and speed value for right and left speed
					if(steerAngle >= 90)
					{
						pwmSlave = CLAMP(scaledSpeed, -1000, 1000);
						pwmMaster = CLAMP(pwmSlave / xScale, -1000, 1000);
					}
					else
					{
						pwmMaster = CLAMP(scaledSpeed, -1000, 1000);
						pwmSlave = CLAMP(xScale * pwmMaster, -1000, 1000);
					}
				}
				else
				{
					// simply boost/drop speed left and right according to steer going from -1.0 to +1.0
					pwmMaster = speed * (1+steer/1000.0);		
					pwmSlave = speed * (1-steer/1000.0);
				}
			#endif
```

The idea is, that still different Remotes can be used to talk to the same Pilot.
For my Hoverbike, the speed coming from a Remote can be used by the Pilot as the max speed.
This max speed could come from RemoteUart, RemoteUartBus but also from a throttle being read by RemoteAdc.
My PilotHoverbike will have the final word on the speed sent to the Driver, based on brake and pedaling detection.

Currently, the new 15% speed boost is hard coded with 

```c++
//#define BLDC_SINE_BOOSTER		// can boost speed by 15% starting from 87% throttle.
```

But when i have merged the two `bldc_get_pwm` functions, a Pilot can deactivate the booster when battery is low, for example.

@WestlingPi , you already wanted to discuss the code of PilotDefault. But it might cause confusion if your RemoteROS2 will also need a PilotROS2. 
So we could add a PilotNull which does nothing at all. `pwmMaster` and `pwmSlave` are global variables anyway, so you could already compute these values in RemoteROS2.

I was tempted to copy RemoteUart into a new RemoteHoverbike and also put my pilot code there.
But i think it is a good idea to give every user a new playground for their onboard code: PilotXY.c

This time, i have not added a "virtual class" Pilot.h, should i ?
To add a new Pilot, add the include to defines.h

```c++
#ifdef PILOT_HOVERBIKE
	#include "PilotHoverbike.h"
#endif
```

and in your .h file put 

```c++
#define PILOT_XY
void 	Pilot(int16_t* pPwmMaster, int16_t* pPwmSlave);
```

The `PILOT_XY` will tell main.c to call the Pilot function instead of the default old code.

Ideas welcome.

RoboDurden, 2025-08-19:
made it a bit clearer in defines.h : 

```c++
#ifdef PILOT_HOVERBIKE
	#include "PilotHoverbike.h"		// https://youtu.be/ihCpCtgXIRA
#elif defined (PILOT_New)		
	#include "PilotNew.h"		// for your new onboard user code :-)
#else
	#define PILOT_DEFAULT
#endif
```

and main.c :

```c++
			#ifdef PILOT_DEFAULT
				if (iDrivingMode == 0)
				{
					// Calculate expo rate for less steering with higher speeds
					expo = MAP((float)ABS(speed), 0, 1000, 1, 0.5);
					...
			#else
				Pilot(&pwmMaster,&pwmSlave);
			#endif
```

@WestlingPi , i have moved your `#define STAND_STILL_THRESHOLD 10` from config.h to your `RemoteROS2.h` file.
In config.h there should be only defines for the user to select, not some underlying constants that make the  magic work.

WestlingPi, 2025-08-20:
I think the new Pilot framework is a great idea. My first thought was to introduce a "ROS2 Pilot" for the reverse translation from leftSpeed/rightSpeed to speed/steer that the ROS2 hoverbaord-driver does. However, for the ROS2 usecase I agree that a null pilot would be a better option. Then all the ROS2 hoverboard-driver adaptations could be placed in RemoteROS2 (would just be confusing to spread the code in 2 different places with dependencies between them).

I think moving `STAND_STILL_THRESHOLD` to RemoteROS2 is good - it was just a hack because the "Default Pilot" wasnt doing what I wanted. With a null pilot it is not needed.

With 3 pilots: Default, HoverBike and Null, maybe a "virtual" Pilot.h should be introduced. But it can alway be introduced later if you would like it to "mature" a little bit more... I have no strong opinion other than I would want to use the "Null Pilot".

RoboDurden, 2025-08-20:
Yes i agree, that your code should be spread onto two files.
But there is an esay way for your to put the Pilot() function into RemoteROS2.h :

```c++
#undef PILOT_DEFAULT
void 	Pilot(int16_t* pPwmMaster, int16_t* pPwmSlave);

#endif // REMOTE_ROS2_H
```

RemoteROS2.c :

```c++
void 	Pilot(int16_t* pPwmMaster, int16_t* pPwmSlave)
{
	*pPwmMaster = CLAMP(speed + steer,-1000,+1000);	// or something like that
	*pPwmMaster = CLAMP(speed - steer,-1000,1000);	// or something like that
}

#endif // REMOTE_ROS2
```

A PilotX.h is included before the RemoteW.h so that the pilot can define the BAUDRATE before the default 19200 is set, like i do with my PilotHoverbike.h : 

```c++
#define REMOTE_BAUD 115200		// increase default baud rate from 19200 to higher speed
```

That is why you have to `#undef PILOT_DEFAULT` to activate a user `Pilot()` function.
I think that is okay. People can always click on `Pilot(&pwmMaster,&pwmSlave);` in `main.c` and hit `F12` to jump to the function definition.

WestlingPi, 2025-08-20:
Undefining `PILOT_DEFAULT` works for me too. With this appoach all the ROS2 adaptations can be turned on with one define in config.h (REMOTE_ROS2) which is what I was after...

I see you have added almost all the code needed commented out in RemoteROS2.h/.c :-). Thanks!