RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #131: Optimal 64 MHz clock for STM32F103 ...

RoboDurden, 2025-08-30:
@WestlingPi  , @hoverboardhavoc  let's continue here with this clock issue.

<blockquote>
Indeed:
>When the HSI (=high-speed internal clock signal) is used as a PLL clock input, the maximum system clock frequency that can be achieved is 64 MHz.

 Please no modifications to system files like system_gd32f10x.c. I think these get easily replace while making changes to the RTE runtime enviroment ?

 Better add a new Keil target and add the define there:
[image]

```c++
#ifdef __SYSTEM_CLOCK_48M_PLL_IRC8M
uint32_t SystemCoreClock = __SYSTEM_CLOCK_48M_PLL_IRC8M;
```
comes first, so the default 72M are ignored.

With this i get 
[image]
on my Gen1 board and msTicks again verified with stop watch.
But 48MHz is not optimal anyway ? How would you achieve 64MHz ?
</blockquote>

So i asked ChatGpt5 again:

<blockquote>
please give me the Keil init code for STM32F103C8 to run without external oscillator at the max clock of 64 MHz
In the datasheet i read: When the HSI is used as a PLL clock input, the maximum system clock frequency that can be achieved is
64 MHz.
</blockquote>
<blockquote>
I do not want to touch system_stm32f10x.c .  And the code is also used for GD32f103, so my system file is system_gd32f10x.c
Is it possible to detect at the beginning of main() that the mcu is a stm32f103 and then set the clock from the default 72 Mhz to 64 Mhz ?
</blockquote>

This is the final code that might work for stm32f103:

```c++
#if TARGET==2		
		
	void clock_init_stm32(void)
	{
		/* 1. Enable internal 8 MHz oscillator (IRC8M = HSI) */
		RCU_CTL |= RCU_CTL_IRC8MEN;
		while((RCU_CTL & RCU_CTL_IRC8MSTB) == 0);

		/* 2. Configure Flash wait states for 64 MHz 
			 (2 wait states needed for 48–72 MHz range) */
		FMC_WS &= ~0x7;   // clear WSCNT[2:0]
		FMC_WS |= 0x2;    // 2 wait states

		/* 3. Configure PLL: IRC8M / 2 * 16 = 64 MHz */
		RCU_CFG0 &= ~(RCU_CFG0_PLLMF | RCU_CFG0_PLLSEL);
		RCU_CFG0 |= (RCU_PLLSRC_IRC8M_DIV2 | RCU_PLL_MUL16);

		/* 4. Set prescalers: 
					AHB = /1 (64 MHz), 
					APB1 = /2 (32 MHz, must be =36 MHz), 
					APB2 = /1 (64 MHz) */
		RCU_CFG0 &= ~(RCU_CFG0_AHBPSC | RCU_CFG0_APB1PSC | RCU_CFG0_APB2PSC);
		RCU_CFG0 |= (RCU_AHB_CKSYS_DIV1 | RCU_APB1_CKAHB_DIV2 | RCU_APB2_CKAHB_DIV1);

		/* 5. Enable PLL */
		RCU_CTL |= RCU_CTL_PLLEN;
		while((RCU_CTL & RCU_CTL_PLLSTB) == 0);

		/* 6. Switch system clock to PLL */
		RCU_CFG0 &= ~RCU_CFG0_SCS;
		RCU_CFG0 |= RCU_CKSYSSRC_PLL;
		while((RCU_CFG0 & RCU_SCSS_PLL) == 0);

		/* 7. Update SystemCoreClock variable */
		SystemCoreClock = 64000000UL;
	}
	uint32_t dev_id = 0;	
	uint32_t pll_mul = 0;
#endif
		
int main (void)
{
	iBug = 1;
	ConfigRead();		// reads oConfig defined in defines.h from flash
	
	#if TARGET==2
		#define DBGMCU_IDCODE   (*(volatile uint32_t*)0xE0042000)
		#define DEV_ID_MASK     0x00000FFF
		#define STM32F103_DEV   0x410   // STM32F103
		// GD32F103 will read as something else (typically 0x419)
		dev_id = DBGMCU_IDCODE & DEV_ID_MASK;		// will be 1044 for GD32F103RC
		if (1)//(dev_id == STM32F103_DEV) 
			clock_init_stm32();   // 64 MHz
		else
			SystemCoreClockUpdate();  // ensure GD32 sets 72 MHz correctly

		pll_mul = (RCU_CFG0 & RCU_CFG0_PLLMF) >> 18;  // bits differ per header, check what value you actually get		
			// will be 1 on GD32F103, should be 14 on true stm32f103.
	#else
		SystemCoreClockUpdate();
	#endif
  SysTick_Config(SystemCoreClock / 1000);	//  Configure SysTick to generate an interrupt every millisecond

	#ifdef MASTER_OR_SINGLE
		FlagStatus chargeStateLowActive = SET;
		int16_t pwmMaster = 0;
		int16_t scaledSpeed = 0;
		int16_t scaledSteer  = 0;
		float expo = 0;
		float steerAngle = 0;
		float xScale = 0;
	#endif
```
I get this for my 2.2.7 with gd32f103RC when i force clock_init_stm32() function:

[image]

But for 10000ms msTicks, my stop watch only gives 9 seconds.
ChatGpt then made me check pll_mul  and this confirmed that for gd32f103, you still get 72M.
But for stm32f103, it should work.

Please test :-)

WestlingPi, 2025-08-30:
Yes, the code works on my stm32f103:

[image]

I would rather use a compile time #define. I dont trust chinese clones to set or dont set `dev_id = DBGMCU_IDCODE & DEV_ID_MASK;` correctly...

However if I run SystemCoreClockUpdate() after clock_init_stm32(); it thinks SystemCoreClock is 48MHz. So I am not sure if it runs on 48MHz or 64MHz... 

Great progress anyway this weekend! I will test more in the evening... Thanks for your effort!

RoboDurden, 2025-08-30:
If 10000 msTicks take 10 seconds, your cpu is running on 64M.
If your mcu is running on 48MHz, then your stop watch should show 13.3 seconds for 10000 msTicks (i guess)

WestlingPi, 2025-08-30:
With AI code modified to:
```
		if (dev_id == STM32F103_DEV) {
			DEBUG_RCU_CFG0_Value1 = RCU_CFG0;
			clock_init_stm32();   // 64 MHz
			DEBUG_RCU_CFG0_Value2 = RCU_CFG0;
			SystemCoreClockUpdate();
		}
```

[image]

RoboDurden, 2025-08-30:
- So how long do 10000 msTicks take with a stop watch for the code i posted ?
- What pll_mul do you get ?

WestlingPi, 2025-08-30:
> * So how long do 10000 msTicks take with a stop watch for the code i posted ?
> * What pll_mul do you get ?

I tried your code with __SYSTEM_CLOCK_48M_PLL_IRC8M in compiler arguments so before your code starts to execute it is configured to run at 48MHz.

30000 msTicks took ~40 sec on stopwatch so 10000 msTicks should have been  13,33sec

RCC_CFGR / RCU_CFG0:

[image]

so pll_mul is: 1010

1010: PLL input clock x 12

I.e. same as set due to  __SYSTEM_CLOCK_48M_PLL_IRC8M

WestlingPi, 2025-08-30:
This code from Deepseek from yesterday I think accually managed to change PLLMUL. I will try if I can modify it:

```
void stm32_clock_config_to_72mhz(void)
{
    /* Define register addresses */
    #define RCC_CR        (*(volatile uint32_t *)(0x40021000U))
    #define RCC_CFGR      (*(volatile uint32_t *)(0x40021004U))
    #define FLASH_ACR     (*(volatile uint32_t *)(0x40022000U))

    /* Define bit masks for RCC_CR */
    #define RCC_CR_HSION    (1 << 0)
    #define RCC_CR_HSIRDY  (1 << 1)
    #define RCC_CR_PLLON   (1 << 24)
    #define RCC_CR_PLLRDY  (1 << 25)

    /* Define bit masks for RCC_CFGR */
    #define RCC_CFGR_PLLSRC_HSI_DIV2 (0 << 16)
    #define RCC_CFGR_PLLMUL_18       (0x10 << 18) // 16 << 18 = 0x04000000

    #define RCC_CFGR_HPRE_DIV1   (0 << 4)
    #define RCC_CFGR_PPRE1_DIV2  (0x4 << 8)  // 0x400
    #define RCC_CFGR_PPRE2_DIV1  (0 << 11)

    #define RCC_CFGR_SW_HSI      (0 << 0)
    #define RCC_CFGR_SW_HSE      (1 << 0)
    #define RCC_CFGR_SW_PLL      (2 << 0)
    #define RCC_CFGR_SWS_PLL     (2 << 2)

    #define FLASH_ACR_LATENCY_2WS (2 << 0)

    /* 0. SAFETY FIRST: Switch system clock back to HSI if it's using the PLL */
    /* Read the current clock source */
    uint32_t reg = RCC_CFGR;
    uint32_t sw = reg & 0x3;

    /* If the system clock is currently sourced from the PLL... */
    if (sw == RCC_CFGR_SW_PLL) {
        /* Switch it back to HSI */
        RCC_CFGR = (reg & ~0x3) | RCC_CFGR_SW_HSI; // Clear SW bits, set to HSI
        /* Wait until the switch is complete */
        while (((RCC_CFGR >> 2) & 0x3) != 0); // Wait for SWS to become 0 (HSI)
    }

    /* 1. Now it's safe to disable the PLL */
    RCC_CR &= ~RCC_CR_PLLON;       // Disable PLL
    /* Wait for PLL to unlock/disable. 
       NOTE: This while loop might be unnecessary or hang if the PLL was in a bad state.
       If it still hangs here, you might need to remove this wait. */
    // while(RCC_CR & RCC_CR_PLLRDY); // Use with caution

    /* 2. Enable HSI (ensure it's on) */
    RCC_CR |= RCC_CR_HSION;
    while((RCC_CR & RCC_CR_HSIRDY) == 0) { /* Wait */ }

    /* 3. Configure Flash for 72 MHz (2 wait states) */
    FLASH_ACR = FLASH_ACR_LATENCY_2WS;

    /* 4. Configure AHB, APB1, APB2 prescalers */
    RCC_CFGR &= ~(0xF << 4);   // Clear HPRE
    RCC_CFGR &= ~(0x7 << 8);   // Clear PPRE1
    RCC_CFGR &= ~(0x7 << 11);  // Clear PPRE2
    RCC_CFGR |= (RCC_CFGR_HPRE_DIV1 | RCC_CFGR_PPRE1_DIV2 | RCC_CFGR_PPRE2_DIV1);

    /* 5. Configure the PLL: Source = HSI/2, Multiplier = 18 */
    RCC_CFGR &= ~( (1 << 16) | (0xF << 18) ); // Clear PLLSRC and PLLMUL
    RCC_CFGR |= (RCC_CFGR_PLLSRC_HSI_DIV2 | RCC_CFGR_PLLMUL_18);

    /* 6. Enable the PLL and wait for it to lock */
    RCC_CR |= RCC_CR_PLLON;
    while((RCC_CR & RCC_CR_PLLRDY) == 0) { /* Wait */ }

    /* 7. Switch the system clock source to the PLL */
    RCC_CFGR &= ~0x3;        // Clear SW bits
    RCC_CFGR |= RCC_CFGR_SW_PLL; // Select PLL
    while((RCC_CFGR & 0xC) != RCC_CFGR_SWS_PLL) { /* Wait for SWS to become 2 */ }

    // Clean up local defines
    #undef RCC_CR
    #undef RCC_CFGR
    #undef FLASH_ACR
    // ... undef all others ...
}
```

RoboDurden, 2025-08-30:
>I tried your code with __SYSTEM_CLOCK_48M_PLL_IRC8M ...

Well that is not what i had in mind. And why don't you simply read pll_mul with StmStudio ? 

I read iTestClock = 3376 for 72 Mhz:

```c++
uint32_t iTestClock;

	uint32_t iTimeStart = millis();
	for (volatile uint32_t i = 72000000/4; i > 0; i--);	// 72000000/4 NOP operations would take 250 ms at 72 MHz ?
	iTestClock = millis()-iTimeStart;
```

Next will be for 48Mhz..

RoboDurden, 2025-08-30:
And yes. i get 0x410 as identifier for my 2.2.1 gd32f103c8 board too. So we definitely need a #define in your stm32f103 defines2_2-y.h

WestlingPi, 2025-08-30:
Sorry, I didnt realize pll_mul was a variable in the code:

[image]

RoboDurden, 2025-08-30:
```c++
uint32_t iTestClock = 42;

	SysTick_Config(SystemCoreClock / 1000);	//  Configure SysTick to generate an interrupt every millisecond

	uint32_t iTimeStart = millis();
	for (volatile uint32_t i = 72000000; i > 0; i--);	// 72000000 NOP operations would take 1000 ms at 72 MHz ?
	iTestClock = millis()-iTimeStart;
```
for 48 Mhz (__SYSTEM_CLOCK_48M_PLL_IRC8M in the Keil Target preprocessor Define) i get iTestClock = 18010 (msTicks) = 10 seconds. And indeed it took approx 18 seconds from power on to the startup melody.
So if you do not trust msTicks, then simply use stop watch.

for 72 Mhz i get iTestClock = 12004 and indeed it took 12.6 seconds on my stop watch.

12 seconds / 48 * 72 = 18 seconds 👍 

I think this is the way to truly test at what clock speed the mcu is running :-)

WestlingPi, 2025-08-30:
Is this the brute force test?
```
uint32_t iTestClock = 42;

	SysTick_Config(SystemCoreClock / 1000);	//  Configure SysTick to generate an interrupt every millisecond

	uint32_t iTimeStart = millis();
	for (volatile uint32_t i = 72000000; i > 0; i--);	// 72000000 NOP operations would take 1000 ms at 72 MHz ?
	iTestClock = millis()-iTimeStart;
```
I get iTestClock = 13505 (with the deepseek code above).
msTicks seems to corespond well to stopwatch...

RoboDurden, 2025-08-30:
Good job, sleep well 🥇 

I will add your code tomorrow and upload it. will put `clock_init_stm32()` into setup.c.
I think we should add the `#define STM32F103` to config.h ?

```c++
	#elif GD32F103		// TARGET = 2
		#define LAYOUT 1
		//#define STM32F103 	// uncomment if your board has stm32f103 instead of gd32f103
	#elif GD32E230		// TARGET = 3
```

I think that some boards come with both MCUs ?

WestlingPi, 2025-08-31:
Yes, the update system call will set it correctly. You can uncomment it if you want. I think the functions in system_gd32f103.c set it too. But I think the update system call should always be used. I trust that code more than the AI code :-)

WestlingPi, 2025-08-31:
New commit looks great! Thanks!